roboticslab-uc3m/yarp-devices #2: Add --recovery option on BodyBot

jgvictores, 2015-02-03:
To recover from motor driver failures.

rsantos88, 2016-04-19:
Now, you can use “checkCanBus” program with “--resetAll” parameter for reset all nodes defined in .ini files (see --help parameter) or “--resetNode” parameter followed by node number, to send reset message to one driver. In this state the drivers perform a software reset and enter the pre-operational state. This is very useful for recover from motor driver failures (red led). 
Also, this application checks the drivers and CUI absolute encoders just after these are reseted or initiated.  
Added in (#56)

jgvictores, 2016-04-19:
Very cool! This actually closes #27!
What is actually pursued with this issue is:
- [ ] Being able to pass a --recovery option to TechnosoftIpos that enables its self recovery during initialization.
- [ ] Allowing this parameter to flow all the way from launchManipulation/launchLocomotion to TechnosoftIpos (passing through CanBusControlboard, etc).

PeterBowman, 2019-07-29:
**Self** recovery is hard or impossible to attain, will talk about that at https://github.com/roboticslab-uc3m/yarp-devices/issues/185. 

> Now, you can use “checkCanBus” program with “--resetAll” parameter for reset all nodes defined in .ini files (see --help parameter) or “--resetNode” parameter followed by node number, to send reset [...]

Once fault state is reached, https://github.com/roboticslab-uc3m/yarp-devices/issues/215 will make it possible to traverse back to *Operation enabled*. Meddling with the NMT state machine feels like overkill.

PeterBowman, 2019-10-04:
I'm closing this as WONTFIX. Fault state recovery will be easily attained via `setControlMode(j, VOCAB_CM_FORCE_IDLE)`, see https://github.com/roboticslab-uc3m/yarp-devices/issues/215. On the other hand, the reset node NMT operation will only happen on device close since all SDO configuration parameters are lost in that step.

closed 2022-06-04 by PeterBowman