roboticslab-uc3m/yarp-devices #9: [bodybot] Could not open CAN device of path: /dev/can0

edwinDOS, 2015-02-05:
When running testBodyBot, I have the following error:
```
[error] CanBusHico.cpp:13 init(): Could not open CAN device of path: /dev/can0
yarpdev: ***ERROR*** driver <bodybot> was found but could not open
cannot make <bodybot>
yarpdev: ***ERROR*** driver <controlboard> was found but could not open
[error] TestBodyBot.cpp:19 configure(): Class instantiation not worked.
[error] TestBodyBot.cpp:20 configure(): Be sure CMake "ENABLE_BodyYarp_bodybot" variable is set "ON"
[error] TestBodyBot.cpp:21 configure(): "SKIP_bodybot is set" --> should be --> "ENABLE_bodybot is set"
```

jgvictores, 2015-02-05:
Could you paste the output of the following commands?
`ls -l /dev/can0`
`grep hcanpci /proc/devices`

jgvictores, 2015-03-10:
Please check if the following new guide on installing the hcanpci module can help:
http://robots.uc3m.es/dox-teo-body/teo_body_install_hcanpci_on_debian_6.html

closed 2015-03-23 by jgvictores