roboticslab-uc3m/yarp-devices #11: Could not "make" ASIBOT code

iagson, 2015-02-18:
Hi, this is my first time posting here, I hope it's good enough. Now to the point.

I was trying to compile the ASIBOT code, but encountered a problem when using "make -j3"
It stops at 78%, right after building the CartesianServerLib. The image below is what the terminal outputs after the error.

[image]

I used this as installation guide, as is posted in the Sourceforge ASIBOT repository:

svn co https://svn.code.sf.net/p/roboticslab/code/asibot asibot
cd asibot/main; mkdir build; cd build
cmake .. -DENABLE_cartesianServer=ON -DENABLE_RlPlugins_cartesianbot=ON -DENABLE_RlPlugins_ravebot=ON
make -j3; sudo make install; cd ../../..

Thank you in advance

jgvictores, 2015-02-18:
@iagson This corresponds to the https://svn.code.sf.net/p/roboticslab/code/asibot repository, which still has not been migrated to GitHub and has become quite outdated. Just for the sake of compiling, I've commited a fix. Please "svn up" in "asibot" and make in "asibot/main/build" again.

iagson, 2015-02-18:
It compiled perfectly, thank you.

jgvictores, 2015-02-20:
Okay, closing issue. For futher reference, this is a https://github.com/roboticslab-uc3m/asibot-main issue, so similar new issues should be posted at https://github.com/roboticslab-uc3m/asibot-main/issues/new.

closed 2015-02-20 by jgvictores