roboticslab-uc3m/yarp-devices #16: BodyBot --mode flag not working?

jgvictores, 2015-03-09:
From gravity compensation tests, mode has to be set explicity further on in the code.

David-Estevez, 2016-01-26:
As @rsantos88  and I have checked, mode is not working because only certain sections of the configuration file are passed to the DeviceDrivers (i.e. only the "devcan0" section is passed to can0 device). Command line flags have to be appended to the concrete sections so that the drivers have access to all comman line flags.

jgvictores, 2016-01-26:
i.e. only the "devcan0" section is passed to can0 device -> This is good, they are initial configuration parameters.
"mode" should issue the change mode command **after** the drivers have been enabled (*enable*, I believe, is performed in what is currently named *setCanBusPtr*). Could you check if this->setVelocityMode() or this->setTorqueMode() reach the driver (TechnosoftIpos) level at runtime?

rsantos88, 2016-01-26:
When I force, for example, torque mode executing "this->setTorqueMode()", the console show this: 
[warning] ICanBusSharerImpl.cpp:257 interpretMessage(): 	-Mode "251" not specified in manual, may be in Fault or not enabled yet. canId(ID)- 
This warning message is indicated for IDs: 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26.

rsantos88, 2016-02-03:
When we force position mode and we put the check status rpc commands: [get] [icmd] [cmds], we can see this:
Response: [is] [cmds] ([posd] [posd] [posd] [posd] [posd] [posd] [posd]) [tsta] 0 1454508630.629655 [ok]
This is good. In the same way, when we force velocity mode, we can see:
Response: [is] [cmds] ([veld] [veld] [veld] [veld] [veld] [veld] [posd]) [tsta] 0 1454508440.639366 [ok]
However, when we force torque mode, it shows:
Response: [is] [cmds] ([fail] [fail] [fail] [fail] [fail] [fail] [posd]) [tsta] 0 1454508689.538105 [ok]

rsantos88, 2016-02-04:
I have done some tests with Yarp RPC commands  without any modification of the code and the results are this:
When you run launchManipulation, it starts in position mode. If you change to torque mode with  [set] [trqd] and you check the status mode with [get] [icmd] [cmds] , it shows: Response: [is] [cmds] ([fail] [fail] [fail] [fail] [fail] [fail] [posd]) [tsta] 1 1454578574.957852 [ok] but you can move the articulations of the robot.
If we change again to a position mode, the articulations turned be rigid and it says Response: [is] [cmds] ([posd] [posd] [posd] [posd] [posd] [posd] [posd]) [tsta] 2 1454578610.86074 [ok] when you check the status. This is right, but if you put the robot in a torque mode once again, it will be regid all the time. Torque mode will stop working.

rsantos88, 2016-02-09:
Now, there is an updated version of launchManipulation in Git with optional “--mode” flash working correctly in the master branch.