roboticslab-uc3m/yarp-devices #26: [idea] Refractor CMake files

jgvictores, 2015-11-28:
A great number of new CMake features would make our CMake files cleaner and better.

jgvictores, 2017-02-16:
As commented with @PeterBowman this issue is mainly related with all the [exporting dependencies for (...) quite manually for now...](https://github.com/roboticslab-uc3m/teo-main/blob/54b5cc7cd9764ff0245e6d2866f486e5b2aa08d3/libraries/TrajectoryLib/CMakeLists.txt#L33) comments.

PeterBowman, 2017-02-17:
I'm afraid we are somehow constrained by having to deal with Debian's outdated (6.0) SO. Of course, these lines could be improved a bit anyway, in that respect I found YARP's use of CMake a valuable source for discovering features and good (I hope so) practices in dealing with project configuration. I noticed yesterday some ongoing work targeted towards next YARP releases to support CMake 3.0, let me link it here just for interest's sake: robotology/yarp@215c823. Keep in mind that teo-body is stuck at 2.8.9 which, incidentally, is guaranteed to be supported in [current YARP distributions](http://wiki.icub.org/wiki/YARP_Supported_Distributions), thus many ideas could be borrowed from them for use in this repo.

PeterBowman, 2017-05-19:
Most of the constraints mentioned in my earlier comment could be collaterally solved with #118.

PeterBowman, 2017-11-29:
Example: roboticslab-uc3m/kinematics-dynamics#14.

PeterBowman, 2017-12-08:
WIP at the `fix-26-cmake` branch. I wonder if using YCM to fetch ColorDebug was a good choice... I'm unable to configure it on my PC with CMake v2.8.9 (same as Debian 6.0) because of a `Unknown CMake command "write_basic_package_version_file"` error at [CMakeLists.txt#L102](https://github.com/robotology/ycm/blob/v0.2.2/CMakeLists.txt#L102). What happened: YCM tries to download `CMakePackageConfigHelpers.cmake` from CMake's GitHub repo at rev. 3.1.3, but for some reason it gets an empty file (!) as a result, which then shadows the local version shipped with CMake itself.

I need to test it on Debian 6.0, but from what I recall, YCM didn't work with this same CMake version on Ubuntu Precise.

PeterBowman, 2017-12-08:
I managed to solve that by digging in our own installation guides: [install-cmake.md#openssl-support](https://github.com/roboticslab-uc3m/installation-guides/blob/a4be467648a8512041e6895b554cf3c47101485a/install-cmake.md#openssl-support). Funnily enough, I run into this same issue a few months ago, identified the cause and added that section in `install-cmake.md`. Now I need to find out whether CMake packages are compiled with OpenSSL support, otherwise tell users to compile it according to the installation guide.

Let me express my relief with a bunch of fine sarcasm.

[image]

PeterBowman, 2017-12-11:
> Now I need to find out whether CMake packages are compiled with OpenSSL support

Confirmed, Debian 6.0 backports support SSL.

PeterBowman, 2017-12-11:
Done at 891df8f.

closed 2017-12-11 by PeterBowman