roboticslab-uc3m/yarp-devices #32: Define the scope of the repository

David-Estevez, 2016-02-03:
I have seen some commits lately for the "waiter" application on this repo, and I was wondering if we should keep this repo just for generic functionalities (drivers o modules that can be used in several different applications) and create new repositories for specific applications.

What do you think, guys?

(See also: [teo-head - issue11](https://github.com/roboticslab-uc3m/teo-head/issues/11) )

David-Estevez, 2017-02-24:
I guess it is now clear that this repo is related only to robot-related code and not application-related code.

closed 2017-02-24 by David-Estevez

PeterBowman, 2017-02-24:
This repo might be slightly repurposed to hold a collection of sensor/actuator YARP devices in the near future (that is, expanding current CAN and jr3 devices + tts/asr from teo-head). See roboticslab-uc3m/QA#2.

David-Estevez, 2017-02-24:
I didn't know...Thanks, @PeterBowman !! :smile: :+1: