roboticslab-uc3m/yarp-devices #33: [launchManipulation] Passing all command-line arguments to Property Object

David-Estevez, 2016-02-08:
When extracting a group from a Resource Finder, only arguments coming from the config file are passed to the Property Object. Command-line arguments are ignored, and this may cause problems in the future.

Possible solutions:
* Append command line arguments to group arguments coming from the file as a new group in the Property Object
* Add support in YARP for extracting parameters depending on the origin of them (either command-line or file)

PeterBowman, 2018-10-07:
Relevant lines upstream: [ResourceFinder.cpp](https://github.com/robotology/yarp/blob/0e67a89e8c147d6fc9547a937ca45768f1ca52d7/src/libYARP_OS/src/ResourceFinder.cpp#L188-L207). I'd deem that placing the call to [`yarp::os::Property::fromConfigFile()`](http://www.yarp.it/classyarp_1_1os_1_1Property.html#a275d87fffe69cd2333b777b721967ab3) last would give precedence to command-line user arguments over `--from`.

PeterBowman, 2018-10-07:
> Add support in YARP for extracting parameters depending on the origin of them (either command-line or file)

Compare:

* [`yarp::os::Property::fromArguments`](http://www.yarp.it/classyarp_1_1os_1_1Property.html#abfd96557e2cebc3473e297afbdcf412e)
* [`yarp::os::Property::fromConfigFile`](http://www.yarp.it/classyarp_1_1os_1_1Property.html#a275d87fffe69cd2333b777b721967ab3)

...and others.

PeterBowman, 2018-10-08:
Reported upstream at https://github.com/robotology/yarp/issues/1889.

PeterBowman, 2019-12-30:
> Reported upstream at [robotology/yarp#1889](https://github.com/robotology/yarp/issues/1889).

The former *launchManipulation* app has been improved and renamed to *launchCanBus* in https://github.com/roboticslab-uc3m/yarp-devices/issues/20. Also, the .ini parse system has been fully revamped (https://github.com/roboticslab-uc3m/yarp-devices/issues/237) and there is currently no urgent need to accept additional command line arguments in our CAN-based tools. Therefore, I'm marking this as invalid and superseded by the upstream issue.

PeterBowman, 2019-12-30:
> Add support in YARP for extracting parameters depending on the origin of them (either command-line or file)

See [`yarp::os::ResourceFinderOptions`](http://www.yarp.it/classyarp_1_1os_1_1ResourceFinderOptions.html) and related methods at [`yarp::os::ResourceFinder`](http://www.yarp.it/classyarp_1_1os_1_1ResourceFinder.html).

closed 2022-06-04 by PeterBowman