roboticslab-uc3m/yarp-devices #43: Create image and partition of working manipulation and locomotion OSs

jgvictores, 2016-02-29:
Create image and bootable partition on PCs of working manipulation and locomotion OSs.

rsantos88, 2016-04-22:
Ya se ha realizado el Backup de los 3 PCs y se han subido los ficheros .fsa a http://robots.uc3m.es/svn/TEOrepo/backups/.

rsantos88, 2016-04-26:
Se ha reducido la partición actual de los PCs: manipulation y locomotion a aproximadamente 90 GB (87,69 GB), creando una nueva partición adicional de aproximadamente 20 GB (19,53 GB) con formato ext4 y etiqueta con nombre "backup". De la misma manera, la partición principal de PC Head se ha reducido la partición a 45,97 GB, creando también una partición adicional (más pequeña debido a que el HDD tiene la mitad de capacidad que los otros PCs) de aproximadamente 10 GB (9,77 GB) con mismo formato y nombre.
Por último, hemos copiado las imágenes que hemos creado anteriormente con FSArchiver a las particiones correspondientes.

rsantos88, 2016-04-29:
**IMPORTANTE:** Se están realizando cambios en el ordenador de manipulación para poder crear una segunda partición arrancable por lo que se ruega no guardar datos importantes en este ordenador sin tener previamente una copia de seguridad. El backup del mismo en caso de pérdida de datos se ha realizado el 22/04/2016.

rsantos88, 2016-05-06:
Ya está realizado el backup de todos los discos de los PCs: _teo-head_, _manipulation_, y _locomotion_. Se han clonado particiones principales a una partición secundaria de aproximadamente 20GB arrancable que podremos seleccionar desde el menú de Grub con nombre "_Linux.....(Backup: 22/04/2016)_ " situada en la parte inferior de la lista de Grub. El procedimiento de como restaurar la imagen del sistema operativo (backup) se encuentra documentada en la wiki: http://robots.uc3m.es/index.php/Tutorial:_C%C3%B3mo_restaurar_un_S.O_en_una_partici%C3%B3n_arrancable

rsantos88, 2018-03-15:
Tal y como he hablado con @jgvictores , se va a instalar en `teo-head`  también una partición limpia con Ubuntu 14.04.

rsantos88, 2018-03-19:
Instalado Ubuntu 14.04 LTS en `teo-head`. Instalado software para cabeza en la nueva distribución. Funcionando :+1:

closed 2018-03-19 by rsantos88

jgvictores, 2018-03-19:
Thanks a LOT!!!! :+1: