roboticslab-uc3m/yarp-devices #44: [TechnosoftIpos] Timeout if no absolute encoder message is received

jgvictores, 2016-03-02:
Implement timeout in [TechnosoftIpos] if no absolute encoder message is received, for instance displaying an error message and then continue by defaulting to using relative encoders.

jgvictores, 2016-03-02:
Implement only on forcing timeout through --externalEncoderWait [s] parameter.

rsantos88, 2016-09-28:
Implemented in #77. 
Now, if you for run launchManipulation or launchLocomotion and you see for example a error message, like:
`Cui Absolute (PIC ID: 120) doesn't respond. Try using --externalEncoderWait [seconds] parameter with timeout higher than 0 ` 
the Absolute Encoder doesn't respond, so use --externalEncoderWait [seconds] parameter for using default relative encoder position.

closed 2016-09-28 by rsantos88