roboticslab-uc3m/yarp-devices #47: yarpmotorgui not getting limits

jgvictores, 2016-03-15:
yarpmotorgui-qt (gtk not tested) not getting robot joint limits. https://github.com/robotology/yarp/blob/eed2a0afaddf3806a7eb551262c2b8a2f3ea6d08/src/yarpmotorgui-qt/partitem.cpp#L180 gets a false return. Same happens on teo-main/programs/teoSim.

jgvictores, 2016-11-24:
As pointed out by @PeterBowman we need to implement yarp::dev::IControlLimits2

@rsantos88 Could you do this on CanBusControlboard and TechnosoftIpos?

This would be a great improvement!!

jgvictores, 2016-11-28:
Fixed at 6328dd5ce4f451b172a6fe176a226e42b463d212

@PeterBowman Thanks again for pointing out the fact that we needed to implement yarp::dev::IControlLimits2 !

closed 2016-11-28 by jgvictores

PeterBowman, 2016-11-28:
@jgvictores we've just tested these changes on ASIBOT with yarpmotorgui and it works like a charm, thanks!

jgvictores, 2016-11-28:
@PeterBowman Great, cool!