roboticslab-uc3m/yarp-devices #50: Velocity control fails in hip (joint 14)

j-lorente, 2016-04-01:
Joint 14 of TEO can be moved with position control but not with velocity control.

Velocity control works fine for other joints.

j-lorente, 2016-04-01:
Please, don't use TEO's hip by now, Santi is looking for the problem with the velocity control.

j-lorente, 2016-04-05:
We checked the driver's configuration and the problem is not there so it must be in a higher level, in the software.

jgvictores, 2016-04-05:
I'd recommend trying with oneCanBusOneWrapper, best if you physically disconnect the rest of the CAN nodes.

j-lorente, 2016-04-06:
Using oneCanBusOneWrapper the joint can be moved with velocity control, so we conclude that the other driver in the hip, which is broken, is probably causing problems.

Then we configured the .ini for ignoring this broken driver and physically took it out, but we still can't move the joint in velocity mode with launchLocomotion.

jgvictores, 2016-04-06:
For the launchLocomotion .ini file, to preserve the wrapper numbering, I'd recommend simply replacing the "TechnosoftIpos" string for "FakeJoint" in the position corresponding to the faulty driver.

smcdiaz, 2016-04-06:
New driver arrived last week. Please keep the .ini as is, without modifications. Raul can change the broken driver for the new one

rsantos88, 2016-04-07:
Now, there are three files .ini in teo-body PC located at “/repos/teo-body/share/launchLocomotion/conf” with the following names: "launchLocomotion.ini"  "launchLocomotion_without_113_114.ini"  "launchLocomotion_without_13_113_114.ini" for testing launchLocomotion with --from parameter.

j-lorente, 2016-04-07:
We changed and configured driver 14 and we were able to move joint 14 in velocity mode using the Technosoft program.

We tried using launchLocomotion (both ignoring and not ignoring driver 13) and still doesn't work.

jgvictores, 2018-05-21:
This issue was moved to roboticslab-uc3m/teo-hardware-issues#25

closed 2018-05-21 by jgvictores