roboticslab-uc3m/yarp-devices #51: [BodyYarp] devices should be shared libraries

jgvictores, 2016-04-04:
[BodyYarp] devices should be shared libraries.

jgvictores, 2016-10-13:
Working on this in teo-main first: https://github.com/roboticslab-uc3m/teo-main/issues/60

jgvictores, 2017-01-23:
Essentially https://github.com/jgvictores/example-yarp-plugin-gtest is a good example of a repository containing a YARP plugin compiled as a shared library. Important:
1. In [root CMake of repo](https://github.com/jgvictores/example-yarp-plugin-gtest/blob/1c13b96a2133635d316983ef2d0111937b0cb7d2/CMakeLists.txt#L16) (maybe in teo-body could be elsewhere):
`option(BUILD_SHARED_LIBS "All shared libs" TRUE)`
2. In reality, `yarp_begin_plugin_library` and `yarp_end_plugin_library` ([line 9](https://github.com/jgvictores/example-yarp-plugin-gtest/blob/1c13b96a2133635d316983ef2d0111937b0cb7d2/libraries/CMakeLists.txt#L9) and [line 11](https://github.com/jgvictores/example-yarp-plugin-gtest/blob/1c13b96a2133635d316983ef2d0111937b0cb7d2/libraries/CMakeLists.txt#L11) of the library parent directory) are not actually required for shared libraries.
3. Need new `.ini` file per plugin. Just imitate [ExampleLibrary.ini](https://github.com/jgvictores/example-yarp-plugin-gtest/blob/1c13b96a2133635d316983ef2d0111937b0cb7d2/libraries/ExampleLibrary/ExampleLibrary.ini).
4. Need to adapt CMakeLists.txt per plugin. Just imitate [the CMakeLists.txt of ExampleLibrary](https://github.com/jgvictores/example-yarp-plugin-gtest/blob/1c13b96a2133635d316983ef2d0111937b0cb7d2/libraries/ExampleLibrary/CMakeLists.txt).

jgvictores, 2017-01-23:
Just in case, I would also recommend having YARP installed with `CREATE_SHARED_LIBRARY` activated.

jgvictores, 2017-01-23:
PS: From https://github.com/asrob-uc3m/robotDevastation-robots I've just seen that https://github.com/jgvictores/example-yarp-plugin-gtest is not updated to the latest of YARP. Here are the two relevant fixes:
- [update yarp_prepare_device -> yarp_prepare_plugin with CATEGORY device](https://github.com/asrob-uc3m/robotDevastation-robots/commit/9dc5d4dff9aae94f541c77f560127d06071dccce)
- [WRAPPER RobotServer -> EXTRA_CONFIG WRAPPER=RobotServer](https://github.com/asrob-uc3m/robotDevastation-robots/commit/73838a2aa3cbba3f3ab4d09e6786234666fab8d3)

PeterBowman, 2017-01-23:
@jgvictores `EXTRA_CONFIG` is a recent addition to the `yarp_prepare_plugin` command, it won't work with pre-2.3.68 YARP.

rsantos88, 2017-01-24:
I tried to do the first shared library with CanBusControlboard. Here is the [commit](https://github.com/rsantos88/teo-body/commit/a839ad43be821fd8d7c5f527b57c65d964e4438b) but, when I've tried to do cmake .., it shows this:
```bash
-- The C compiler identification is GNU 5.2.1
-- The CXX compiler identification is GNU 5.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Libraries go to /home/teo/repos/rsantos88/teo-body/build/lib
-- Executables go to /home/teo/repos/rsantos88/teo-body/build/bin
-- Setting up installation of teo-body.ini to /usr/local/share/yarp/config/path.d folder.
--  +++ plugin CuiAbsolute, SKIP_CuiAbsolute is set
--  +++ plugin TechnosoftIpos, SKIP_TechnosoftIpos is set
--  +++ plugin LacqueyFetch, SKIP_LacqueyFetch is set
--  +++ plugin TextilesHand, SKIP_TextilesHand is set
--  +++ plugin FakeJoint, SKIP_FakeJoint is set
--  +++ plugin CanBusHico, SKIP_CanBusHico is set
--  +++ plugin CanBusControlboard, SKIP_CanBusControlboard is set
CMake Error at libraries/BodyYarp/CMakeLists.txt:32 (install):
  install TARGETS given target "BodyYarp" which does not exist in this
  directory.

CMake Error: Could not open file for write in copy operation /BodyYarp_yarpdev.cpp.tmp
CMake Error: : System Error: Permission denied
CMake Error at /usr/local/share/yarp/cmake/YarpPlugin.cmake:357 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  libraries/BodyYarp/CMakeLists.txt:36 (YARP_ADD_PLUGIN_YARPDEV_EXECUTABLE)

CMake Error at libraries/BodyYarp/CMakeLists.txt:37 (install):
  install TARGETS given target "launchBodyYarp" which does not exist in this
  directory.

-- Found PythonInterp: /usr/bin/python (found version "2.7.10") 
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring incomplete, errors occurred!
See also "/home/teo/repos/rsantos88/teo-body/build/CMakeFiles/CMakeOutput.log".
See also "/home/teo/repos/rsantos88/teo-body/build/CMakeFiles/CMakeError.log".
```

rsantos88, 2017-01-24:
Now, with this [commit](https://github.com/rsantos88/teo-body/commit/3dfd8b2a139a68600a27519b7eff118ba1c1b0a2), the last problem is solved, but when I try to compile, it shows this error:
```bash
[ 46%] Building CXX object programs/recordLocomotion/CMakeFiles/recordLocomotion.dir/RecordRateThread.cpp.o
[ 50%] /usr/local/lib/libBodyYarp.a(add_BodyYarp_plugins.cpp.o): En la función `add_BodyYarp_plugins':
add_BodyYarp_plugins.cpp:(.text+0x1f): referencia a `add_owned_CuiAbsolute(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x29): referencia a `add_owned_TechnosoftIpos(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x33): referencia a `add_owned_LacqueyFetch(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x3d): referencia a `add_owned_TextilesHand(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x47): referencia a `add_owned_FakeJoint(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x51): referencia a `add_owned_CanBusHico(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x5b): referencia a `add_owned_CanBusControlboard(char const*)' sin definir
collect2: error: ld returned 1 exit status
programs/recordManipulation/CMakeFiles/recordManipulation.dir/build.make:166: recipe for target 'bin/recordManipulation' failed
make[2]: *** [bin/recordManipulation] Error 1
CMakeFiles/Makefile2:869: recipe for target 'programs/recordManipulation/CMakeFiles/recordManipulation.dir/all' failed
make[1]: *** [programs/recordManipulation/CMakeFiles/recordManipulation.dir/all] Error 2
Building CXX object programs/playbackLocomotion/CMakeFiles/playbackLocomotion.dir/PlaybackThread.cpp.o
[ 53%] Building CXX object programs/playbackLocomotion/CMakeFiles/playbackLocomotion.dir/main.cpp.o
Linking CXX executable ../../bin/recordLocomotion
Linking CXX executable ../../bin/playbackLocomotion
/usr/local/lib/libBodyYarp.a(add_BodyYarp_plugins.cpp.o): En la función `add_BodyYarp_plugins':
add_BodyYarp_plugins.cpp:(.text+0x1f): referencia a `add_owned_CuiAbsolute(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x29): referencia a `add_owned_TechnosoftIpos(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x33): referencia a `add_owned_LacqueyFetch(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x3d): referencia a `add_owned_TextilesHand(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x47): referencia a `add_owned_FakeJoint(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x51): referencia a `add_owned_CanBusHico(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x5b): referencia a `add_owned_CanBusControlboard(char const*)' sin definir
collect2: error: ld returned 1 exit status
programs/recordLocomotion/CMakeFiles/recordLocomotion.dir/build.make:141: recipe for target 'bin/recordLocomotion' failed
make[2]: *** [bin/recordLocomotion] Error 1
CMakeFiles/Makefile2:969: recipe for target 'programs/recordLocomotion/CMakeFiles/recordLocomotion.dir/all' failed
make[1]: *** [programs/recordLocomotion/CMakeFiles/recordLocomotion.dir/all] Error 2
/usr/local/lib/libBodyYarp.a(add_BodyYarp_plugins.cpp.o): En la función `add_BodyYarp_plugins':
add_BodyYarp_plugins.cpp:(.text+0x1f): referencia a `add_owned_CuiAbsolute(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x29): referencia a `add_owned_TechnosoftIpos(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x33): referencia a `add_owned_LacqueyFetch(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x3d): referencia a `add_owned_TextilesHand(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x47): referencia a `add_owned_FakeJoint(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x51): referencia a `add_owned_CanBusHico(char const*)' sin definir
add_BodyYarp_plugins.cpp:(.text+0x5b): referencia a `add_owned_CanBusControlboard(char const*)' sin definir
collect2: error: ld returned 1 exit status
programs/playbackLocomotion/CMakeFiles/playbackLocomotion.dir/build.make:141: recipe for target 'bin/playbackLocomotion' failed
make[2]: *** [bin/playbackLocomotion] Error 1
CMakeFiles/Makefile2:1019: recipe for target 'programs/playbackLocomotion/CMakeFiles/playbackLocomotion.dir/all' failed
make[1]: *** [programs/playbackLocomotion/CMakeFiles/playbackLocomotion.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

```

PeterBowman, 2017-01-24:
You are still linking to `/usr/local/lib/libBodyYarp.a` somewhere, in my case the compiler doesn't even get that far if `libBodyYarp.a` is not installed.

PeterBowman, 2017-01-24:
@jgvictores

> 2. In reality, yarp_begin_plugin_library and yarp_end_plugin_library (line 9 and line 11 of the library parent directory) are not actually required for shared libraries.

I'd keep them as long as the `BUILD_SHARED_LIBS` option/variable is not forced since users may still want to link statically. Similarly, `yarp_add_plugin_yarpdev_executable` CMake plugin and `YARP_DECLARE_PLUGINS` / `YARP_REGISTER_PLUGINS` C macros (used in `main.cpp` in most programs) should be kept for the same reason.

Just as a side note: if the `yarp_begin_plugin_library` / `yarp_end_plugin_library` bundle is omitted and `BUILD_SHARED_LIBS` is set, all contained plugins are disabled by default (as you can see [here](https://github.com/roboticslab-uc3m/teo-body/issues/51#issuecomment-274790928): `SKIP_CuiAbsolute is set` and so on) in that they will still compile, but cannot be dynamically loaded.

jgvictores, 2017-01-27:
In fact, the consequences are quite brutal regarding code and cmake (take a look at the current SharedLibraries branch at https://github.com/roboticslab-uc3m/teo-body/network). I'm extremely inclined to force shared libraries, keeping static libraries seems like a giant effort (possibly requiring ugly hacks or at least ugly code) with no visible benefits.

PS: `SKIP_CuiAbsolute` etc were `SKIP_BodyYarp_CuiAbsolute` etc before. This has been corrected on the branch (https://github.com/rsantos88/teo-body/commit/7a7b378cb1c32dc69f4a89b690d1a634e7891664).

PeterBowman, 2017-01-27:
I believe some of those changes would be necessary even for static libraries in terms of a major cleanup, I just recently found that instructions like this do not make much sense (apart from populating a public interface, which may or may not be intended):

```cmake
target_link_libraries(TEO_BODY_STATIC_PLUGIN ${YARP_LIBRARIES} ${TEO_BODY_LIBRARIES})
```

It might not be that hard nor hackish to provide support for static libraries. Having that said, I'm not against forcing shared YARP plugins/TEO libraries/both, but in that case, I'd stress that `BUILD_SHARED_LIBS` should be set as a CMake variable rather than an user-configurable option. BTW there is a YARP variable available from 2.3.64 that could be investigated: `YARP_FORCE_DYNAMIC_PLUGINS`.

PeterBowman, 2017-02-07:
@jgvictores you might also want to add b342044 to the list at https://github.com/roboticslab-uc3m/teo-body/issues/51#issuecomment-274515888.

jgvictores, 2017-02-07:
Done at 92a3cf72b102ed92d690fc80d16dd883eafe1df3.

closed 2017-02-07 by jgvictores