roboticslab-uc3m/yarp-devices #55: GIT error: server certificate verification failed

lpinel, 2016-04-18:
I tried to pull, push... and later clone a git repository on locomotion PC and something went wrong. :'(

error: server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none while accessing https://github.com/roboticslab-uc3m/project-generator.git/info/refs

fatal: HTTP request failed

lpinel, 2016-04-18:
I tried : 
 git config --global http.sslverify false

and it worked, but don't know if it is the best way

jgvictores, 2016-04-19:
Seems like we'll be living dangerously for a while.

BTW, just accessing https://github.com/roboticslab-uc3m/project-generator.git/info/refs through a web browser returns: `Please upgrade your git client.
GitHub.com no longer supports git over dumb-http:` https://github.com/blog/809-git-dumb-http-transport-to-be-turned-off-in-90-days`

Closing issue as git is functional again.

closed 2016-04-19 by jgvictores