roboticslab-uc3m/yarp-devices #58: Error installing Gpart in teo-manipulation PC

rsantos88, 2016-04-25:
I have tried to install Gpart  with Software Center and it shows an error:

> "Requieres installation of untrusted packages: this action would require the installation of packages from not authenticated source. "

In "Edit>Software Sources" I have checked the following options: 
officially supported (main)

- DSFG-compatible
- Software with Non-Free Dependencies (contrib)
- Non-DSFG-compatible
- Software (non-free)
- Source code

When I close the options windows it says: 

> "Failed to download repository information. Check your Internet connection." 

Other form (and the easiest) through terminal:

`teo@manipulation:~$ sudo apt-get install gparted`
`[sudo] password for teo: `
`Reading package lists... Done`
`Building dependency tree`
`Reading state information... Done`
`N: Ignoring file 'google-chrome.list.save' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension`
`E: Unable to locate package gparted`

Then, I have tried to update the packages list with "sudo apt-get update" but it shows this error:

`teo@manipulation:~$ sudo apt-get update`
`Ign http://ftp.es.debian.org squeeze-updates Release.gpg`
`Ign http://ftp.es.debian.org/debian/ squeeze-updates/main Translation-en`
`Ign http://ftp.es.debian.org squeeze-updates Release`
`Ign http://ftp.es.debian.org squeeze-updates/main Sources/DiffIndex`
`Ign http://ftp.es.debian.org squeeze-updates/main i386 Packages/DiffIndex      `
`Ign http://ftp.es.debian.org squeeze-updates/main Sources                      `
`Ign http://ftp.es.debian.org squeeze-updates/main i386 Packages                `
`Err http://ftp.es.debian.org squeeze-updates/main Sources                      `
`  404  Not Found`
`Err http://ftp.es.debian.org squeeze-updates/main i386 Packages                `
`  404  Not Found`
`Get:1 http://security.debian.org squeeze/updates Release.gpg [836 B]           `
`Ign http://security.debian.org/ squeeze/updates/contrib Translation-en         `
`Ign http://security.debian.org/ squeeze/updates/main Translation-en            `
`Ign http://security.debian.org/ squeeze/updates/non-free Translation-en        `
`Hit http://security.debian.org squeeze/updates Release                         `
`Get:2 http://ppa.launchpad.net maverick Release.gpg [316 B]                    `
`Ign http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu/ maverick/main `
`Translation-en Ign http://debian.cc.lehigh.edu squeeze/updates Release.gpg                    `
`Ign http://debian.cc.lehigh.edu squeeze Release.gpg                            `
`Ign http://debian.cc.lehigh.edu/debian/ squeeze/contrib Translation-en         `
`Ign http://debian.cc.lehigh.edu/debian/ squeeze/main Translation-en            `
`Ign http://debian.cc.lehigh.edu/debian/ squeeze/non-free Translation-en        `
`Hit http://ppa.launchpad.net maverick Release                                  `
`Ign http://http.us.debian.org squeeze Release.gpg                              `
`Hit http://security.debian.org squeeze/updates/main i386 Packages              `
`Ign http://http.us.debian.org/debian/ squeeze/contrib Translation-en           `
`Ign http://http.us.debian.org/debian/ squeeze/main Translation-en              `
`Ign http://http.us.debian.org/debian/ squeeze/non-free Translation-en          `
`Ign http://http.us.debian.org squeeze/updates Release.gpg                      `
`Ign http://http.us.debian.org squeeze-proposed-updates Release.gpg             `
`Ign http://http.us.debian.org/debian/ squeeze-proposed-updates/contrib Translation-en`
`Ign http://http.us.debian.org/debian/ squeeze-proposed-updates/main Translation-en`
`Ign http://http.us.debian.org/debian/ squeeze-proposed-updates/non-free Translation-en`
`Hit http://ppa.launchpad.net maverick/main i386 Packages                       `
`Hit http://security.debian.org squeeze/updates/contrib i386 Packages           `
`Hit http://security.debian.org squeeze/updates/non-free i386 Packages          `
`Ign http://debian.cc.lehigh.edu squeeze/updates Release              `
`Ign http://debian.cc.lehigh.edu squeeze Release                      `
`Get:3 http://backports.debian.org squeeze-backports Release.gpg [819 B]`
`Hit http://backports.debian.org/debian-backports/ squeeze-backports/main Translation-en`
`Ign http://http.us.debian.org squeeze Release  `
`Ign http://debian.cc.lehigh.edu squeeze/updates/main Sources`
`Ign http://debian.cc.lehigh.edu squeeze/updates/contrib Sources`
`Ign http://debian.cc.lehigh.edu squeeze/updates/non-free Sources`
`Ign http://http.us.debian.org squeeze/updates Release`
`Hit http://backports.debian.org squeeze-backports Release`
`Ign http://http.us.debian.org squeeze-proposed-updates Release`
`N: Ignoring file 'google-chrome.list.save' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension`
`E: Release file expired, ignoring http://backports.debian.org/debian-backports/dists/squeeze-backports/Release (invalid since 36d 6h 41min 9s)`

I would be very grateful if somebody help me.
Thanks!!

jgvictores, 2016-04-25:
Looks similar to #49.
Comment out or erase everything from /etc/apt/sources.list leaving only
deb http://archive.debian.org/debian/ squeeze main non-free contrib deb-src http://archive.debian.org/debian/ squeeze main non-free contrib
Also get any apt.d or sources.list.d etc out of there.
sudo apt-get update # should go good

jgvictores, 2016-04-25:
BTW, we should also add backports to get the right cmake version (note for the future :smile:).

closed 2016-04-28 by rsantos88