roboticslab-uc3m/yarp-devices #62: [Jr3] Card connection fails sometimes

lpinel, 2016-06-06:
A veces al encender el PC de manipulación, los LEDs de la tarjeta DAQ están apagados (los de comunicación con la tarjeta y los de alimentación de los sensores).  Y el comando lsipci no reconoce el adaptador PCI->PCIe "Pericon".
Entonces se debe encender y apagar el ordenador  (con reiniciarlo no basta). ¿A qué se puede deber el problema?

lpinel, 2016-06-13:
Hoy al ejecutar el programa de lectura de datos de los sensores, se apagaron de golpe todos los LED de la tarjeta. 
Al hacer lspci aparece el adaptador Pericon PCI bridge.
Al hacer lsmod, aparece el módulo jr3pci_driver por lo que no es tema de fallo de inclusión del módulo.

jgvictores, 2018-05-21:
This issue was moved to roboticslab-uc3m/teo-hardware-issues#24

closed 2018-05-21 by jgvictores