roboticslab-uc3m/yarp-devices #63: [Wifi] Slow communications

rsantos88, 2016-06-08:
Se han detectado lentitud en las comunicaciones con los ordenadores de manipulación y locomoción utilizando Wifi en las siguientes situaciones:
 * Con el qtcreator abierto por ssh en el ordenador de manipulación, se aprecian bloqueos constantes del programa cuando se da la situación de que Loli está trabajando a la vez con otra terminal conectada por ssh al mismo PC para realizar otras tareas (en este caso, únicamente la adquisición de datos de los sensores).

* Loli haciendo pruebas con las piernas ha detectado también lentitud. Para ello necesita recibir datos de los sensores de los pies a través de Yarp desde el ordenador de manipulación. La conexión entre puertos tarda en hacerse (más de 5 segundos). Al editar un documento por ssh, navegar por la terminal..etc se percibe lentitud.

jmgarciah, 2016-06-08:
Creo que mientras el robot no camine, sería una buena idea volver con los cables ethernet. Nos estaríamos asegurando la velocidad de las comunicaciones y a la vez no habría problemas en los tirones, pues no se lanzan trayectorias de caminata. :+1:

jgvictores, 2016-06-09:
El router actual tiene 4 conectores para cables. Si se vuelve a la configuración con cables, no habrán conectores libres: manipulation, locomotion, head, main (mandatory, e.g. because yarp server there).

lpinel, 2016-06-15:
Hoy ha sido necesario resetear el router por fallo en las comunicaciones. No se tenía acceso a los pc's de TEO ni a Internet

rsantos88, 2016-06-15:
Las comunicaciones siguen yendo bastante lentas cuando trabajan dos personas con la misma conexión. Mientras se esté desarrollando con Teo, (y no perjudique) propongo el uso de cable. Si el router está limitado a 4 conectores... ¿no se puede pedir comprar otro con más puertos o un switch?
Saludos cordiales

smcdiaz, 2016-06-15:
Me encargo de buscar soluciones hardware

rsantos88, 2016-06-15:
Gracias @smcdiaz !!

smcdiaz, 2016-06-15:
http://www.netgear.es/home/products/networking/wifi-routers/R8500.aspx?cid=gwmng#tab-especificacionestécnicas

http://www.netgear.es/home/products/networking/wifi-adapters/A6100.aspx?cid=gwmng

rsantos88, 2016-06-15:
Propongo también la posibilidad de comprar simplemente un switch para ampliar el número de puertos que tenemos actualmente: http://www.netgear.es/home/products/networking/switches/soho-ethernet-switches.aspx?cid=gwmng

smcdiaz, 2016-06-15:
Ok. Pero aun así es necesario mejorar las comunicaciones inalámbricas.

rsantos88, 2016-07-18:
Se ha añadido en la Wiki información sobre la instalación de los drivers para los adaptadores A6100 WiFi USB en: http://robots.uc3m.es/index.php/TEO_Network_information.

rsantos88, 2016-09-22:
Al tratar de instalar el paquete "sudo apt-get install linux-headers-generic build-essential git" en el PC de manipulación, me saca el siguiente error:

teo@manipulation:~$ sudo apt-get install linux-headers-generic build-essential git
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package linux-headers-generic is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-headers-generic' has no installation candidate

rsantos88, 2016-09-26:
Se han corregido los pasos a seguir para instalar correctamente el nuevo modelo de wifi (Netgear A6100) en la [Wiki](http://robots.uc3m.es/index.php/TEO_Network_information#Setting_up_Wifi_connection:_Netgear_A6100_WiFi_USB_Mini_Adapter) en el apartado **Setting up Wifi connection: Netgear A6100 WiFi USB Mini Adapter**.
Seguidos los pasos, nos encontramos con el problema de que, aún habiendo instalado correctamente los drivers, el sistema operativo no asocia correctamente los drivers instalados con el dispositivo en cuestión.

rsantos88, 2017-11-15:
Posiblemente este problema se pueda solucionar cuando se lleve a cabo el cambio de distribución de linux en los PCs de TEO ([issue](https://github.com/roboticslab-uc3m/yarp-devices/issues/118) relacionada). Por el momento, todos los PCs están conectados a través de cable ethernet.

jgvictores, 2017-11-15:
This issue was moved to roboticslab-uc3m/teo-main#24

closed 2017-11-15 by jgvictores