roboticslab-uc3m/yarp-devices #68: [Cui Absolute] Integrating Cui 113 in launchLocomotion.ini

rsantos88, 2016-06-29:
Actualmente, el launchLocomotion está funcionando sin los encoders absolutos 113 y 114. El encoder 114 aún se encuentra sin el cable de CAN. En cambio el 113 se encuentra operativo y mandando correctamente su posición. Al  integrar el 113 en el launchManipulation.ini, cuando accedemos al tronco, los valores de los encoders, tanto del 113 como del 114, permanecen con valor 0. Además, cuando se trata de mover la cintura (giro axial), no responde. La solución temporal es desconectar por el momento el encoder 113 y tenerlos desactivados en el fichero .ini hasta que ambos encoders estén operativos.

jgvictores, 2016-06-29:
Vi algún .ini provisional que evita el 119. ¿Algo ocurre con este CuiAbsolute?

rsantos88, 2016-06-29:
Si. En un par de ocasiones, la conexión entre el PIC y el Cui ha fallado y se han dejado de recibir mensajes.

rsantos88, 2016-06-29:
Aún así @jgvictores , ahora mismo la issue se centra en los CuiAbsolute del tronco y su integración en el launchLocomotion.ini. Gracias.

rsantos88, 2016-07-06:
Recientemente se han conectado los encoders 113 y 114. Ambos encoders se encuentran enviando mensajes correctamente. Cuando arrancamos el launchLocomotion y accedemos al tronco, ambos encoders dan valor 0.0. Tras mover ambas articulaciones, los valor de los encoders siguen marcando 0.0. ¿alguna solución?
Gracias

rsantos88, 2016-07-06:
Añadir que también se ha probado a usar el oneCanBusOneWrapper modificando el .ini e incluyendo el driver 14 y el Cui 114. Se ha probado y el valor obtenido al mover la articulación es igualmente (0.0)

rsantos88, 2016-09-07:
Se han reprogramado los encoders absolutos 113 y 114. Ambos envían mensajes correctamente.
Se ha descartado por tanto que sea un problema de hardware.  Además, añadir que cuando en el .ini se integran los encoders 113 y 114, al acceder a la posición de los encoders correspondientes a la pierna derecha y a la pierna izquierda, estos marcan 0 en todas sus articulaciones.

jgvictores, 2016-09-07:
IMHO, reprogramar involucra algo de hardware. ;-)

rsantos88, 2016-09-08:
IMO el problema de esta Issue es HW. 
NSFW hablar de HW, si ya se ha descartado. Todo apunta a que es el SW del LM.
Pls contesta ASAP ;-)

jgvictores, 2016-09-08:
AFAIK, si hay una EEPROM que se borra sin motivo, sería HW. Más si no hay cambios en el SW.
NSFW podría ser JLo, no HW.

rsantos88, 2016-09-08:
LOL! +1 yo también creo que puede ser JLo.

David-Estevez, 2016-09-08:
OMG! WTF??

rsantos88, 2016-09-22:
Tema solucionado. Con las nuevas modificaciones del nuevo launchManipulation & launchLocomotion (#75), los encoders absolutos (113 y 114) son reconocidos perfectamente por el software.
Cierro esta issue :stuck_out_tongue_winking_eye:

closed 2016-09-22 by rsantos88