roboticslab-uc3m/yarp-devices #74: [CanBusControlboard] clean open() and close()

jgvictores, 2016-09-19:
CanBusControlboard open() and close() functions contain class-specific functionalities that can and should be moved to the corresponding classes "TechnosoftIpos", "CuiAbsolute", etc.

PeterBowman, 2017-05-19:
We spoke recently about moving CANopen-related method to a new interface (perhaps `ICanOpen`?), namely `start`, `readyToSwitchOn`, `switchOn` and `enable` (now located in [`ICanBusSharer.h`](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef/libraries/BodyYarp/ICanBusSharer.h)), which would essentially end up leaving the most relevant method in the `ICanBusSharer` interface (and not much more): `interpretMessage`. Among what will be left, `setCanBusPtr` could be dropped in favour of passing an `ICanBusHico` pointer through YARP properties prior to each device initial configuration. We can see an example in the openrave-yarp-plugins tree: [setter](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/5483749/openraveplugins/OpenraveYarpControlboard/OpenraveYarpControlboard.cpp#L163-L164), [getter](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/4afbb7c/yarpplugins/YarpOpenraveControlboard/DeviceDriverImpl.cpp#L10). Furthermore, we would like to explore the idea of using multiple threads, one per device's open and close, in order to speed up things and avoid *staging* time-consuming CAN commands across several for-loops with delays ([ref](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef/libraries/BodyYarp/CanBusControlboard/DeviceDriverImpl.cpp#L265-L288)).

jgvictores, 2017-05-19:
1. `ICanOpen`. Perfect name. I think those methods are good. Some more inspiration at [1](http://docs.ros.org/fuerte/api/canopen/html/classcanopen_1_1CANopenMotorDriver.html), but before adding more methods it would be best to contrast at [the official source](https://www.can-cia.org/canopen/) whether if these are actual CAN Open commands (downside is accessing documentation is not easy).

2. Replacing `setCanBusPtr` as described. Great.

3. More than speeding up things through multiple threads, I'd start by replacing empirically set hard-coded delays by checks that assure the stages have been reached.

PeterBowman, 2017-12-24:
> Among what will be left, `setCanBusPtr` could be dropped in favour of passing an ICanBusHico pointer through YARP properties prior to each device initial configuration.

Confirming that this was not meant to last per https://github.com/roboticslab-uc3m/yarp-devices/pull/158#issuecomment-353778180:

> I like the idea of passing the pointer via blob. The additional setter methods were just hacks before I had experimented doing so on the openrave-yarp-plugins repo. :-)

See how to implement this with proper casts in https://github.com/roboticslab-uc3m/yarp-devices/pull/158#issuecomment-353777110.

PeterBowman, 2018-10-17:
I don't whink we'd benefit from that library anyway, at least not in the scope of this issue:

> It is not the intention of CANpie to incorporate higher layer functionality (e.g. CANopen, J1939).

We encapsulate CAN frame functionality within every CAN bus device (CanBusHico, CanBusPeak, future CanBusSocket and CanBusEsd, even CanBusFake with fake frames). There is no need to expose a common API in that respect as our intermediate layer is already the YARP `ICanBus` and related interfaces.

jgvictores, 2018-12-21:
Blocks #171.

PeterBowman, 2019-05-04:
> Among what will be left, `setCanBusPtr` could be dropped in favour of passing an `ICanBusHico` pointer through YARP properties prior to each device initial configuration.

I'd rather abandon this approach, let's restrict CAN comms to the CanBusControlboard device (if possible): https://github.com/roboticslab-uc3m/yarp-devices/issues/209#issuecomment-489372536.

PeterBowman, 2019-07-29:
Recently, the `ICanBusSharer::initialize` method was added to configure drives via CAN SDO protocol prior to NMT initialization: https://github.com/roboticslab-uc3m/yarp-devices/commit/297b070596af3fa2487a0e53cda732b5adc8b8f5.

PeterBowman, 2019-09-10:
Next step: re-design the `ICanBusSharer` interface ([develop](https://github.com/roboticslab-uc3m/yarp-devices/blob/380480966fd16bed8c279c6fc32d241799bf9f08/libraries/YarpPlugins/ICanBusSharer.h), [current WIP](https://github.com/roboticslab-uc3m/yarp-devices/blob/9a9d1f94d31ac3722f64aa9e072a89860981780a/libraries/CanBusSharerLib/ICanBusSharer.hpp)).

PeterBowman, 2019-09-10:
Workflow:

* Device instantiation: `CanBusControlboard` calls `DeviceDriver::open` on each raw subdevice.
* Pass CAN sender handle (currently `CanSenderDelegate`, name is subject to change) so that subdevices can actually write stuff. 
* Initialization phase: this corresponds to the NMT pre-operational state for CANopen-enabled subdevices. Here, SDOs are used in order to perform initial configuration, including PDO mappings and communications. No PDOs allowed.
* Issue NMT start command (CANopen-based): transition to operational state, PDOs allowed.
* Commands related to CiA 402 state machine: shutdown, switch on, enable operation and disable voltage. I'm deferring the deletion or refactorization of these commands to issue https://github.com/roboticslab-uc3m/yarp-devices/issues/215.
* Interpret message: handles incoming CAN data.

So far, no changes?

PeterBowman, 2019-09-10:
I think the following comment from https://github.com/roboticslab-uc3m/yarp-devices/issues/211#issuecomment-516919532 belongs to here:

> Following the .ini rework proposal, it might be wise to parse an additional per-device type group (e.g. all Cuis, all iPOS nodes, etc.) to gather common configuration parameters within (e.g. PT/PVT mode period).
>
> Besides, I'd like to take advantage of the .ini file transclusion feature: let's put all iPOS/Cuis/grippers on their own separate .ini file and transclude them if needed from the parent .ini storing the configuration of the CanBusControlboard master device.

PeterBowman, 2019-09-10:
CuiAbsolute stuff cleaned at https://github.com/roboticslab-uc3m/yarp-devices/commit/a0ad0340158a0fcea4ed99dc7def54f6adce88a2, solved https://github.com/roboticslab-uc3m/yarp-devices/issues/207 by the way.

PeterBowman, 2019-09-13:
CiA state machine transitions (e.g. switch on, enable, etc.) with hardcoded timeouts refactored at https://github.com/roboticslab-uc3m/yarp-devices/commit/972f8758d246bbb2a85c40cf5af11bd5855597a0.

PeterBowman, 2019-09-13:
Mode change shall be performed by the launcher application: https://github.com/roboticslab-uc3m/yarp-devices/commit/00916c9fe138fcec700c8809f138a477112c6bcb. CanBusControlboard's open procedure will be strictly focused on drive/node initialization, further actions are performed by external actors.

PeterBowman, 2019-09-14:
Homing refactored at https://github.com/roboticslab-uc3m/yarp-devices/commit/e1995b68e0422d4684a47990e50a798bb0168ce9, see also https://github.com/roboticslab-uc3m/yarp-devices/issues/215#issuecomment-531500948.

PeterBowman, 2019-09-25:
Mostly ready at https://github.com/roboticslab-uc3m/yarp-devices/blob/ad741da402f44683e7362c07366c9a215a054872/libraries/YarpPlugins/CanBusControlboard/DeviceDriverImpl.cpp, see https://github.com/roboticslab-uc3m/yarp-devices/blob/8726fa368f8b3becf26a2f263f4d451e1836eb1b/libraries/YarpPlugins/CanBusControlboard/DeviceDriverImpl.cpp (develop) for comparison. The .ini file cleanup will be discussed in https://github.com/roboticslab-uc3m/yarp-devices/issues/237.

closed 2019-12-31 by PeterBowman

PeterBowman, 2020-01-28:
The workflow has been slightly altered to allow deferred CAN node initialization, see https://github.com/roboticslab-uc3m/yarp-devices/issues/242. Now, `CanBusControlboard::open` is supposed to create wrapped subdevices (CAN nodes and buses) and start threads. If possible, CAN nodes are started at this point, too, but this is not required for the application to keep running. On power-on, a boot-up signal is issued and caught by the monitor thread to initiate the CAN configuration sequence.