roboticslab-uc3m/yarp-devices #81: Cannot control axis 2-5 in ASIBOT with TeoBody code

PeterBowman, 2016-11-12:
ASIBOT's PC (more precisely, a RoBoard computer with a Debian system installed on a SD card) stores a local copy of teo-body to communicate with CAN drivers at a low level. We've observed that, out of 5 axes, only the first one moves as expected and the rest stand still, while RPC responses always return an `[ok]` acknowledgement. Tested code corresponds to an old version of the repo since we first issued RPC commands by means of a now-defunct program called `testBodyBot`. This used to work some months ago before the loss of SD card memory due to an unexpected shutdown, later reinstated from a saved .iso file. Once fixed, we found out that only axis 1 produced motion. Newly fetched and built teo-body code did not help; moreover, `oneCanBusOneWrapper` won't communicate with CAN until we launch `checkCanBus` or `dumpCanBus` in another terminal, but the end result is the same anyway.

This is the command that used to work and is currently installed in ASIBOT with old teo-body libraries, all of this dates from ~1.5 year ago. Note the `motoripos` device type, currently called `TechnosiftIpos`:

`$ testBodyBot --canDevice /dev/can0 --canBitrate 6 --ids 1 2 3 4 5 --types motoripos motoripos motoripos motoripos motoripos --trs 160 160 160 160 160 --maxs 90 90 90 90 90 --mins -90 -90 -90 -90 -90`

And these are the contents of the .ini file the `oneCanBusOneWrapper` command is launched with (axes/ids 2 to 5 are failing):
```
[devCan0]
device CanBusControlboard
canDevice /dev/can0
types TechnosoftIpos
ids   2
maxs  90
mins -90
trs   100
refAccelerations  0.575437
refSpeeds  5.0

[wrapper0]
device controlboardwrapper2
name /wrapper0
threadrate 60
joints 1
networks (devCan0)
devCan0 0 0 0 0
```

The output of `oneCanBusOneWrapper` upon launch:
```
[info] ICanBusSharerImpl.cpp:451 interpretMessage(): Got "manufacturer status register" from driver. 43 2 10 0 40 2 0 2. canId(2) via(580).
[info] ICanBusSharerImpl.cpp:468 interpretMessage(): 	-Motor supply voltage is absent. canId: 2.
[info] ICanBusSharerImpl.cpp:471 interpretMessage(): 	-Performing a quick stop. canId: 2.
[info] ICanBusSharerImpl.cpp:474 interpretMessage(): 	-Switch on disabled. canId: 2.
[info] ICanBusSharerImpl.cpp:483 interpretMessage(): 	-Remote: drive parameters may be modified via CAN and the drive will execute the command message. canId: 2.
[info] ICanBusSharerImpl.cpp:490 interpretMessage(): 	-Target not reached. canId: 2.
[info] ICanBusSharerImpl.cpp:498 interpretMessage(): 	-No event set or the programmed event has not occurred yet. canId: 2.
[info] ICanBusSharerImpl.cpp:503 interpretMessage(): 	-Axis off. Power stage is disabled. Motor control is not performed. canId: 2.
[info] ICanBusSharerImpl.cpp:534 interpretMessage(): 	*Target command reached. canId: 2.
[success] IControlModeRawImpl.cpp:225 getControlModeRaw3(): Sent Motion Error Register query. 40 0 20 0 0 0 0 0. canId(2) via(600).
[info] ICanBusSharerImpl.cpp:399 interpretMessage(): Got SDO ack "Motion Error Register" from driver. 4b 0 20 0 0 0 0 0. canId(2) via(580).
[success] IControlModeRawImpl.cpp:240 getControlModeRaw4(): Sent Detailed Error Register query. 40 2 20 0 0 0 0 0. canId(2) via(600).
[info] ICanBusSharerImpl.cpp:553 interpretMessage(): Got SDO ack "Detailed Error Register" from driver. 4b 2 20 0 0 0 0 0. canId(2) via(580).
[success] IControlModeRawImpl.cpp:179 getControlModeRaw1(): Sent modes of operation display. 40 61 60 0 0 0 0 0. canId(2) via(600).
[info] ICanBusSharerImpl.cpp:280 interpretMessage(): Got SDO "modes of operation display" from driver. 4f 61 60 0 0 0 0 0. canId(2) via(580).
[warning] ICanBusSharerImpl.cpp:329 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(2).
[success] IControlModeRawImpl.cpp:210 getControlModeRaw2(): Sent manufacturer status query. 40 2 10 0 0 0 0 0. canId(2) via(600).
```

The same for `dumpCanBus`:
```
[success] ThreadImpl.cpp:32 run(): Read CAN message: 4b 7e 20 0 0 0 0 0. canId(2) via(580)
[success] ThreadImpl.cpp:32 run(): Read CAN message: 4b 0 20 0 0 0 0 0. canId(2) via(580)
[success] ThreadImpl.cpp:32 run(): Read CAN message: 4b 2 20 0 0 0 0 0. canId(2) via(580)
[success] ThreadImpl.cpp:32 run(): Read CAN message: 43 64 60 0 0 0 0 0. canId(2) via(580)
[success] ThreadImpl.cpp:32 run(): Read CAN message: 4f 61 60 0 0 0 0 0. canId(2) via(580)
[success] ThreadImpl.cpp:32 run(): Read CAN message: 43 2 10 0 40 2 0 2. canId(2) via(580)
```

If we connect to `/wrapper0/rpc:i` and send `set pos 0 5`, this is the only extra line `dumpCanBus` spits out:
```
[success] ThreadImpl.cpp:32 run(): Read CAN message: 60 7a 60 0 0 0 0 0. canId(2) via(580)
```

I didn't observe any change in `oneCanBusOneWrapper`'s output, but I'll be running more tests in the next days to try to gather more data.

Ping @edwinDOS.

jgvictores, 2016-11-15:
Okay, let's go little by little.
1. As a basic rule, do not run programs simultaneously.
2. Is possible to physically connect/disconnect the TechnosoftIpos power without stopping the roboard?
3.  Copy and paste the output of `dumpCanBus` here. If 2 is true: run `dumpCanBus`, turn off TechnosoftIpos power, wait 5 seconds, do a ctrl-c to stop `dumpCanBus`, run `dumpCanBus`, turn on TechnosoftIpos power.
4. Try a `oneCanBusOneWrapper` ini more like:
```[devCan0]
device CanBusControlboard
canDevice /dev/can0
types TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos
ids   1 2 3 4 5
maxs  90 90 90 90 90
mins -90 -90 -90 -90 -90
trs   100 100 100 100 100
refAccelerations  0.575437  0.575437  0.575437  0.575437  0.575437
refSpeeds  5.0 5.0 5.0 5.0 5.0

[wrapper0]
device controlboardwrapper2
name /wrapper0
threadrate 60
joints 5
networks (devCan0)
devCan0 0 4 0 4
```

PeterBowman, 2016-11-15:
After collecting `testBodyBot`'s output in a file (~86 MB worth of ~~useless~~ tests), stripping dupes and other garbage, these lines may point at a new suspect - the motor constant `k`. Let me check it in a few hours, I'll follow your pointers if my hunch is wrong.
```
[success] DeviceDriverImpl.cpp:21 open(): Created MotorIpos with canId 1, tr 160.000000, k 0.070600, and all local parameters set to 0.
[success] ICanBusSharerImpl.cpp:10 setCanBusPtr(): Ok pointer to CAN bus device 1.
[success] DeviceDriverImpl.cpp:21 open(): Created MotorIpos with canId 2, tr 160.000000, k 0.000000, and all local parameters set to 0.
[success] ICanBusSharerImpl.cpp:10 setCanBusPtr(): Ok pointer to CAN bus device 2.
[success] DeviceDriverImpl.cpp:21 open(): Created MotorIpos with canId 3, tr 160.000000, k 0.000000, and all local parameters set to 0.
[success] ICanBusSharerImpl.cpp:10 setCanBusPtr(): Ok pointer to CAN bus device 3.
[success] DeviceDriverImpl.cpp:21 open(): Created MotorIpos with canId 4, tr 160.000000, k 0.000000, and all local parameters set to 0.
[success] ICanBusSharerImpl.cpp:10 setCanBusPtr(): Ok pointer to CAN bus device 4.
[success] DeviceDriverImpl.cpp:21 open(): Created MotorIpos with canId 5, tr 160.000000, k 0.000000, and all local parameters set to 0.
[success] ICanBusSharerImpl.cpp:10 setCanBusPtr(): Ok pointer to CAN bus device 5.
```

PeterBowman, 2016-11-16:
We managed to move all 5 axes by replacing the SD card with a freshly restored system backup (previous SD had some additional WiFi drivers installed and recent teo-body code pulled from GH). I have attached the output from running `testBodyBot` on both SDs in [this Gist](https://gist.github.com/PeterBowman/77ae4762822a2ea04d35db18f6f8f3fd). The first file refers to the previous card, only axis 1 responds to our commands (note that reference speeds and accelerations are set to zero for axes 2-5); the latter corresponds to the restored SD, all motors produce motion. This command was compiled at 0e58438. What could be failing here?

jgvictores, 2016-11-16:
It's reasonable that the a joint does not move upon a position movement command if the reference velocity and acceleration is zero. Does that answer you, or am I misunderstanding your question?

PeterBowman, 2016-11-16:
That's what I would expect to happen (no velocity/acceleration = no movement), but it **was** unclear to me why these values had been set to zero for joints 2 to 5. After a second thought, it seemed plausible that the omission of `refSpeeds` and `refAccelerations` in the command line while on the first SD may have had some influence, hence the bolded past tense. Unfortunately, I can't discard it until we finish some hardware tweaks currently in process (we've decided to raise up the CAN bus speed from 500 kb/s to 1 Mb/s). Once we reach a more "stable" state, I'll follow up. Sorry for my messy reports!

PeterBowman, 2016-11-18:
Long story short, our main issue got solved with a thorough inspection of .ini files and parameter defaults. That is, we did not specify any reference speeds nor accelerations for `testBodyBot`, thus taking a default value of zero for joints 2-5. I'm pasting the full command with params (updated for 1 Mb/s):

`$ testBodyBot --canDevice /dev/can0 --canBitrate 8 --ids 1 2 3 4 5 --types motoripos motoripos motoripos motoripos motoripos --trs 160 160 160 160 160 --maxs 90 90 90 90 90 --mins -90 -90 -90 -90 -90 --refSpeeds 737.2798 737.2798 737.2798 737.2798 737.2798 --refAccelerations 0.575437 0.575437 0.575437 0.575437 0.575437
`

Such a simple solution, unnoticed at first, was inadvertently being buried deeper as more things came into play. In the end, the motor constant `k` had no distinguishable effect (in fact, it may be left unset, or zeroed) and, more interestingly, I recall that your proposed .ini file in conjunction with `oneCanBusOneWrapper` could not produce motion in any joint. All of this could've been caused by a confluence of factors that we may not be able to reproduce anymore; however, we are convinced that several things were learned as a result of these misadventures. Thank you for your assistance!

jgvictores, 2016-11-19:
1. refSpeeds and refAccelerations are needed for positionMode commands, whereas k is required for torqueMode commands.

2. I'm reopening this issue. Would it be correct to rename this issue including "depreciated testBodyBot", close this issue, and open a new issue for `oneCanBusOneWrapper`?

PeterBowman, 2016-11-21:
I've run another test today - this time, we were able to control all joints with `oneCanBusOneWrapper` and the .ini file you specified [earlier](https://github.com/roboticslab-uc3m/teo-body/issues/81#issuecomment-260555317). Previous conditions cannot be reproduced because of recent hardware changes. I'd leave this issue closed as invalid unless you notice any other topic worth investigating further.

BTW in every piece of software I've worked with, [including YARP](http://www.yarp.it/deprecated.html), the verb *deprecate* is preferred over *depreciate* :). Both may be synonyms, though.

jgvictores, 2016-11-21:
> I've run another test today - this time, we were able to control all joints with oneCanBusOneWrapper and the .ini file you specified earlier. Previous conditions cannot be reproduced because of [...]

Ok, you can close this issue!

> BTW in every piece of software I've worked with, including YARP, the verb deprecate is preferred over depreciate :). Both may be synonyms, though.

Ok, lesson learned! :smile:

closed 2022-06-04 by PeterBowman