roboticslab-uc3m/yarp-devices #83: [TechnosoftIpos] enable digital outputs

jgvictores, 2016-11-24:
For the ASIBOT, and possibly for TEO, it would be great if we could control the digital outputs of the TechnosoftIpos. However, before going down this road on the TechnosoftIpos side, we have to see how thos goes through the remote_controlboard and CanBusControlboard flow.

PeterBowman, 2016-12-30:
i wrote a simple C program to control the OUT3/READY digital output of TechnosoftIpos drivers: https://gist.github.com/PeterBowman/db3c67205e83363ac02d6cac427b2646 (BTW it was supposed to be a first step to accomplish #85, but I'll defer the task of creating a fully working CLI program to a later phase). This is the output pin we'll use to command each brake in ASIBOT.

PeterBowman, 2016-12-31:
I forgot to mention a discordance between what the *iPOS CANopen Programming User Manual* says and the observed behavior. According to section **5.4.2. Object 60FEh: Digital outputs** (ref. P091.063.iPOS.UM.0615):

> The object controls the digital outputs of the drive. The first sub-index sets the outputs state to high or low if it is allowed by the mask in the second sub-index which defines the outputs that can be controlled.
> If any of the bits is SET, then the corresponding drive output will be switched to logical ‘1’ (high) and Switched ON. If the bit is RESET, then the corresponding drive output will be switched to logical ‘0’ (low) and Switched OFF.

It continues with a detailed object description and an example for setting digital outputs (5.4.2.1). According to that, the object entry mapped with sub-index 1 should hold the desired output states, and a bitmask at sub-index 2 specifies which outputs will be controlled. It didn't work for me with this configuration until I reversed the order. You can see [here](https://gist.github.com/PeterBowman/db3c67205e83363ac02d6cac427b2646/bbb5a3827eb7757791e86a793942370f1c9529a7#file-control_brakes-c-L153) that the corresponding bit is *always* set in the first entry (0x08 refers to OUT3) and expected states are defined a few lines further on in the second entry, depending on the value of a `brakeFlag` variable.

Actual output states were monitored with information obtained from object 2045h.

PeterBowman, 2019-12-30:
The iPOS drives can probably handle brakes on their own. That is, according to our experience with EasySetup and given available options, we presume the iPOS disables brakes on Motor On condition and enables them again on Motor Off. This is highly desirable.

This issue raised due to the fact that ASIBOT's electronics have serious issues. Namely, somebody forgot to wire the digital outputs to the iPOS pin that is actually supposed to activate the brake. Hence, we would like to command pin OUT3, which is the one connected to physical brakes.

The control flow in new TechnosoftIpos code is not that hard to attain thanks to the recent work on https://github.com/roboticslab-uc3m/yarp-devices/pull/229, but any progress on ASIBOT is blocked now due to https://github.com/roboticslab-uc3m/asibot-main/issues/44.

Since 1. handling of digital outputs has been proposed as a workaround for the proper solution (fix the electronics), 2. its implementation would feel like a hack (use general purpose digital IO to accomplish a specific goal), 3. this issue has been stalled for three years, and 4. nobody currently cares about ASIBOT, I'm marking this as WONTFIX for now.

closed 2022-06-04 by PeterBowman