roboticslab-uc3m/yarp-devices #84: Outdated "threadrate" parameter breaks on YARP 2.3.66.1

PeterBowman, 2016-11-24:
From [yarp/releases/tag/v2.3.66.1](https://github.com/robotology/yarp/releases/tag/v2.3.66.1):

> The `controlboardwrapper2` device (implemented in `yarp::dev::ControlBoardWrapper` has removed the supported for the `threadrate` parameter, that has been deprecated since 2014. Invalid configuration files of the form `threadrate 10` can be update with the valid form `period 10`.

Several .ini files have to be updated.

closed 2016-11-25 by PeterBowman

PeterBowman, 2016-11-25:
@jgvictores this might actually be worth a cherry pick to master.

jgvictores, 2016-12-01:
@PeterBowman I really never understood what you meant with that last comment :smile: ...to what do you refer? Something like we should check this among all repos? Or just the fact that keeping up with dependencies can be really tedious in general?

PeterBowman, 2016-12-01:
@jgvictores a [cherry pick](https://git-scm.com/docs/git-cherry-pick) is basically a copy-paste of one or more commits from one branch to another (your IDE of choice should have such a GUI option). Since `master` branch is currently broken for YARP >2.3.66.1 installations, I referred to applying 1e45ba3 there, too.

jgvictores, 2016-12-03:
Mmmm ended up doing a `git cherry-pick --abort`, wasn't very fast-forward... ended up doing a brute-force update of master to develop, sorry! :smile:

commit 1e45ba39 (2016-11-25): [hotfix] Replace 'threadrate' parameter (removed in YARP 2.3.66.1) with 'period'

Closes #84.
commit efb89621 (2016-11-25): Replace 'threadrate' parameter (removed in YARP 2.3.66.1) with 'period'

Closes #84.