roboticslab-uc3m/yarp-devices #85: Create CLI utility for sending data through CAN bus

PeterBowman, 2016-12-03:
We already have `dumpCanBus`, a handy program for collecting transmitted data, and `checkCanBus`, some sort of maintenance script that checks the availability of one or several CAN devices. As there is a growing need for communicating with low-level devices at the ASIBOT team, we could benefit from another utility that would simply send CAN messages issued at the command line, let's say `sendCanBus`. The idea originated from a set of SocketCAN utilities for Linux: [can-utils](https://github.com/linux-can/can-utils).

jgvictores, 2016-12-03:
Another idea would be to make a very simple python class, the use through `ipython` could be good (not having to copy+paste or use readline). But yes, a very simple CLI utility is definitely a great idea and can be implemented very quickly.

PeterBowman, 2016-12-06:
I have no experience with IPython, but it looks interesting. This would require binding from C code to a Python class, am I right? I was considering a YARP `RpcServer` to intercept custom vocabs (=CAN messages) and grant a basic shell for easier interaction. It might be overkill for our needs, though.

jgvictores, 2016-12-06:
> I have no experience with IPython, but it looks interesting. This would require binding from C code to a Python class, am I right?

Yes, possibly. As an interface, I was thinking of something like [this](https://github.com/robotics/open_abb/wiki/Python-Control). Maybe just making an SWIG `.i` is enough, it's a matter of selecting which class and making a `send()` member available if it doesn't exist yet.

> I was considering a YARP `RpcServer` to intercept custom vocabs (=CAN messages) and grant a basic shell for easier interaction. It might be overkill for our needs, though. 

This is another possibility. People used to use [yarp::os::RFModule::attachTerminal](http://www.yarp.it/classyarp_1_1os_1_1RFModule.html#a32998fe2ac83a4a9f7e4a52694c36c58) to provide a CLI embedded in the program attached to a port's methods. However, I kind of find the Python solution much more reusable (SWIG also provides Lua, etc) for a number of possible future use cases.

> Edit: a simple callback port should be enough (I was thinking of ASIBOT's `CartesianServer` with this regard) as bidirectional communication could be emulated with an instance of `dumpCanBus` running on another terminal.

I'm not a big fan of running two programs (dumpCanBus and another one) using the same HicoCan (I'd have to review what [the manual](https://github.com/roboticslab-uc3m/teo-body/blob/develop/doc/hcanpci_linux_v1535en_edi1878.pdf)) says. That's why I'd probably go at the [CanBusControlboard](https://github.com/roboticslab-uc3m/teo-body/tree/develop/libraries/BodyYarp/CanBusControlboard) (includes read) level or similar.

PeterBowman, 2016-12-26:
I'd give a try to Python's native access to low-level IO with functions like `os.open()`, `os.read()`, etc. ([2.x](https://docs.python.org/2/library/os.html?highlight=os.read#file-descriptor-operations), [3.x](https://docs.python.org/3/library/os.html?highlight=os.fdopen#file-descriptor-operations)). [python-can](https://python-can.readthedocs.org/en/stable/) looks promising, too, although it may not support HicoCan interfaces.

PeterBowman, 2017-02-07:
A nice example of such implementation can be found in asrob-uc3m/robotDevastation-robots@3385429.

jgvictores, 2017-02-16:
`teo-main` now also has some similar bindings (see [here](https://github.com/roboticslab-uc3m/teo-main/tree/54b5cc7cd9764ff0245e6d2866f486e5b2aa08d3/bindings)).

jgvictores, 2017-04-02:
Updated comment from `teo-main` to `kinematics-dynamics`

PeterBowman, 2017-12-24:
> I was considering a YARP RpcServer to intercept custom vocabs (=CAN messages) and grant a basic shell for easier interaction.

Now that YARP interfaces are making their way through (#122), I'd consider this option once again: a client/server device pair to talk remotely with the CAN master via RPC. No custom shell will be needed (new apps, ipython with bindings), just use the `yarp rpc` utility.

PeterBowman, 2017-12-24:
To expand on my last comment, the CAN bus logic (master + nodes, i.e. CanBusControlboard + TechnosoftIpos, CuiAbsolute and so on) could live in a remote PC, quite the contrary as it is stated in our install guide: [yarp-devices-install.md#Specific dependencies](https://github.com/roboticslab-uc3m/yarp-devices/blob/8be3065e6529536826e22aac41ff2e163a41d157/doc/yarp-devices-install.md#specific-dependencies) (check the *hcanpci* row). The only thing that requires to run closely tied to the robot CPU and consume its resources is the `yarp::dev::ICanBus` implementation (currently: CanBusHico). We'd need the `remote_controlboard` equivalent and the server pair, thus CanBusHico (or whatever) would act as the subdevice of the latter.

PeterBowman, 2017-12-25:
I've opened #160 to cover the RPC thing, which is focused on working with distributed archs (send CAN data over the network) instead of talking to the CAN master locally.

Now, this issue is far more straightforward if we follow your proposal, @jgvictores. Let's enable SWIG bindings for the `yarp::dev::ICanBus` interface and thus gain access to the upcoming HicoCan (#122), SocketCan and PeakCan (#146) implementations, which will integrate the `canWrite` and `canRead` methods for sending and receiving data, respectively. The usual boilerplate (check YARP network, configure and open device, view interfaces, etc.) could be wrapped in a convenient Python script to make this truly a one-line call via terminal, such as `canUtility send --device /dev/can0 --type HicoCan --id 605 --data 23 81 20 00 05`.

BTW there is no support for CAN interfaces in upstream `yarp.i` yet.

jgvictores, 2017-12-25:
Perfect!

PeterBowman, 2017-12-30:
Yet another approach, after completing the new Peak CAN device (**not** SocketCan, see #146). The manufacturer already provides API bindings and useful sample apps, including GUIs that enable read/write operations:

https://www.peak-system.com/PCAN-miniPCI.205.0.html?&L=1

Look for *PCAN-Basic API* in the *Downloads* tab.

jgvictores, 2018-04-20:
Also, think about: https://github.com/google/python-fire

PeterBowman, 2019-10-04:
On a second thought, such CLI/whatever utility makes little sense whenever we want to run that alongside the CAN motor control application. Besides, CanBusControlboard is required to instantiate its **raw** subdevices. Best option is: make CanBusControlboard expose CANopen commands via RPC port: https://github.com/roboticslab-uc3m/yarp-devices/issues/160#issuecomment-538381625.

The Python thing could be summed up as invalid, too. I found a nice library that might do as a YARP-free replacement, but this is not our use case: https://github.com/christiansandberg/canopen.

closed 2022-06-04 by PeterBowman