roboticslab-uc3m/yarp-devices #93: [CuiAbsoluteEncoder] New wrong values found... and 109 missing to fix!!

rsantos88, 2017-02-23:
Después de realizar diversas pruebas, se ha encontrado que:
* Al arrancar el launchLocomotion, los encoder 114 (frontal de la cintura) y el 105 (sagital de la pierna derecha) dan aleatoriamente un valor, que puede ser entorno a los 67º (ej: 67.557129) , -67º (ej: -67.398941) y su valor correcto. A partir de este valor (sea correcto o incorrecto), empieza a variar, incrementandose o decrementandose según movamos la articulación. Con esto quiero dejar claro de que no se trata de un valor basura que se muestra solo al principio, como si ocurría en la issue  #80.
* Para aislar el problema, se ha conectado directamente el encoder 114 al MPLAB y se ha utilizado el debugger para comprobar el valor que da el encoder. De 5 pruebas realizadas, 4 ha dado el valor 3.251318 y a la 5ª, el valor era de 295.6942. Tras apagar y encender el encoder encoder (conectar y desconectar del MPLAB ICD 2), el valor nuevamente era de 3.251318.
* Una vez realizas estas pruebas, se ha realizado una puesta a cero de dicho encoder y se ha reprogramado nuevamente el firmware. Tras volver a arrancar el launchLocomotion, vuelve a producirse las mismas anomalías.
* Falta arreglar el encoder 109

rsantos88, 2017-02-24:
Añado el encoder 103 (frontal rodilla derecha) con exactamente el mismo problema comentado en el primer punto.

jgvictores, 2017-02-25:
@rsantos88 En tu opinión, esto es software?
Se puede aislar porblemas hardware/software en esta issue?

rsantos88, 2017-02-26:
En mi opinión, es un problema de hardware.

rsantos88, 2017-02-27:
Issue migrada a [#13](https://github.com/roboticslab-uc3m/teo-hardware-manual/issues/13) y [#12](https://github.com/roboticslab-uc3m/teo-hardware-manual/issues/12)

rsantos88, 2017-03-10:
Se reabre esta issue, ya que no está claro totalmente la naturaleza del problema (hardware o software).
Hasta ahora, con las pruebas que se han realizado, se comprueba que, una vez vuelto a un commit anterior a la utilización de las librerías compartidas  (pull request #79) , el valor erróneo de los encoders sigue produciéndose... La única diferencia con respecto a antes ha sido el cambio de versión de YARP ( #89). Se seguirá analizando el problema...

rsantos88, 2017-03-14:
Para prescindir del LaunchManipulation y launchLocomotion, se ha mejorado la aplicación de `testCuiAbsolute` para que ahora devuelva el valor del encoder absoluto [(ver aquí)](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/tests/testCuiAbsolute/testCuiAbsolute.cpp#L221-L227), además de testear las diferentes formas de publicar el PIC. El resultado es exactamente el mismo y se siguen mostrando los mismos valores de forma aleatoria. En las pruebas, se ha quitado el driver correspondiente a la rodilla y se ha sacado también el valor a través del debugger, conectando directamente el programador al encoder, siendo también este erroneo en algunos casos.
Resultado de las pruebas hechas:
Valor correcto: **testCuiAbsolute** (-2.53) **Debugger** (356.4698)
Valor erroneo: **testCuiAbsolute** (-70)     **Debugger** (289.9824)

Nota: entre una prueba y otra, únicamente se ha conectado y desconectado el PIC del programador MPLAB para apagarlo y alimentarlo de nuevo.

rsantos88, 2017-04-17:
Tras diferentes pruebas realizadas, se ha procedido a cambiar los encoders de las piernas que fallaban por unos completamente nuevos. Se han reprogramado todos los encoders en su posición home, y el problema finalmente se ha sido solucionado.

Además de esto, los drivers de la pierna izquierda se reiniciaban aleatoriamente, produciéndose la perdida del control de los motores. Finalmente, el problema era este:

[image]

Tras realizar las pruebas esta misma mañana, las piernas de Teo vuelven a funcionar correctamente :)

Cierro issue!!

rsantos88, 2017-04-18:
Abro de nuevo esta issue para comentar que los encoders del tobillo de la pierna izquierda (111 y 112) empiezan a dar valores erroneos al arrancar de forma aleatoria... (valores en torno a 67 o -67 grados) 
En estos encoders no se habían detectado problemas hasta ahora...  :sweat:

rsantos88, 2017-04-19:
Después de hacer diferentes pruebas también en los brazos, los encoders 115 y 116 devuelven de forma aleatoria los valores erroneos comentados:
```bash
[debug] DeviceDriverImpl.cpp:217 open(): ---> First CUI message has been reached 
Absolute encoder value -----> 67.574692
[INFO]created device <CuiAbsolute>. See C++ class teo::CuiAbsolute for documentation.
[debug] DeviceDriverImpl.cpp:217 open(): ---> First CUI message has been reached 
Absolute encoder value -----> 67.398941
[INFO]created device <CuiAbsolute>. See C++ class teo::CuiAbsolute for documentation.
[debug] DeviceDriverImpl.cpp:217 open(): ---> First CUI message has been reached 
Absolute encoder value -----> -0.333923
[INFO]created device <CuiAbsolute>. See C++ class teo::CuiAbsolute for documentation.
[debug] DeviceDriverImpl.cpp:217 open(): ---> First CUI message has been reached 
Absolute encoder value -----> 0.000000
[INFO]created device <CuiAbsolute>. See C++ class teo::CuiAbsolute for documentation.
[debug] DeviceDriverImpl.cpp:217 open(): ---> First CUI message has been reached 
Absolute encoder value -----> -0.158173
[INFO]created device <CuiAbsolute>. See C++ class teo::CuiAbsolute for documentation.
[debug] DeviceDriverImpl.cpp:217 open(): ---> First CUI message has been reached 
Absolute encoder value -----> 0.000000

```

rsantos88, 2017-05-16:
**Nota:** Se han realizado las pruebas al robot y se ha detectado en el encoder del tobillo frontal izquierdo el valor `-67.662567` (encoder que ha sido cambiado recientemente por @smcdiaz )

rsantos88, 2018-01-24:
resucito un poco esta issue para comentar que los encoders absolutos de los brazos que fallan actualmente son: 115, 116 y 118

jgvictores, 2018-01-27:
This issue was moved to roboticslab-uc3m/teo-hardware-issues#10

closed 2018-01-27 by jgvictores