roboticslab-uc3m/yarp-devices #99: Inminent repo name change

jgvictores, 2017-02-27:
On Friday March 3rd 2017 evening, this repo will be renamed: yarp-devices

Cheers!

jgvictores, 2017-02-27:
PS: This is the main reason why we've been migrating teo hardware issues to https://github.com/roboticslab-uc3m/teo-hardware-manual

PeterBowman, 2017-03-04:
Renaming done. All references to the old location should redirect now to `yarp-plugins`, including fetch/push operations. In order to update your remotes, see [Changing a remote's URL](https://help.github.com/articles/changing-a-remote-s-url/). I'd leave this issue open until every single reference in the organization has been updated accordingly: [search](https://github.com/search?q=org%3Aroboticslab-uc3m+teo-body&type=Code).

jgvictores, 2017-03-04:
Thanks for doing that! 

Btw, edited a typo in your post, from yarp-plugins to the final yarp-devices. 😀

jgvictores, 2017-03-05:
cc https://github.com/roboticslab-uc3m/QA/issues/2

PeterBowman, 2017-03-05:
I've tracked down all occurrences of `teo-body` in the repos at our org, only these are left:

* [waiter/programs/waiterExecutionCore/CMakeLists.txt.user#L62](https://github.com/roboticslab-uc3m/waiter/blob/beea9ad/programs/waiterExecutionCore/CMakeLists.txt.user#L62)
* follow-me/doc/fig/editable/follow-me-app.graphml:
  * [line 389](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c/doc/fig/editable/follow-me-app.graphml#L389)
  * [line 405](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c/doc/fig/editable/follow-me-app.graphml#L405)

The wiki has been updated, too. Leftovers:
* [Procedimiento Revisión Drivers](http://robots.uc3m.es/index.php/Procedimiento_Revisi%C3%B3n_Drivers)
* TEO Network information:
  * [#manipulation (robot front right PC)](http://robots.uc3m.es/index.php/TEO_Network_information#manipulation_.28robot_front_right_PC.29)
  * [#locomotion (robot front left PC)](http://robots.uc3m.es/index.php/TEO_Network_information#locomotion_.28robot_front_left_PC.29)

jgvictores, 2017-12-02:
Changing priority to high, we should check and replace these occurrences.

PeterBowman, 2018-01-05:
According to my updated comment (https://github.com/roboticslab-uc3m/yarp-devices/issues/99#issuecomment-284267709), the only thing left to do here is tracked at roboticslab-uc3m/follow-me#36.

jgvictores, 2018-01-06:
Can close, because tracked elsewhere, right?

closed 2018-01-06 by PeterBowman