roboticslab-uc3m/yarp-devices #103: Yarpmotorgui control modes

jgvictores, 2017-03-08:
_From @rsantos88 on February 27, 2017 14:5_

Implement all the necessary functions to do Yarpmotorgui fully functional

_Copied from original issue: roboticslab-uc3m/teo-software-manual#2_

jgvictores, 2017-03-08:
_From @rsantos88 on February 28, 2017 11:36_

Fixed error `[ERROR] iint->getInteractionlMode failed` implementing IInteractionMode.cpp with empty functions (https://github.com/rsantos88/teo-body/commit/420d59dfa623e9c64ca2bbc99a1b71f91e8b8689).
To do: complete implementation!

jgvictores, 2017-03-08:
_From @rsantos88 on February 28, 2017 11:47_

Implement functions are missing... [here!](https://github.com/rsantos88/teo-body/blob/develop/libraries/BodyYarp/TechnosoftIpos/ITorqueControlRawImpl.cpp)

jgvictores, 2017-03-08:
_From @rsantos88 on March 7, 2017 12:27_

After adding the update of the new [libraries](https://github.com/rsantos88/teo-body/tree/issue%2390/libraries/BodyYarp/TechnosoftIpos), when I try to change to Velocity mode, it shows this warning message:
```bash 
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 180).
[warning] ICanBusSharerImpl.cpp:252 interpretMessage(): pt movement ended. canId: 6 (via 280).
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "0" not specified in manual, may be in Fault or not enabled yet. canId(9).
```
The interface changes to blue color but I can't move the velocity bar...

[image]

jgvictores, 2017-03-08:
_From @PeterBowman on March 7, 2017 13:31_

~~Yarpmotorgui calls `IVelocityControl2::setRefAcceleration` [here](https://github.com/robotology/yarp/blob/a0af248bfc2b454772b617aa18ff85e17cd50245/src/yarpmotorgui/partitem.cpp#L947), but I don't see this method implemented in your code: [a](https://github.com/rsantos88/teo-body/blob/issue%2390/libraries/BodyYarp/CanBusControlboard/IVelocityControl2.cpp), [b](https://github.com/rsantos88/teo-body/blob/issue%2390/libraries/BodyYarp/TechnosoftIpos/IVelocityControl2RawImpl.cpp) + http://www.yarp.it/classyarp_1_1dev_1_1IVelocityControl2.html + http://www.yarp.it/classyarp_1_1dev_1_1IVelocityControl2Raw.html.~~

**Edit:** sorry, I didn't realize this same method is implemented in the `IPositionControl2` interface (wouldn't compile otherwise). I'll keep looking.

jgvictores, 2017-03-08:
_From @rsantos88 on March 7, 2017 14:21_

Trying to change to torque mode, the interface doesn't turn to other color and it shows this error in launchLocomotion terminal:
```bash
[warning] ICanBusSharerImpl.cpp:366 interpretMessage(): 	-Mode "251" not specified in manual, may be in Fault or not enabled yet. canId(6).
```

jgvictores, 2017-03-08:
_From @rsantos88 on March 7, 2017 15:0_

Fixed torque mode issue with this [commit](https://github.com/rsantos88/teo-body/commit/cf81ab7305451ae3114af9528e173c2599e56bab)

jgvictores, 2017-03-08:
_From @rsantos88 on March 7, 2017 15:58_

Returning with the problem to change to velocity mode, I've checked the status of the joints and I can see that the velocity mode is changed  successfully:
```bash
yarp rpc /teo/rightLeg/rpc:i
get icmd cmds
Response: [is] [cmds] ([pos] [pos] [pos] [vel] [pos] [pos]) [tsta] 2440 1488901823.83269 [ok]
```

jgvictores, 2017-03-08:
Mini-comment: this type of issues belong to https://github.com/roboticslab-uc3m/yarp-devices :smirk:

rsantos88, 2017-04-17:
All the changes made in the libraries are integrated now into `develop` branch of [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices)

jgvictores, 2017-04-18:
Great! Does that mean we can close this issue?

rsantos88, 2017-04-18:
There are things that do not work correctly yet... like `velocity` mode:

> The interface changes to blue color but I can't move the velocity bar...
[image]

PeterBowman, 2017-07-12:
I'm unable to change to velocity mode at YARP v2.3.68.1 and latest `yarp-devices` develop. That is, yarpmotorgui reports a successful mode change, but the background doesn't switch to blue and I still see both position and velocity sliders.

PeterBowman, 2019-12-30:
Tested again on YARP 3.3 and https://github.com/roboticslab-uc3m/yarp-devices/pull/229. Note the internals of CanBusControlboard and raw subdevices varied wildly. I checked this against yarpmotorgui and I can say it is mostly functional, but not perfect. Velocity mode can be switched successfully, but commands would not be accepted unless this mode is enabled in the *View* dropdown menu. There is room for improvements, e.g. we can't switch to idle mode and back to e.g. position control apparently due to the lack of a calibration routine. On the other hand, per-joint, per-part and multiple-part homing works like a charm.

Since basic functionality is achieved, I'm finally closing this.

closed 2019-12-30 by PeterBowman