roboticslab-uc3m/yarp-devices #104: [launchManipulation & launchLocomotion] Segmentation fault in the end

rsantos88, 2017-03-10:
Se ha observado últimamente que al terminar de ejecutarse la aplicación, esta cierra con un Segmentation fault. Copio un ejemplo de ello:
```bash
[info] ICanBusSharerImpl.cpp:96 stopPublishingMessages(): Send: "stopPublishingMessages" to Cui Absolute Encoder. ICanBusSharerImpl.cpp
[warning] DeviceDriverImpl.cpp:366 close(): Resending stop message to Cui Absolute PIC (ID: 111)
[info] ICanBusSharerImpl.cpp:96 stopPublishingMessages(): Send: "stopPublishingMessages" to Cui Absolute Encoder. ICanBusSharerImpl.cpp
[warning] DeviceDriverImpl.cpp:366 close(): Resending stop message to Cui Absolute PIC (ID: 111)
[info] ICanBusSharerImpl.cpp:96 stopPublishingMessages(): Send: "stopPublishingMessages" to Cui Absolute Encoder. ICanBusSharerImpl.cpp
[success] DeviceDriverImpl.cpp:350 close(): Time out passed and CuiAbsolute ID (111) was stopped successfully
[info] DeviceDriverImpl.cpp:336 close(): Stopping Cui Absolute PIC (ID: 112)
[info] ICanBusSharerImpl.cpp:96 stopPublishingMessages(): Send: "stopPublishingMessages" to Cui Absolute Encoder. ICanBusSharerImpl.cpp
[success] DeviceDriverImpl.cpp:350 close(): Time out passed and CuiAbsolute ID (112) was stopped successfully
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:58 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:28 close(): 
[info] DeviceDriverImpl.cpp:381 close(): End.
[INFO]RFModule finished.
Segmentation fault
```
Si además, probamos a ejecutarlo con el comando `--help`, nos encontramos con que lo muestra al final:
```bash
TwoCanBusThreeWrappers options:
	--help (this help)	--from [file.ini]	--context [path]	--homePoss	--externalEncoderWait [s]

Note: if the Absolute Encoder doesn't respond, use --externalEncoderWait [seconds] parameter for using default relative encoder position
(devCan0 (device CanBusControlboard) (canDevice "/dev/can0") (types TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute) (ids 14 6 5 4 3 2 1 114 106 105 104 103 102 101) (maxs 50 90 20 45 80 25 25 0 0 0 0 0 0 0) (mins -10 -90 -20 -30 -5 -25 -25 0 0 0 0 0 0 0) (maxVels 1000 1000 1000 1000 1000 1000 1000 0 0 0 0 0 0 0) (minVels 0 0 0 0 0 0 0 0 0 0 0 0 0 0) (trs 480 400 523.2 192 235.2 270.4 235.2 -1 -1 1 1 1 1 1) (refAccelerations 10.0 10.0 10.0 10.0 10.0 10.0 10.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) (refSpeeds 10.0 10.0 10.0 10.0 10.0 10.0 10.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0)) (from launchLocomotion.ini) (devCan1 (device CanBusControlboard) (canDevice "/dev/can1") (types TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute) (ids 13 7 8 9 10 11 12 113 107 108 109 110 111 112) (maxs 45 90 20 45 80 25 25 0 0 0 0 0 0 0) (mins -45 -90 -20 -30 -5 -25 -25 0 0 0 0 0 0 0) (maxVels 1000 1000 1000 1000 1000 1000 1000 0 0 0 0 0 0 0) (minVels 0 0 0 0 0 0 0 0 0 0 0 0 0 0) (trs 160 400 523.2 192 235.2 270.4 235.2 1 1 -1 -1 -1 -1 -1) (refAccelerations 10.0 10.0 10.0 10.0 10.0 10.0 10.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) (refSpeeds 10.0 10.0 10.0 10.0 10.0 10.0 10.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0)) (help) (wrapper0 (device controlboardwrapper2) (name "/teo/rightLeg") (period 60) (joints 6) (networks (devCan0)) (devCan0 0 5 1 6)) (wrapper1 (device controlboardwrapper2) (name "/teo/leftLeg") (period 60) (joints 6) (networks (devCan1)) (devCan1 0 5 1 6)) (wrapper2 (device controlboardwrapper2) (name "/teo/trunk") (period 60) (joints 2) (networks (devCan0 devCan1)) (devCan0 1 1 0 0) (devCan1 0 0 0 0))
[INFO]RFModule failed to open.
Segmentation fault
```
Además, cuando probamos a ejecutar otra aplicación como `dumpCanBus` y la cerramos, ocurre exactamente lo mismo:
```bash
||| clearing context
||| adding context [dumpCanBus]
||| configuring
||| default config file specified as dumpCanBus.ini
||| checking [/home/teo/repos/yarp-devices/build/bin/dumpCanBus.ini] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/ICUBcontrib/robots/default] (robot yarp.d)
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/robots/default] (robot yarp.d)
||| checking [/home/teo/.config/yarp/contexts/dumpCanBus] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/dumpCanBus] (context YARP_DATA_HOME)
||| checking [/etc/yarp/contexts/dumpCanBus] (context YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/contexts/dumpCanBus] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/dumpCanBus] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/ICUBcontrib/contexts/dumpCanBus] (context yarp.d)
||| checking [/usr/local/share/iCub/contexts/dumpCanBus] (context yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/dumpCanBus] (context yarp.d)
||| found /usr/local/share/roboticslab-yarp-devices/contexts/dumpCanBus
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/dumpCanBus/dumpCanBus.ini] (context)
||| found /usr/local/share/roboticslab-yarp-devices/contexts/dumpCanBus/dumpCanBus.ini
[info] main.cpp:57 main(): Checking for yarp network...
[success] main.cpp:64 main(): Found yarp network.
[debug] DumpCanBus.cpp:23 configure(): (from dumpCanBus.ini) (canBitrate 8) (device CanBusHico) (canDevice "/dev/can1")
[success] DeviceDriverImpl.cpp:20 open(): Opened CAN device of path: /dev/can1
[success] DeviceDriverImpl.cpp:30 open(): Bitrate set on CAN device: /dev/can1
[success] DeviceDriverImpl.cpp:40 open(): IOC_START ok on CAN device: /dev/can1
[INFO]created device <CanBusHico>. See C++ class teo::CanBusHico for documentation.
[info] ThreadImpl.cpp:10 run(): Started DumpCanBus reading thread run.
^C[INFO][try 1 of 3] Trying to shut down.
[INFO]RFModule closing.
[info] ThreadImpl.cpp:39 run(): Stopping DumpCanBus reading thread run.
[INFO]RFModule finished.
Segmentation fault
```
Supongo que este problema ha surgido desde que se han empezado a utilizar librerías compartidas...
¿Qué opinas @jgvictores ?

jgvictores, 2017-03-11:
Puedes copiar+pegar los outputs de:
```bash
yarpdev --list
```

Y
```bash
yarp plugin --all
```
Please?

rsantos88, 2017-03-14:
Realizado en el PC-locomotion:

`yarpdev --list`
```bash
Here are devices listed for your system:
Device "test_grabber", C++ class yarp::dev::TestFrameGrabber, wrapped by "grabber"
Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
Device "sound_grabber", C++ class yarp::dev::ServerSoundGrabber, is a network wrapper.
Device "pipe", C++ class yarp::dev::DevicePipe, has no network wrapper
Device "group", C++ class yarp::dev::DeviceGroup, has no network wrapper
Device "remote_controlboard", C++ class yarp::dev::RemoteControlBoard, wrapped by "controlboardwrapper2"
Device "inertial", C++ class yarp::dev::ServerInertial, is a network wrapper.
Device "controlboardwrapper2", C++ class yarp::dev::ControlBoardWrapper2, is a network wrapper.
Device "batteryClient", C++ class batteryClient, has no network wrapper
Device "analogsensorclient", C++ class AnalogSensorClient, has no network wrapper
Device "analogServer", C++ class yarp::dev::AnalogWrapper, is a network wrapper.
Device "virtualAnalogServer", C++ class yarp::dev::VirtualAnalogWrapper, is a network wrapper.
Device "batteryWrapper", C++ class yarp::dev::BatteryWrapper, is a network wrapper.
Device "Rangefinder2DWrapper", C++ class yarp::dev::Rangefinder2Dwrapper, is a network wrapper.
Device "RGBDSensorWrapper", C++ class yarp::dev::RGBDSensorWrapper, is a network wrapper.
Device "RGBDSensorClient", C++ class yarp::dev::RGBDSensorClient, is a network wrapper.
Device "controlboardremapper", C++ class yarp::dev::ControlBoardRemapper, wrapped by "controlboardwrapper2"
Device "remotecontrolboardremapper", C++ class yarp::dev::RemoteControlBoardRemapper, wrapped by "controlboardwrapper2"
Device "navigation2DClient", C++ class navigation2DClient, has no network wrapper
Device "clientcontrolboard", C++ class yarp::dev::ClientControlBoard, wrapped by "controlboard"
Device "controlboard", C++ class yarp::dev::ServerControlBoard, is a network wrapper.
Device "CanBusControlboard", available on request (found in /usr/local/lib/yarp/CanBusControlboard.so library).
Device "CanBusHico", available on request (found in /usr/local/lib/yarp/CanBusHico.so library).
Device "CuiAbsolute", available on request (found in /usr/local/lib/yarp/CuiAbsolute.so library).
Device "FakeJoint", available on request (found in /usr/local/lib/yarp/FakeJoint.so library).
Device "Jr3", available on request (found in /usr/local/lib/yarp/Jr3.so library), wrapped by "analogServer".
Device "LacqueyFetch", available on request (found in /usr/local/lib/yarp/LacqueyFetch.so library).
Device "PlaybackThread", available on request (found in /usr/local/lib/yarp/PlaybackThread.so library).
Device "TechnosoftIpos", available on request (found in /usr/local/lib/yarp/TechnosoftIpos.so library).
Device "TextilesHand", available on request (found in /usr/local/lib/yarp/TextilesHand.so library).
Device "xsensmtx", available on request (found in /usr/local/lib/iCub/xsensmtx.so library).
```
`yarp plugin --all`
```bash
Runtime plugins found:

device CanBusControlboard
  * ini file:       /usr/local/share/yarp/plugins/CanBusControlboard.ini
  * config:         CanBusControlboard (type device) (name CanBusControlboard) (library CanBusControlboard) (part CanBusControlboard) (code CanBusControlboard)
  * library:        /usr/local/lib/yarp/CanBusControlboard.so
  * system version: 5
  * class name:     teo::CanBusControlboard
  * base class:     yarp::dev::DeviceDriver

device CanBusHico
  * ini file:       /usr/local/share/yarp/plugins/CanBusHico.ini
  * config:         CanBusHico (type device) (name CanBusHico) (library CanBusHico) (part CanBusHico) (code CanBusHico)
  * library:        /usr/local/lib/yarp/CanBusHico.so
  * system version: 5
  * class name:     teo::CanBusHico
  * base class:     yarp::dev::DeviceDriver

device CuiAbsolute
  * ini file:       /usr/local/share/yarp/plugins/CuiAbsolute.ini
  * config:         CuiAbsolute (type device) (name CuiAbsolute) (library CuiAbsolute) (part CuiAbsolute) (code CuiAbsolute)
  * library:        /usr/local/lib/yarp/CuiAbsolute.so
  * system version: 5
  * class name:     teo::CuiAbsolute
  * base class:     yarp::dev::DeviceDriver

device FakeJoint
  * ini file:       /usr/local/share/yarp/plugins/FakeJoint.ini
  * config:         FakeJoint (type device) (name FakeJoint) (library FakeJoint) (part FakeJoint) (code FakeJoint)
[... 80 lines cut ...]
carrier tcpros
  * ini file:       /usr/local/share/yarp/plugins/ros.ini
  * config:         tcpros (type carrier) (name tcpros) (library yarp_tcpros) (part tcpros_carrier) (code (any) (any) "\0\0" (any) (any) "\0\0")
    Cannot find or load shared library

carrier rossrv
  * ini file:       /usr/local/share/yarp/plugins/ros.ini
  * config:         rossrv (type carrier) (name rossrv) (library yarp_tcpros) (part rossrv_carrier) (code (any) (any) "\0\0" (any) (any) "\0\0")
    Cannot find or load shared library

carrier xmlrpc
  * ini file:       /usr/local/share/yarp/plugins/xmlrpc.ini
  * config:         xmlrpc (type carrier) (name xmlrpc) (library yarp_xmlrpc) (code "POST /")
error while opening /usr/local/lib/yarp/yarp_xmlrpc.so:
  /usr/local/lib/yarp/yarp_xmlrpc.so: undefined symbol: xmlrpc
    Cannot find or load shared library

device xsensmtx
  * ini file:       /usr/local/share/iCub/plugins/xsensmtx.ini
  * config:         xsensmtx (type device) (name xsensmtx) (library xsensmtx)
  * library:        /usr/local/lib/iCub/xsensmtx.so
  * system version: 5
  * class name:     yarp::dev::XSensMTx
  * base class:     yarp::dev::DeviceDriver
```

jgvictores, 2017-03-14:
¿Ningún segfault entonces?

Lo siguiente en lo que pensaría sería en que se esté llamando `dd.close()` varias o ninguna vez. Podemos investigarlo en `dumpCanBus`, que es un programa sencillo.

jgvictores, 2017-03-14:
Tb: `gdb`.

rsantos88, 2017-03-30:
**Nota**: con la versión `YARP version 2.3.68+193-20170216.12+git0f7a8cb` no se produce ningún segfault al final de la ejecución de nuestros programas.

rsantos88, 2017-04-05:
**Nota**:  Al ejecutar los programas con la misma versión de YARP en un Ubuntu 15.10 no hay ni rastro de segfault...

rsantos88, 2017-12-18:
Este problema se resolverá cuando se realice una correcta actualización de la versión de YARP. 
Queda bloqueada por la issue: https://github.com/roboticslab-uc3m/installation-guides/issues/24 que a su vez depende la issue https://github.com/roboticslab-uc3m/yarp-devices/issues/118

rsantos88, 2018-02-01:
Versión de YARP actualizada a la 2.3.70.2 gracias a la resolución de [esta](https://github.com/roboticslab-uc3m/teo-main/issues/26) issue.
Problema de `segmentation fault` resuelto :)
Cierro issue!

closed 2018-02-01 by rsantos88