roboticslab-uc3m/yarp-devices #105: [checkCanBus] Bad Configuration of TechnosoftIpos :(

rsantos88, 2017-03-10:
```bash
||| clearing context
||| adding context [checkCanBus]
||| configuring
||| default config file specified as checkCanBus.ini
||| checking [/home/teo/repos/yarp-devices/build/bin/checkCanBus.ini] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/ICUBcontrib/robots/default] (robot yarp.d)
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/robots/default] (robot yarp.d)
||| checking [/home/teo/.config/yarp/contexts/checkCanBus] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/checkCanBus] (context YARP_DATA_HOME)
||| checking [/etc/yarp/contexts/checkCanBus] (context YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/contexts/checkCanBus] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/checkCanBus] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/ICUBcontrib/contexts/checkCanBus] (context yarp.d)
||| checking [/usr/local/share/iCub/contexts/checkCanBus] (context yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/checkCanBus] (context yarp.d)
||| found /usr/local/share/roboticslab-yarp-devices/contexts/checkCanBus
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/checkCanBus/checkCanBus.ini] (context)
||| found /usr/local/share/roboticslab-yarp-devices/contexts/checkCanBus/checkCanBus.ini
[info] main.cpp:69 main(): Checking for yarp network...
[success] main.cpp:76 main(): Found yarp network.
[debug] CheckCanBus.cpp:53 configure(): (from checkCanBus.ini) (canBitrate 8) (device CanBusHico) (canDevice "/dev/can0")
[success] DeviceDriverImpl.cpp:20 open(): Opened CAN device of path: /dev/can0
[success] DeviceDriverImpl.cpp:30 open(): Bitrate set on CAN device: /dev/can0
[success] DeviceDriverImpl.cpp:40 open(): IOC_START ok on CAN device: /dev/can0
[INFO]created device <CanBusHico>. See C++ class teo::CanBusHico for documentation.
[success] DeviceDriverImpl.cpp:51 open(): Created TechnosoftIpos with canId 24, tr 160.000000, k 0.000000, refAcceleration 5.000000, refSpeed 0.575000, and all local parameters set to 0.
[INFO]created device <TechnosoftIpos>. See C++ class teo::TechnosoftIpos for documentation.
[error] CheckCanBus.cpp:84 configure(): Bad Configuration of TechnosoftIpos :(
```
Probably it hasn't been updated correctly to use shared libraries...

PeterBowman, 2017-03-10:
> Probably it hasn't been updated correctly to use shared libraries...

I'm working at c3171cd (23/02/17) in ASIBOT w/ shared libs, no problem in controlling all joints (`oneCanBusOneWrapper`). However, this line returns `false` at [CheckCanBus.cpp#L67](https://github.com/roboticslab-uc3m/yarp-devices/blob/c3171cd/programs/checkCanBus/CheckCanBus.cpp#L67):

```c++
ok &= canNodeDevice.view( iControlLimitsRaw );
```

PeterBowman, 2017-03-10:
@jgvictores pointed out that this issue might be caused by a upstream bug in the `yarp::dev::IControlLimits2Raw` interface due to inheriting privately from its base class (solved in robotology/yarp#1108). Then, we upgraded the inheritance chain of `TechnosoftIpos` at ec0c452, including the offending class.

Solutions (not tested):
1. Inherit from both `IControlLimits2Raw` and its base class at `TechnosoftIpos.hpp`, that is:
    ```c++
    [...] public yarp::dev::IControlLimitsRaw, public yarp::dev::IControlLimits2Raw, [...]
    ```

2. Change type of `iControlLimitsRaw` pointer to `yarp::dev::IControlLimits2Raw*` in `CheckCanBus.hpp`

rsantos88, 2017-03-14:
Done it! (94947645dd548736fd3089434e1b89d68d60f0a6). 
Thanks @PeterBowman. I'll close this issue!

closed 2017-03-14 by rsantos88