roboticslab-uc3m/yarp-devices #106: ControlBoardWrapper2 not wrapping correctly for joint group commands

jgvictores, 2017-03-13:
As commented by @jmgarciah and @rsantos88 using `YARP version 2.3.68+181-20170203.11+gite122128`, ControlBoardWrapper2 is not wrapping correctly for joint group commands, e.g. Moving head actually moves the arms. Note that individual movePosition commands map correctly.

Relevant links:
- [Our .ini](https://github.com/roboticslab-uc3m/yarp-devices/blob/76b774499bdd986bd28605014fa882c5093fc80a/share/launchManipulation/conf/launchManipulation.ini#L31-L54)
- http://www.yarp.it/classyarp_1_1dev_1_1ControlBoardWrapper.html
- Maybe? https://github.com/robotology/yarp/pull/753 https://github.com/robotology/yarp/pull/988 https://github.com/robotology/yarp/issues/310

Ideas:
- [ ] Update YARP
- [ ] Migrate ini to xml

rsantos88, 2017-03-16:
To begind with this problem, we could migrate `oneCanBusOneWrapper.ini` to XML to check the program runs well and then, we can continue with the migration of `launchManipulation.ini`...

rsantos88, 2017-03-17:
I've created a new branch to test input parameters in XML format (cfc0ff86bf043bb082cbfb62f3629718e5f90662) with oneCanBusOneWrapper but I don't know if it runs correctly, so... please check it!

jgvictores, 2017-03-18:
Updated YARP, same effect on `2.3.68+218-20170316.1+git3050aeb`.

jgvictores, 2017-03-18:
Downgraded to YARP `2.3.66.2`. Created [launchManipulation-fake.ini](https://github.com/roboticslab-uc3m/yarp-devices/blob/a0b3d7a4791f0c585cc4ad5ebb053660b0765369/share/launchManipulation/conf/launchManipulation-fake.ini) to avoid using robot.
- Group commands are not being matched to correct implementation via RPC.
- Also working on [cpp example](https://github.com/roboticslab-uc3m/yarp-devices/blob/improve-exampleRemoteControlboard/example/cpp/exampleRemoteControlboard/ExampleRemoteControlboard.cpp) to assure use of `yarp::dev::IPositionControl2` interface.

jgvictores, 2017-03-18:
Still strange behaviors at `2.3.66`.

Note to self:
The following section of http://www.yarp.it/classyarp_1_1dev_1_1ControlBoardWrapper.html
```
 <!-- Following parameters are meaningful ONLY for yarprobotinterface -->
 <action phase="startup" level="5" type="attach">
     <paramlist name="networks">
         <!-- The param value must match the device name in the corresponding emsX file -->
         <elem name="FirstSetOfJoints">  left_upper_arm_mc </elem>
         <elem name="SecondSetOfJoints"> left_lower_arm_mc </elem>
      </paramlist>
 </action>
 <action phase="shutdown" level="5" type="detach" />
```

jgvictores, 2017-03-18:
Same at `2.3.65` with fake.ini.

jgvictores, 2017-03-18:
Okay, found problem and have working version with `YARP version 2.3.68+181-20170203.11+gite122128` which is a fairly recent version of YARP.

Conclusion: There must be a bug upstream in [these YARP lines](https://github.com/robotology/yarp/blob/3050aeb118e3ca514a6296afa954207ea7cd50a2/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L1515-L1525). We hadn't noticed it because we were still using "Classic Position Control" (`IPositionControl`), not "Position Control 2" (`IPositionControl2`). Note that the IPositionControl2 group move also displayed buggy behavior.

Temporary fix: I went back to [b342044d10194303a2e4b195b4953dfd4946f14b](https://github.com/roboticslab-uc3m/yarp-devices/tree/b342044d10194303a2e4b195b4953dfd4946f14b), about a month ago when we didn't have IPositionControl2 on `manipulation` PC.

Lemma: The show goes on for the demo on Tuesday... Yay! :smiley:

jgvictores, 2017-03-18:
Just recalled how our implementation of IPositionControl2 may be wrong (also due to lack of API documentation, I recall we implemented a mask instead of a map).

rsantos88, 2017-03-22:
In this case... can it be useful  to continue the migration of .ini to XML?
I've observed that ResourceFinder class uses only .ini files. I haven't found any examples using xml configuration files... (see [ResourceFinder Tutorials and Specification](http://www.yarp.it/yarp_resource_finder_tutorials.html))

jgvictores, 2017-03-22:
- Regarding closing this issue: we should re-implement our `IPositionControl2` to use the `vector<int>` as a `map` (not the binary mask we currently implement). <- high priority
- Regarding understanding how XML is used in YARP in this context: either look at some icub robot code, or ask at https://github.com/robotology/QA or https://github.com/robotology/yarp. <- relatively low priority.

rsantos88, 2017-03-22:
Sorry @jgvictores , but I don't know what do you mean with this:
> we should re-implement our IPositionControl2 to use the vector<int> as a map (not the binary mask we currently implement)

I suspect you want to change this [line](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/libraries/BodyYarp/CanBusControlboard/IPositionControl2.cpp#L247) currently used as a mask...isn't it?

jgvictores, 2017-03-22:
Yes, probably the whole loop:
```c++
    for(int j=0; j<nodes.size(); j++)
    {
        if( joints[j] )
        {
            ok &= this->positionMove(j,refs[j]);
        }
   }
```
Should look more like:
```c++
    for(int j=0; j<n_joint; j++)  // Note the use of n_joint here!!
    {
            ok &= this->positionMove(joints[j],refs[j]);
   }
```
Which agrees a bit more with the [description in the API](http://www.yarp.it/classyarp_1_1dev_1_1ControlBoardWrapper.html#a95b2cd7bd93dabd6b0e3fa2b951a4594).

rsantos88, 2017-03-23:
Perfect!! this problem has been fixed now!! (44b6cc353149dbdf998dded12d84901956b2373b)

rsantos88, 2017-03-23:
merged in develop branch (44b6cc353149dbdf998dded12d84901956b2373b)

jgvictores, 2017-03-23:
@rsantos88 Could you check getRefAcceleration ?
I'm not sure if there's still a `nodes.size()` that should be `n_joint` to avoid future segfaults.

jgvictores, 2017-03-23:
Same on getTargetPosition. Best to `ctrl-f` for all `nodes.size()` of `IPositionControl2`.

rsantos88, 2017-03-23:
Se va a proceder a integrar los cambios realizados en las librerías de CanBusControlboard (nuevas declaraciones e implementaciones añadidas: IInteractionMode, IPositionControl2, IVelocityControl2 and IControlMode2) como consecuencia de la [issue#103](https://github.com/roboticslab-uc3m/yarp-devices/issues/103) al fork actual de yarp-devices. Para ello se ha creado una branch [integrated-changes-issue#103](https://github.com/roboticslab-uc3m/yarp-devices/tree/integrated-changes-issue%23103) que apunta a develop como medida de seguridad ante posibles catástrofes  :smile:

rsantos88, 2017-03-23:
New pull request!! (#109)

rsantos88, 2017-03-24:
Con los nuevos cambios aplicados, haciendo uso de `launchLocomotion ` todo funciona correctamente. En cambio, cuando intentamos arrancar `launchManipulation` nos salta el siguiente error al realizar el arranque:

```bash
....
[success] IControlModeRawImpl.cpp:221 getControlModeRaw2(): Sent manufacturer status query. 40 2 10 0 0 0 0 0. canId(20) via(600).
[success] IControlModeRawImpl.cpp:221 getControlModeRaw2(): Sent manufacturer status query. 40 2 10 0 0 0 0 0. canId(26) via(600).
[info] ICanBusSharerImpl.cpp:298 interpretMessage(): Got SDO "modes of operation display" from driver. 4f 61 60 0 1 0 0 0. canId(26) via(580).
[info] ICanBusSharerImpl.cpp:338 interpretMessage(): 	-Profile Position Mode. canId: 26.
[info] ICanBusSharerImpl.cpp:534 interpretMessage(): Got "manufacturer status register" from driver. 43 2 10 0 37 86 1 2. canId(20) via(580).
[info] ICanBusSharerImpl.cpp:538 interpretMessage(): 	-Ready to switch on. canId: 20.
[info] ICanBusSharerImpl.cpp:542 interpretMessage(): 	-Switched on. canId: 20.
[info] ICanBusSharerImpl.cpp:546 interpretMessage(): 	-Operation Enabled. canId: 20.
[info] ICanBusSharerImpl.cpp:554 interpretMessage(): 	-Motor supply voltage is present. canId: 20.
[info] ICanBusSharerImpl.cpp:578 interpretMessage(): 	-Remote: drive parameters may be modified via CAN and the drive will execute the command message. canId: 20.
[info] ICanBusSharerImpl.cpp:586 interpretMessage(): 	-Target reached. canId: 20.
[info] ICanBusSharerImpl.cpp:602 interpretMessage(): 	-No event set or the programmed event has not occurred yet. canId: 20.
[info] ICanBusSharerImpl.cpp:606 interpretMessage(): 	-Axis on. Power stage is enabled. Motor control is performed. canId: 20.
[info] ICanBusSharerImpl.cpp:615 interpretMessage(): 	*Drive/motor initialization performed. canId: 20.
[info] ICanBusSharerImpl.cpp:651 interpretMessage(): 	*Target command reached. canId: 20.
[info] ICanBusSharerImpl.cpp:534 interpretMessage(): Got "manufacturer status register" from driver. 43 2 10 0 37 86 1 2. canId(26) via(580).
[info] ICanBusSharerImpl.cpp:538 interpretMessage(): 	-Ready to switch on. canId: 26.
[info] ICanBusSharerImpl.cpp:542 interpretMessage(): 	-Switched on. canId: 26.
[info] ICanBusSharerImpl.cpp:546 interpretMessage(): 	-Operation Enabled. canId: 26.
[info] ICanBusSharerImpl.cpp:554 interpretMessage(): 	-Motor supply voltage is present. canId: 26.
[info] ICanBusSharerImpl.cpp:578 interpretMessage(): 	-Remote: drive parameters may be modified via CAN and the drive will execute the command message. canId: 26.
[info] ICanBusSharerImpl.cpp:586 interpretMessage(): 	-Target reached. canId: 26.
[info] ICanBusSharerImpl.cpp:602 interpretMessage(): 	-No event set or the programmed event has not occurred yet. canId: 26.
[info] ICanBusSharerImpl.cpp:606 interpretMessage(): 	-Axis on. Power stage is enabled. Motor control is performed. canId: 26.
[info] ICanBusSharerImpl.cpp:615 interpretMessage(): 	*Drive/motor initialization performed. canId: 26.
[info] ICanBusSharerImpl.cpp:651 interpretMessage(): 	*Target command reached. canId: 26.
[success] IControlModeRawImpl.cpp:237 getControlModeRaw3(): Sent Motion Error Register query. 40 0 20 0 0 0 0 0. canId(20) via(600).
[success] IControlModeRawImpl.cpp:237 getControlModeRaw3(): Sent Motion Error Register query. 40 0 20 0 0 0 0 0. canId(26) via(600).
[info] ICanBusSharerImpl.cpp:464 interpretMessage(): Got SDO ack "Motion Error Register" from driver. 4b 0 20 0 0 0 0 0. canId(20) via(580).
[info] ICanBusSharerImpl.cpp:464 interpretMessage(): Got SDO ack "Motion Error Register" from driver. 4b 0 20 0 0 0 0 0. canId(26) via(580).
[success] IControlModeRawImpl.cpp:253 getControlModeRaw4(): Sent Detailed Error Register query. 40 2 20 0 0 0 0 0. canId(20) via(600).
[debug] IInteractionMode.cpp:10 getInteractionMode(): (1)
[info] IInteractionModeRawImpl.cpp:10 getInteractionModeRaw(): (1)
[debug] IInteractionMode.cpp:10 getInteractionMode(): (2)
[info] IInteractionModeRawImpl.cpp:10 getInteractionModeRaw(): (2)
[debug] IInteractionMode.cpp:10 getInteractionMode(): (3)
[info] IInteractionModeRawImpl.cpp:10 getInteractionModeRaw(): (3)
[debug] IInteractionMode.cpp:10 getInteractionMode(): (4)
[info] IInteractionModeRawImpl.cpp:10 getInteractionModeRaw(): (4)
[debug] IInteractionMode.cpp:10 getInteractionMode(): (5)
[info] IInteractionModeRawImpl.cpp:10 getInteractionModeRaw(): (5)
[debug] IInteractionMode.cpp:10 getInteractionMode(): (6)
[info] IInteractionModeRawImpl.cpp:10 getInteractionModeRaw(): (6)
[debug] IInteractionMode.cpp:10 getInteractionMode(): (7)
Segmentation fault

```

jgvictores, 2017-03-24:
On [launchManipulation.ini](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/share/launchManipulation/conf/launchManipulation.ini#L6) we have `LacqueyFetch` on (7), which has no `IInteractionModeRawImpl.cpp`.

On second, I'm also going to commit (via web) a couple of fixes.

jgvictores, 2017-03-24:
Ok, I think fb075586cb52650d55fed4b46a84103320d4a541 is an improvement.

Could you work on inheriting from `IInteractionModeRaw` and implementing `IInteractionModeRawImpl.cpp` on `LacqueyFetch`?

----

PS: We could avoid segfaults by checking the return values on [all these lines](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/libraries/BodyYarp/CanBusControlboard/DeviceDriverImpl.cpp#L89-L98). Could be with the classic format:
```c++
ok &= device->view( .... );
```

PeterBowman, 2017-03-24:
> PS: We could avoid segfaults by checking the return values on all these lines. Could be with the classic format:
> 
> ok &= device->view( .... );

I'd short-circuit the assignment of `ok`:

```c++
ok = ok && device->view( .... );
```

It doesn't offer any real advantage in this case, though (check [`yarp::dev::DeviceDriver::view`](http://www.yarp.it/DeviceDriver_8h_source.html#l00074)), just wanted to comment what striked me when I first saw these lines.

jgvictores, 2017-03-24:
@PeterBowman It's just shorthand... In fact, I like verbose solutions like in [this example](https://github.com/roboticslab-uc3m/yarp-devices/blob/c01640b24fd17c3d473bade3ac26e623426f0798/example/cpp/exampleRemoteControlboard/ExampleRemoteControlboard.cpp#L41-L66), at least on error (not needed for success).

PeterBowman, 2017-03-24:
> @PeterBowman It's just shorthand...

Not really, you still have to evaluate the result of `device->view()`, which is avoided with the logical AND operator `&&` if `ok` resolves to `false`. Bitwise operators (`&`, `|`, etc.) and their assignment counterparts (`&=`, `|=`) do not perform short-circuiting ([SO](http://stackoverflow.com/q/23107162)).

jgvictores, 2017-03-24:
> Not really, you still have to evaluate the result of device->view(), which is avoided with the logical AND operator && if ok resolves to false. Bitwise operators (&, |, etc.) and their assignment counterparts (&=, |=) do not perform short-circuiting (SO).

Yes, I agree `ok = ok && device->view( .... );` is better than `ok &= device->view( .... );` since the it will not evaluate the second operand if the first operand is `false` (in addition to [your ref](http://stackoverflow.com/q/23107162), I like [this one](http://stackoverflow.com/questions/2488406/why-doesnt-c-have-or-for-booleans)). Good one! :+1: 

In my [more verbose proposal](https://github.com/roboticslab-uc3m/yarp-devices/blob/c01640b24fd17c3d473bade3ac26e623426f0798/example/cpp/exampleRemoteControlboard/ExampleRemoteControlboard.cpp#L41-L66), we would exit the function upon failure, so the next checks/views would obviously not be performed.

PeterBowman, 2017-03-24:
+1 for verbosity as in your proposal, it would have spared me some time figuring what failed [here](https://github.com/roboticslab-uc3m/yarp-devices/issues/105#issuecomment-285638558).

rsantos88, 2017-03-29:
I have noticed that many of the implementations are directly integrated into .hpp (for simplicity?)(e.g: [IPositionDirectRaw](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/libraries/BodyYarp/CuiAbsolute/CuiAbsolute.hpp#L197-L219) in CuiAbsolute.hpp). Would it be more correct to implement each function (even if it has empty functions) in a cpp file? (like for example..[IEncodersRawImpl.cpp](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/libraries/BodyYarp/CuiAbsolute/IEncodersRawImpl.cpp)) ... or is it not necessary?

jgvictores, 2017-03-29:
While inline implementations can potentially be more efficient, the only main reason the source code is like that is because it was written in a hurry.

While either way is technically correct, modularity helps code maintainability, so it would be great if you could chop these files up into their corresponding implementation files.

rsantos88, 2017-03-30:
I've done it with [TechnosoftIpos](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/libraries/BodyYarp/TechnosoftIpos/TechnosoftIpos.hpp#L75-L219), [CuiAbsolute](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/libraries/BodyYarp/CuiAbsolute/CuiAbsolute.hpp#L76-L209) and [LacqueyFetch](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/libraries/BodyYarp/LacqueyFetch/LacqueyFetch.hpp#L71-L196) (I think, the most used and important devices). Should I do the same with `TextilesHand` and `FakeJoint`?

jgvictores, 2017-03-30:
That would un factor be really helpful! 🙂

rsantos88, 2017-03-31:
Done it with [FakeJoint](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/libraries/BodyYarp/FakeJoint/FakeJoint.hpp#L70-L196) and [TextilesHand](https://github.com/roboticslab-uc3m/yarp-devices/blob/integrated-changes-issue%23103/libraries/BodyYarp/TextilesHand/TextilesHand.hpp#L81-L207). And it works!  :grin:
So, if you see everything ok, I can merge this temporal branch [integrated-changes-issue#103](https://github.com/roboticslab-uc3m/yarp-devices/tree/integrated-changes-issue%23103) with `develop` branch

jgvictores, 2017-03-31:
Could you please PR?

rsantos88, 2017-03-31:
here it is! (#114)

jgvictores, 2017-04-03:
I've merged #114 ...@rsantos88 Could you install `develop` on TEO and make sure it's okay to close this issue?

Thanks!

rsantos88, 2017-04-06:
Installed and tested in teo-manipulation and teo-locomotion. All it's okay!! It's runs correctly  :thumbsup:
I'll close this issue.

closed 2017-04-06 by rsantos88

jgvictores, 2018-01-28:
To sum up: the bug was we implemented a mask instead of a map.

PeterBowman, 2018-05-17:
For the record (prior to deleting an old branch), this is how an .ini might look like if re-written as XML (@jgvictores reported this doesn't work, though):

```xml
<paramlist name="devCan0">
  <param name="device">FakeControlboard</param>
  <param name="axes">7</param>
</paramlist>

<paramlist name="devCan1">
  <param name="device">FakeControlboard</param>
  <param name="axes">7</param>
</paramlist>
              
<paramlist name="wrapper0">
  <param name="device">controlboardwrapper2</param>
  <param name="name">/teo/rightArm</param>
  <param name="period">60</param>
  <param name="joints">7</param>
  <paramlist name="networks">
    <elem name="devCan0">  0  6  1  7  </elem>
  </paramlist>
</paramlist>

<paramlist name="wrapper1">
  <param name="device">controlboardwrapper2</param>
  <param name="name">/teo/leftArm</param>
  <param name="period">60</param>
  <param name="joints">7</param>
  <paramlist name="networks">
    <elem name="devCan1">  0  6  1  7  </elem>
  </paramlist>
</paramlist>
              
<paramlist name="wrapper2">
  <param name="device">controlboardwrapper2</param>
  <param name="name">/teo/head</param>
  <param name="period">60</param>
  <param name="joints">2</param>
  <paramlist name="networks">
    <elem name="devCan1">  0  0  0  0  </elem>
    <elem name="devCan0"> 1  1  0  0  </elem>
  </paramlist>
</paramlist>
```