roboticslab-uc3m/yarp-devices #111: Run Travis CI scripts on specific versions of YARP

PeterBowman, 2017-03-30:
Several BodyYarp plugins need to run on a Debian 6.0 and a particular revision of YARP ([docs](https://github.com/roboticslab-uc3m/yarp-devices/blob/0c2411/doc/yarp_devices_install_on_debian_6.md)). Travis builds would be more meaningful if this (quite basic) dependency was adjusted to meet a realistic scenario. Furthermore, we could benefit from caching (example: roboticslab-uc3m/project-generator@722ffc4).

PeterBowman, 2017-04-29:
Also, decide whether to keep running tests on Precise or switch to Trusty. Assuming the former, add a `dist: precise` line to the YAML file. Info:

https://blog.travis-ci.com/2017-04-17-precise-EOL

PeterBowman, 2017-05-02:
This issue might block roboticslab-uc3m/QA#17.

jgvictores, 2017-05-06:
> Also, decide whether to keep running tests on Precise or switch to Trusty.

No doubts here, forget `Precise` and move on to `Trusty`.

PeterBowman, 2017-05-06:
I was thinking more about getting as close as possible to the software ecosystem provided by Debian, which may be no longer reproducible if switching to a newer release of Ubuntu. Remember the issues we encountered during compilation of latest YARP with Debian's outdated gcc (#89).

jgvictores, 2017-05-06:
I think we're blocked by https://github.com/roboticslab-uc3m/yarp-devices/issues/118 here then. If hc-driver-4.9 provides good results, we can begin to forget about outdated Distro versions.

PeterBowman, 2017-11-29:
Just noting that Travis builds no longer run on Precise images by default, thus we are currently testing against Trusty.

PeterBowman, 2017-12-12:
I think we need this, regardless of the outcome of distro-related debates. Providing backwards compatibility is fine, but requires downgrading specific software to make sure that version requirements are fulfilled. For instance, how should I know that my code will work on CMake v2.8.9 (or whichever version I set in `cmake_minimum_required(VERSION)`) without compiling CMake itself at that specific version, re-compiling the code and running some tests? Unexpected errors may easily occur, see https://github.com/roboticslab-uc3m/yarp-devices/issues/26#issuecomment-350398178, https://github.com/roboticslab-uc3m/questions-and-answers/issues/39. That's what continuous integration is meant for, compare https://travis-ci.org/robotology/ycm/builds/216344488 (robotology/ycm#114).

PeterBowman, 2018-05-23:
We've moved on to CMake 3.1 and dropped support for Debian 6.0 (check the `old-debian-6.0` tag). I'm prone to abandon this issue, or at least move it to QA since the above considerations could be easily extended to other repositories. We are prone to require CMake 3.5, and Trusty is *de facto* the oldest distro supported. Therefore, working with software releases so close to the minimum requirements make all this pondering rather pointless. Caching is already discussed at https://github.com/roboticslab-uc3m/questions-and-answers/issues/48.

PeterBowman, 2018-06-07:
Trusty has moved on to CMake 3.9, way ahead of current Xenial packages (3.5). We've discovered recently a nasty bug at YARP devel that prevented it from working on systems with CMake 3.5, but **didn't show up in Travis builds**: https://github.com/robotology/yarp/issues/1727.

PeterBowman, 2018-06-19:
> Therefore, working with software releases so close to the minimum requirements makes all this pondering rather pointless.

To sum up:

* CMake: in spite of building against 3.9 in Travis, most of us are working with the current minimum supported release: CMake 3.5 (Trusty, Xenial). Bionic goes a bit beyond that ([3.10](https://packages.ubuntu.com/bionic/cmake)), but I don't think anyone is going to introduce new and exotic features without noticing such pitfalls.

* YARP: I've just set 2.3.70 as the minimum supported release in several repositories, we can go even a bit further (2.3.72). New jobs will be added to cover 2.x and 3.x in the same build, see https://github.com/roboticslab-uc3m/questions-and-answers/issues/65.

* YCM: introducing cron jobs for *master* and *devel* (https://github.com/roboticslab-uc3m/questions-and-answers/issues/55#issuecomment-397803268), just like YARP.

* Other dependencies: so far not an issue, handle specific cases if needed.

Closing as invalid.

closed 2022-06-04 by PeterBowman