roboticslab-uc3m/yarp-devices #115: teo and teoSim legs movement mismatch

munozyanez, 2017-04-20:
Launching ´´yarpdev --device BasicTwoLimbCartesianControl´´ results in a faulty movement on teo. 
teoSim is working correctly with the same parameters.

Find below some points as example.

´´
New trajectory segment : 1->2, Segment index: 1
> rfX: 0,-0.124531,-0.825,> rot: -0.707107,-4.32978e-17,0.707107,180,
New trajectory segment : 1->2, Segment index: 1
> lfX: 0,0.132469,-0.825,> rot: 0,-1,0,90,
[STEP] [0.239693] x 0.000000 -0.124531 -0.825000 -0.707107 -0.000000 0.707107 180.000000 0.000000 0.132469 -0.825000 0.000000 -1.000000 0.000000 90.000000 xdot 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
[STEP] [0.239693] 0.001512 0.005314 0.019545 0.003074 -0.008279 -0.001534 -> 0.090844 -0.446273 -43.171995 90.580035 -47.882676 0.270900 [deg/s]
rightLeg encoders: [-0.0878735, 0.0878735, -2.0914, 4.48155, -1.91565, 0.0878735]
[STEP] [0.239693] -0.000482 0.007203 0.019518 0.004293 -0.001233 0.002763 -> 0.157589 -0.599489 -47.676554 100.258585 -52.652003 0.353329 [deg/s]
leftLeg encoders: [-0.158173, 0.246063, -2.02109, 4.11249, -2.02109, -0]
> rfX: 0,-0.11952,-0.825,> rot: -0.707107,-4.32978e-17,0.707107,180,
> lfX: 0,0.13748,-0.825,> rot: 0,-1,0,90,
[STEP] [0.289795] x 0.000000 -0.119520 -0.825000 -0.707107 -0.000000 0.707107 180.000000 0.000000 0.137480 -0.825000 0.000000 -1.000000 0.000000 90.000000 xdot 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
[STEP] [0.289795] -0.005570 0.010314 0.019619 0.003062 0.007058 -0.001534 -> 0.082125 -0.902988 -47.336021 100.284197 -52.544070 0.726950 [deg/s]
rightLeg encoders: [-0.0878735, 0.0878735, -2.35501, 4.04218, -2.0914, 0.0878735]
[STEP] [0.289795] -0.001189 0.007640 0.019607 -0.003067 0.000311 0.002761 -> 0.158424 -0.731755 -46.845017 98.606666 -51.744314 0.907526 [deg/s]
leftLeg encoders: [-0.158173, -0.0878735, -2.10896, 4.20035, -2.10896, -0.0878735]

´´

munozyanez, 2017-04-25:
Corrected some issues about motor turn directions.
Redone the xml openrave teo  ( https://github.com/roboticslab-uc3m/teo-openrave-models ) definition for both legs according to real robot. Ready to pull request if no worries.
Also redone DH for both legs according to new turn directions. Also ready to pull request.

munozyanez, 2017-04-25:
But... the robot is still not moving correctly, next we will review the motor speeds for legs to avoid errors in that direction.

jgvictores, 2017-04-26:
Okay, my main concern is to keep matching descriptions across files/repositories.

Kinematics:
- https://github.com/roboticslab-uc3m/teo-software-manual (both DH table and DH diagram of assets directory, .svg source in assets/src !!).
- https://github.com/roboticslab-uc3m/teo-configuration-files

Motor:
- https://github.com/roboticslab-uc3m/yarp-devices but as seen in #116 should be moved to https://github.com/roboticslab-uc3m/teo-configuration-files
- https://github.com/roboticslab-uc3m/teo-openrave-models

Could you please be very careful with this and keep updated and documented ??
-

munozyanez, 2017-04-27:
I can update the drawings, but I will need the source files. Cant find in ( http://robots.uc3m.es/svn/TEOrepo/modelling/diagrams/editable/ ). Is there another place to look for apart from this one?

jgvictores, 2017-04-27:
- https://github.com/roboticslab-uc3m/teo-software-manual/tree/master/assets
- https://github.com/roboticslab-uc3m/teo-software-manual/tree/master/assets/src

munozyanez, 2017-04-27:
Ok, shall I pull request or just update the files?

jgvictores, 2017-04-27:
I think its okay to update directly, just try to cross-reference as much as possible (mention this issue in commits, etc). A branch and merge --no-ff is good as well. Thanks!

munozyanez, 2017-04-27:
Ok, done updates. All but https://github.com/roboticslab-uc3m/yarp-devices , as it seems to be deprecated.
Not closing issue, because leg movement is still faulty.
Maybe related with Absolute encoders signs in configuration files?

munozyanez, 2017-04-27:
Sorry, I mean the files will be no longer in this repo, not that repo will be deprecated.

jgvictores, 2017-11-15:
This issue was moved to roboticslab-uc3m/teo-main#25

closed 2017-11-15 by jgvictores