roboticslab-uc3m/yarp-devices #116: Export TEO files to teo-configuration-files

jgvictores, 2017-04-26:
Export TEO files (from share) to [teo-configuration-files](https://github.com/roboticslab-uc3m/teo-configuration-files):
- [ ] launchManipulation
- [ ] launchLocomotion
- [ ] ???

rsantos88, 2017-12-05:
You've this files:
* checkCanBus
* dumpCanBus
* launchLocomotion
* launchManipulation
* oneCanBusOneWrapper
* spaceNavigator
* testCuiAbsolute

Do you want to export all?? especially, I have doubt with `spaceNavigator`

jgvictores, 2017-12-05:
The main idea is to move the teo-specific files, which usually mention `manipulation` or `locomotion` in its name. This is:
- From [share/checkCanBus/conf](https://github.com/roboticslab-uc3m/yarp-devices/tree/9b92f99eb99d6de553dbf1ed748cfa5ccef65556/share/checkCanBus/conf): `checkCanBus.ini` should stay; **the rest** should be moved (:truck:).
- dumpCanBus: stay.
- launchLocomotion: move (:truck:).
- launchManipulation: move (:truck:).
- oneCanBusOneWrapper: stay.
- spaceNavigator: stay.
- testCuiAbsolute: stay.

PeterBowman, 2017-12-05:
Don't delete [`launchManipulation.ini`](https://github.com/roboticslab-uc3m/yarp-devices/blob/9b92f99eb99d6de553dbf1ed748cfa5ccef65556/share/launchManipulation/conf/launchManipulation.ini)/[`launchLocomotion.ini`](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/share/launchLocomotion/conf/launchLocomotion.ini), just keep a minimal working example. Compare with [`oneCanBusOneWrapper.ini`](https://github.com/roboticslab-uc3m/yarp-devices/blob/9b92f99eb99d6de553dbf1ed748cfa5ccef65556/share/oneCanBusOneWrapper/conf/oneCanBusOneWrapper.ini): there is some placeholder data, but we'll want to preserve a template so that users may adjust it to their needs.

(I'd go a step ahead and merge both files into `twoCanBusThreeWrappers.ini`, see https://github.com/roboticslab-uc3m/yarp-devices/issues/20#issuecomment-346185955)

jgvictores, 2017-12-05:
@PeterBowman Let's agree on that final proposal, which involves the following workflow:
1. Copy `share/launchManipulation/conf/launchManipulation.ini` to a new `share/twoCanBusThreeWrappers/conf/twoCanBusThreeWrappers.ini`.
1. Move `share/launchManipulation/conf/launchManipulation.ini` and `share/launchLocomotion/conf/launchLocomotion.ini` to teo-configuration-files.

jgvictores, 2017-12-05:
PS: Don't worry about breaking things, I'll work on #20 as a follow-up to this issue.

PeterBowman, 2017-12-05:
@jgvictores OK!

rsantos88, 2017-12-05:
> 1. Copy `share/launchManipulation/conf/launchManipulation.ini` to a new `share/twoCanBusThreeWrappers/conf/twoCanBusThreeWrappers.ini`.

done it here (2db6a0d9d1b71deb77adeb88df3dffa304bfb5f3)

rsantos88, 2017-12-05:
There are other files (maybe useful files) together with `launchManipulation.ini` ([see](https://github.com/roboticslab-uc3m/yarp-devices/tree/develop/share/launchManipulation/conf)) like `launchManipulation-fake.ini`. Do you want to move them too? or can I delete them?

rsantos88, 2017-12-05:
One note: I was thinking... `twoCanBusThreeWrappers.ini` not depend of any application, so... we can delete [this](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/share/CMakeLists.txt#L25) and [this](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/share/CMakeLists.txt#L27) lines if we want to install this file by default...?? or do you want use it like a example...

PeterBowman, 2017-12-05:
I'd keep them, we'll need this at #20.

BTW what do we need [these comments](https://github.com/roboticslab-uc3m/yarp-devices/commit/2db6a0d9d1b71deb77adeb88df3dffa304bfb5f3#diff-44b95faacaf72fc3b2754f8e266389d0R29) for? It looks like :wastebasket: and bloats the source code.

jgvictores, 2017-12-05:
Yep, essentially all the non-author-header cmake comment lines can be removed.

jgvictores, 2017-12-05:
And yes, something that requires twoCanBusThreeWrappers.ini will emerge from #20

PeterBowman, 2017-12-09:
Adding the `blocked` label as a notice/reminder that there is ongoing CMake-refactorization work at #26, which I'm going to close as soon as I get my hands on ASIBOT and run some tests. Don't let that block your progress, though, I'll fix any conflicts on my side if your commits are merged first.

rsantos88, 2017-12-11:
> There are other files (maybe useful files) together with `launchManipulation.ini` ([see](https://github.com/roboticslab-uc3m/yarp-devices/tree/develop/share/launchManipulation/conf)) like `launchManipulation-fake.ini`. Do you want to move them too? or can I delete them?
 
and how about this?

jgvictores, 2017-12-11:
In general, I'd say move them. For the specific case of `launchManipulation-fake.ini`, delete, since `FakeControlboard` no longer exists.
Regarding the new `twoCanBusThreeWrappers.ini` we could fit in a `FakeJoint` somewhere to serve as example.

rsantos88, 2017-12-11:
> Move share/launchManipulation/conf/launchManipulation.ini and share/launchLocomotion/conf/launchLocomotion.ini to teo-configuration-files.

done it [here](https://github.com/roboticslab-uc3m/teo-configuration-files/commit/185dfa9675aaa251fe21504478e214ed8c4396ac)

rsantos88, 2017-12-11:
>Regarding the new twoCanBusThreeWrappers.ini we could fit in a FakeJoint somewhere to serve as example.

Do you want to replace some `TechnosoftIpos` by `FakeJoint` in `twoCanBusThreeWrappers.ini`.
For example... with the frontal wrists?

jgvictores, 2017-12-11:
Sure, great!

PeterBowman, 2017-12-11:
> Adding the `blocked` label as a notice/reminder that there is ongoing CMake-refactorization work...

Unblocking, I'm going to close that issue as soon as file shuffling is over.

rsantos88, 2017-12-11:
done at (ee6ee025621c13dad60510f3f50ef79be3dc31d3)

jgvictores, 2017-12-11:
@PeterBowman I think we're done with file shuffling. All yours, right?

PeterBowman, 2017-12-11:
@jgvictores done, thanks!

jgvictores, 2017-12-12:
Closing, as #26 is also closed! Thanks @rsantos88 @PeterBowman !!!

closed 2017-12-12 by jgvictores