roboticslab-uc3m/yarp-devices #117: problems with the inizialitation of jr3 devices

jmgarciah, 2017-05-04:
~$ yarpdev --device Jr3 --period 20 --name /jr3 --ports "(ch0:o ch1:o ch2:o ch3:o)" --channels 24 --ch0:o 0 5 0 5 --ch1:o 6 11 0 5 --ch2:o 12 17 0 5 --ch3:o 18 23 0 5
[INFO]No ROS group found in config file ... skipping ROS initialization. 
[ERROR]AnalogWrapper: unable to open the analog server, address conflict 
[ERROR]AnalogServer Error initializing YARP ports 
yarpdev: ***ERROR*** driver <analogServer> was found but could not open
yarpdev: ***ERROR*** device not available.
Suggestions:
+ Do "yarpdev --list" to see list of supported devices.
+ Or append "--verbose" option to get more information.

PeterBowman, 2017-05-04:
> [ERROR]AnalogWrapper: unable to open the analog server, address conflict

This line says that the ports you are trying to open were already taken by another process. Alternatively, a previous run did not close cleanly and said ports were not garbage collected by the nameserver. Launch a `yarp name list` and look for `/serial/in` and `/serial/out`. These two should disappear after running `yarp clean --timeout 1` (provided you are not using a serialport device anywhere in your YARP network).

jgvictores, 2017-05-04:
Just to not start hunting zombies and invoking `pgrep`, for now, I recommended the IT Crowd solution. ;-)

[image]

jgvictores, 2017-05-05:
Right now what was actually happening was the device was launched on a different PC of the network.

This situation was detected by running a `yarp name list`: `jr3` ports were on an IP that was not the expected one.

Solutions to avoid this kind of situations in the future:
- Physically disconnect from same network to launch same program on a different PC.
- Use `yarp namespace` to be on two YARP networks on the same physical network.
- Launch `export YARP_PORT_PREFIX=/fancyPrefix` before running device/program on different PC.

closed 2017-05-26 by jgvictores