roboticslab-uc3m/yarp-devices #118: Try hc-driver-4.9

jgvictores, 2017-05-06:
If https://github.com/munozyanez/hc-driver-4.9 works, this may affect the Distro we use, propagating to #111 and even https://github.com/roboticslab-uc3m/QA/issues/17.

munozyanez, 2017-05-07:
Well, after some tests it seems to work in debian 9. At least it is sending and receiving.
There is a test bench in teo lab, with that driver working (on debian 9). I would plan some heavy tests over that bench before rely on it. We can add another disk with Ubuntu.

PeterBowman, 2017-05-11:
With regard to the integration in TEO and ASIBOT, @jgvictores and I have concluded that Ubuntu Xenial (16.04) would be a reasonable choice considering compatibility with existing systems and availability of newer software releases. Concerning Travis CI support, see travis-ci/travis-ci#5821.

munozyanez, 2017-05-11:
My vote goes for debian, but I see no problem in using Ubuntu.

PeterBowman, 2017-05-11:
If we finally go the Ubuntu way, a valid concern for ASIBOT's RoBoard and its scarce resources would lead us to limiting installed tools and applications (see https://askubuntu.com/q/16137). That being said, we don't really need to configure the exact same image for both robots. Let's revisit the SO choice at a later stage.

munozyanez, 2017-05-11:
I can't say about the resources consumed by each one, but afaik, debian allows more custom configurations from fresh installs, and that can be good for different platforms.
On the other hand, Ubuntu is a bit more user friendly.
We should be able to run the same software on both, so yes, let's revisit later.

jgvictores, 2017-05-19:
@PeterBowman @munozyanez I've started https://github.com/roboticslab-uc3m/QA/issues/20 to start the distro flameware elseware. LOL

@munozyanez I'm really grateful for the effort you've put into this repo, and hope to be able to test this as soon as possible and obtaining positive results!

munozyanez, 2017-05-19:
An easy trial could be to launch launchLocomotion on the test bench, but unfortunately, only one of the channels is working for the mounted hicocan.
Maybe we can borrow the card from locomotion cpu and make a test.

jgvictores, 2017-05-20:
> Maybe we can borrow the card from locomotion cpu and make a test.

Yes, that would be great!

dgomezmartinez, 2017-07-19:
Just borrowed the hicocan board and tested both drivers.
Got partial results, for it is sending and receiving, but the program launchLocomotion is not fully working. See the output below.
```
teo@pruebas:~/launches$ launchLocomotion --from /usr/local/share/roboticslab-yarp-devices/contexts/launchLocomotion/launchLocomotion.ini
||| clearing context
||| adding context [launchLocomotion]
||| configuring
||| default config file specified as /usr/local/share/roboticslab-yarp-devices/contexts/launchLocomotion/launchLocomotion.ini
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/launchLocomotion/launchLocomotion.ini] (pwd)
||| found /usr/local/share/roboticslab-yarp-devices/contexts/launchLocomotion/launchLocomotion.ini
[info] main.cpp:57 main(): Checking for yarp network...
[success] main.cpp:64 main(): Found yarp network.
[debug] TwoCanBusThreeWrappers.cpp:47 configure(): devCan0 (device CanBusControlboard) (canDevice "/dev/can0") (types TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos TechnosoftIpos CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute CuiAbsolute) (ids 14 6 5 4 3 2 1 114 106 105 104 103 102 101) (maxs 50 90 20 45 80 25 25 0 0 0 0 0 0 0) (mins -10 -90 -20 -30 -5 -25 -25 0 0 0 0 0 0 0) (maxVels 1000 1000 1000 1000 1000 1000 1000 0 0 0 0 0 0 0) (minVels 0 0 0 0 0 0 0 0 0 0 0 0 0 0) (trs 480 400 523.2 192 235.2 270.4 235.2 -1 -1 1 1 1 1 1) (refAccelerations 10.0 10.0 10.0 10.0 10.0 10.0 10.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) (refSpeeds 10.0 10.0 10.0 10.0 10.0 10.0 10.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0)
[success] DeviceDriverImpl.cpp:20 open(): Opened CAN device of path: /dev/can0
[success] DeviceDriverImpl.cpp:30 open(): Bitrate set on CAN device: /dev/can0
[success] DeviceDriverImpl.cpp:40 open(): IOC_START ok on CAN device: /dev/can0
[INFO]created device <CanBusHico>. See C++ class teo::CanBusHico for documentation.
[info] ThreadImpl.cpp:10 run(): Started CanBusControlboard reading thread run.
[success] DeviceDriverImpl.cpp:57 open(): Created TechnosoftIpos with canId 14, tr 480.000000, k 0.000000, refAcceleration 10.000000, refSpeed 10.000000, and all local parameters set to 0.
[INFO]created device <TechnosoftIpos>. See C++ class teo::TechnosoftIpos for documentation.
[success] ICanBusSharerImpl.cpp:11 setCanBusPtr(): Ok pointer to CAN bus device 14.
[info] IPositionControl2RawImpl.cpp:226 setRefAccelerationRaw(): (0, 10.000000)
[success] IPositionControl2RawImpl.cpp:248 setRefAccelerationRaw(): Sent "posmode_acc". 23 83 60 0 fb d 0 0. canId(14) via(600).
[info] IPositionControl2RawImpl.cpp:179 setRefSpeedRaw(): (0, 10.000000)
[info] ICanBusSharerImpl.cpp:735 interpretMessage(): Got SDO ack "posmode_acc" from driver. 60 83 60 0 0 0 0 0. canId(14) via(580).
[success] IPositionControl2RawImpl.cpp:203 setRefSpeedRaw(): Sent "posmode_speed". 23 81 60 0 c0 9f 36 0. canId(14) via(600).
[info] IControlLimits2RawImpl.cpp:9 setLimitsRaw(): (0,-10.000000,50.000000)
[info] ICanBusSharerImpl.cpp:740 interpretMessage(): Got SDO ack "posmode_speed" from driver. 60 81 60 0 0 0 0 0. canId(14) via(580).
[success] IControlLimits2RawImpl.cpp:35 setMinLimitRaw(): Sent "position min". 23 7d 60 1 a0 2a ff ff. canId(14) via(600).
[info] ICanBusSharerImpl.cpp:747 interpretMessage(): Got SDO ack "msg_position_min" from driver. 60 7d 60 1 0 0 0 0. canId(14) via(580).
[success] IControlLimits2RawImpl.cpp:59 setMaxLimitRaw(): Sent "position max". 23 7d 60 2 e0 2a 4 0. canId(14) via(600).
[info] ICanBusSharerImpl.cpp:751 interpretMessage(): Got SDO ack "msg_position_max" from driver. 60 7d 60 2 0 0 0 0. canId(14) via(580).
[success] DeviceDriverImpl.cpp:57 open(): Created TechnosoftIpos with canId 6, tr 400.000000, k 0.000000, refAcceleration 10.000000, refSpeed 10.000000, and all local parameters set to 0.
[INFO]created device <TechnosoftIpos>. See C++ class teo::TechnosoftIpos for documentation.
[success] ICanBusSharerImpl.cpp:11 setCanBusPtr(): Ok pointer to CAN bus device 6.
[info] IPositionControl2RawImpl.cpp:226 setRefAccelerationRaw(): (0, 10.000000)
[success] IPositionControl2RawImpl.cpp:248 setRefAccelerationRaw(): Sent "posmode_acc". 23 83 60 0 a7 b 0 0. canId(6) via(600).
[info] IPositionControl2RawImpl.cpp:179 setRefSpeedRaw(): (0, 10.000000)
[info] ICanBusSharerImpl.cpp:735 interpretMessage(): Got SDO ack "posmode_acc" from driver. 60 83 60 0 0 0 0 0. canId(6) via(580).
[success] IPositionControl2RawImpl.cpp:203 setRefSpeedRaw(): Sent "posmode_speed". 23 81 60 0 20 85 2d 0. canId(6) via(600).
[info] IControlLimits2RawImpl.cpp:9 setLimitsRaw(): (0,-90.000000,90.000000)
[info] ICanBusSharerImpl.cpp:740 interpretMessage(): Got SDO ack "posmode_speed" from driver. 60 81 60 0 0 0 0 0. canId(6) via(580).
[success] IControlLimits2RawImpl.cpp:35 setMinLimitRaw(): Sent "position min". 23 7d 60 1 b0 bf f9 ff. canId(6) via(600).
[info] ICanBusSharerImpl.cpp:747 interpretMessage(): Got SDO ack "msg_position_min" from driver. 60 7d 60 1 0 0 0 0. canId(6) via(580).
[success] IControlLimits2RawImpl.cpp:59 setMaxLimitRaw(): Sent "position max". 23 7d 60 2 50 40 6 0. canId(6) via(600).
[info] ICanBusSharerImpl.cpp:751 interpretMessage(): Got SDO ack "msg_position_max" from driver. 60 7d 60 2 0 0 0 0. canId(6) via(580).
[success] DeviceDriverImpl.cpp:57 open(): Created TechnosoftIpos with canId 5, tr 523.200000, k 0.000000, refAcceleration 10.000000, refSpeed 10.000000, and all local parameters set to 0.
[... cut ...]
[success] IControlLimits2RawImpl.cpp:59 setMaxLimitRaw(): Sent "position max". 23 7d 60 2 62 5 1 0. canId(1) via(600).
[info] ICanBusSharerImpl.cpp:751 interpretMessage(): Got SDO ack "msg_position_max" from driver. 60 7d 60 2 0 0 0 0. canId(1) via(580).
[success] DeviceDriverImpl.cpp:21 open(): Created CuiAbsolute with canId 114 and tr -1.000000, and all local parameters set to 0.
[INFO]created device <CuiAbsolute>. See C++ class teo::CuiAbsolute for documentation.
[success] ICanBusSharerImpl.cpp:11 setCanBusPtr(): Ok pointer to CAN bus device 114.
Instrucción ilegal
teo@pruebas:~/launches$ 
```

dgomezmartinez, 2017-07-19:
Following the error, it seems to break at Cui absolute encoder loading.
Removing the Cui configuration from launchLocomotion.ini , launchLocomotion is working fine (legs are moving). Output is too large to paste here.

jgvictores, 2017-11-15:
Opened https://github.com/roboticslab-uc3m/teo-main/issues/26 for TEO-specific tests. Please continue here with any advances on any other test-bench.

jgvictores, 2018-02-02:
Closed via https://github.com/roboticslab-uc3m/teo-main/issues/26 (a humanoid is a complex system, good enough for proof-of-concept! )

closed 2018-02-02 by jgvictores