roboticslab-uc3m/yarp-devices #119: Provide encoderSlots config option

PeterBowman, 2017-05-10:
There are several places throughout the code in the `TechnosoftIpos` tree where we may encounter the *magic* number 11.38 involved in unit translation calculations: [example](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef/libraries/BodyYarp/TechnosoftIpos/IVelocityControl2RawImpl.cpp#L31). This value originates from the number of TEO's encoder slots and is further used to convert to internal units (IU) of the Technosoft driver. This should be parameterized in order to support different configurations.

PeterBowman, 2017-05-10:
@jgvictores and I have concluded that a more generic approach would fit better: encoderPulses. This value encompasses the pulses-per-slot factor (usually 4) and number of total slots of the encoder (currently: 4 * 1024).

rsantos88, 2017-11-27:
Ok, where do you think is better to provide this config option? Maybe in a .ini or where do you want to add it?

jgvictores, 2017-11-27:
Yes, for example in [this .ini](https://github.com/roboticslab-uc3m/yarp-devices/blob/9b92f99eb99d6de553dbf1ed748cfa5ccef65556/share/launchLocomotion/conf/launchLocomotion.ini) (which is still around until #116 is done), it would mean an additional line under `refSpeeds`.

rsantos88, 2017-11-27:
I need to test it on the robot to confirm that it's working well.
I've created a temporal branch to check it. Here it is #144

PeterBowman, 2017-12-04:
> I need to test it on the robot to confirm that it's working well.

OK to close (#144 has been merged)?

jgvictores, 2017-12-04:
Yes! Thanks @rsantos88  @PeterBowman !!

closed 2017-12-04 by jgvictores