roboticslab-uc3m/yarp-devices #121: Read current speed and acceleration values from TechnosoftIpos device via CAN

PeterBowman, 2017-05-10:
Currently, commanded speeds and accelerations get stored in their respective member fields, `TechnosoftIpos::refSpeed` and `TechnosoftIpos::refAcceleration`, upon each set-like function call (involving thread locks): [example](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef/libraries/BodyYarp/TechnosoftIpos/IPositionControl2RawImpl.cpp#L258). It would be preferable to pull the current value from the driver instead of evaluating said fields: [clic/ref](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef/libraries/BodyYarp/TechnosoftIpos/IPositionControl2RawImpl.cpp#L283).

PeterBowman, 2017-12-12:
Rebased on `develop` and WIP at b076348, now I have to apply #119/#144 and run some tests (again).

PeterBowman, 2017-12-13:
Done at ef0fad9, tested on ASIBOT.

PeterBowman, 2017-12-13:
Just for clarification, now we may pull from the driver:

* actual reference speeds and accelerations (position mode)
* current velocity (velocity mode)

Don't be scared when you'll see lots of decimals instead of fixed integers (e.g. 4.9999 instead of 5).

closed 2017-12-13 by PeterBowman

jgvictores, 2017-12-14:
Great!

Thanks to this, I also reviewed the code and started #153.