roboticslab-uc3m/yarp-devices #122: Replace current ICanBusHico interface with yarp::dev::ICanBus

PeterBowman, 2017-05-11:
Consider migrating [ICanBusHico](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef/libraries/BodyYarp/ICanBusHico.h) to [yarp::dev::ICanBus](http://www.yarp.it/classyarp_1_1dev_1_1ICanBus.html). This process would involve implementing new methods in the `CanBusHico` device and updating the type of current CAN interface handles (`canDevicePtr`).

jgvictores, 2017-05-11:
We would also have to wrap the current [struct can_msg](https://github.com/roboticslab-uc3m/yarp-devices/blob/cce12ef517b333bd1ecab93cc8eb5cc65868f66f/libraries/BodyYarp/CanBusHico/hico_api.h#L302) in a [yarp::dev::CanMessage](http://www.yarp.it/classyarp_1_1dev_1_1CanMessage.html) and then offer some kind of buffer mechanism as [yarp::dev::CanBuffer](http://www.yarp.it/classyarp_1_1dev_1_1CanBuffer.html) is the intermediate actor used by [yarp::dev::ICanBus](http://www.yarp.it/classyarp_1_1dev_1_1ICanBus.html).

PeterBowman, 2017-05-19:
I was having a hard time understanding bit masks in CAN acceptance filters (regarding the implementation of `yarp::dev::ICanBus::canIdAdd`). I'll drop here a couple of links I might want to revisit in the future:

* http://www.kvaser.com/canlib-webhelp/page_user_guide_misc_code_and_mask.html
* http://can-bus.996267.n3.nabble.com/Setting-CAN-MASK-amp-ID-registers-td2304.html

Plus more links in the commit description at 5651204.

PeterBowman, 2017-12-22:
The `roboticslab::ICanBusHico` interface has been killed and replaced by `yarp::dev::ICanBus` in b579638 and d2af6b5. There is one caveat to using `yarp::dev::CanMessage`: no support for timestamps. I had to replace the `ts` field from the `can_msg` structure with a plain call to `yarp::os::Time::now()`: 1a4ba55. I hope that there's no noticeable difference in behavior. Still WIP, requires testing.

Anyway, notice that `can_msg::ts` uses a totally different scale and reference than `yarp::os::Time::now()`:

* https://github.com/roboticslab-uc3m/yarp-devices/blob/f2f659ec3c4575bb691799ba9948d494bdf0c01d/libraries/YarpPlugins/CanBusHico/hico_api.h#L344-L345

* [yarp::os::Time::now](http://www.yarp.it/classyarp_1_1os_1_1Time.html#adf94a9bd3f41b6e57793be5ec536c707): *Return the current time in seconds, relative to an arbitrary starting point.*

A quick search throughout roboticslab-uc3m repos revealed that we already adopted the latter. Encoder timestamps are used by `controlboardwrapper2` to build the time envelope in status streams: [ref1](https://github.com/robotology/yarp/blob/8be516188ec354f82e4e6852a61364a16ef288d7/src/libYARP_dev/src/devices/ControlBoardWrapper/SubDevice.h#L124), [ref2](https://github.com/robotology/yarp/blob/8be516188ec354f82e4e6852a61364a16ef288d7/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L855).

jgvictores, 2017-12-23:
Agreed, we had a minimal implementation of support for `can_msg::ts`, e.g. [here](https://github.com/roboticslab-uc3m/yarp-devices/blob/f6fc15d06682a4569e6ec23eb7f1483d06f5186c/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L271) and [here](https://github.com/roboticslab-uc3m/yarp-devices/blob/f6fc15d06682a4569e6ec23eb7f1483d06f5186c/libraries/YarpPlugins/TechnosoftIpos/IEncodersTimedRawImpl.cpp#L37), but the information was not being used anywhere.
Given the case that the information would be discarded anyway, I think it's safe to live without it for now.

Thank you so much!

PeterBowman, 2018-07-19:
Mostly done. Read more about the new policies introduced by PR #158 in its description.

PR #158 (2018-07-19): Adopt YARP interfaces and implementations for CAN messages and bus drivers
Resolves #122. Main goals and improvements:

* Replace the `roboticslab::ICanBusHico` interface with [`yarp::dev::ICanBus`](http://www.yarp.it/classyarp_1_1dev_1_1ICanBus.html).
* Wrap CAN messages in [`yarp::dev::CanMessage`](http://www.yarp.it/classyarp_1_1dev_1_1CanMessage.html).
* New feature: [CAN acceptance filtering capability](http://www.cse.dmu.ac.uk/~eg/tele/CanbusIDandMask.html) (disabled by default) -> [**policy**](https://github.com/roboticslab-uc3m/yarp-devices/pull/158#issuecomment-404898303).
* Support f [... cut]
closed 2018-07-19 by jgvictores