roboticslab-uc3m/yarp-devices #125: YARP device for Aravis Camera

David-Estevez, 2017-07-03:
I created this issue as a place to discuss the ongoing development of the YARP device for a GigE camera using Aravis that is currently taking place in my [fork of this repo](https://github.com/David-Estevez/yarp-devices).

I have achieved to stream both a fake Aravis driver and the real GigE camera, altough is still no clear to me which is the best FrameGrabber interface to send the data. Currently I am using `IFrameGrabberImage`, but I will likely change this by a `IFrameGrabber` to send the raw buffer. Then, the demosaicing can be applied through the corresponding YARP carrier (in this case, `udp+recv.bayer+order.gbrg`). Or within the class through another interface such as `IFrameGrabberImage` or `IFrameGrabberImageRaw`.

I am currently working on exposing some methods to control the camera parameters. YARP has two interfaces for controlling the cameras (`FrameGrabberControl` and `FrameGrabberControl2`). The first one seams to deal with concrete camera parameters (gain, exposure, fps), but some of the parameters I need to expose (zoom, focus) seem not supported. They are listed on the parameter enum, but getters/setter where never implemented. The second interface seems to be a more general control inteface, but has the disadvantage that the user must know the name of the parameters he wants to access as they are set through a  `setFeature` generic function. Seems to be a lower-level interface, but still not useful for us.

That's why I created a `ICameraLensControls` interface, and I will try to use it as a proof of concept. If it works nicely I will create similar interfaces for other controls.

David-Estevez, 2017-07-03:
I have tested the `ICameraLensControls` and it works, now we can control the camera zoom from C++ code using YARP. Next step is to provide support for controlling the camera parameters such as zoom from a YARP command-line interface, as we would do with motors, for instance.

According to @jgvictores there are currently two choices: 
1. Do it by hand, as in the [CartesianControlServer](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/develop/libraries/TeoYarp)

2. Use the newest [IDL interfaces](http://www.yarp.it/thrift_tutorial_simple.html)

David-Estevez, 2017-07-04:
After a deeper converstation with @jgvictores, it seem that the best choice is to use the `IFrameGrabberControls2` interface that supports all the different parameters I need. It is also supported by the YARP rpc mechanism, so I won't have to do the server myself after all.

Commands are sent through the `/grabber` port in RPC mode:

```
$ yarp rpc /grabber
```

And the format of the command, for a `IFrameGrabberControls2` interface is more or less the following:

```
fgc2 get feat zoom
Response: [fgc2] [feat] [is] 0.0
```

Therefore, I'll change the existing code to work with this new `IFrameGrabberControls2` interface.

David-Estevez, 2017-11-06:
Note to myself: due to https://github.com/roboticslab-uc3m/questions-and-answers/issues/15 and https://github.com/roboticslab-uc3m/yarp-devices/pull/129 I need to update [my fork ](https://github.com/David-Estevez/yarp-devices) to follow the current conventions.

jgvictores, 2017-11-06:
Great! Thanks for taking this into account!!

David-Estevez, 2017-11-08:
This PR also affects this repo: https://github.com/roboticslab-uc3m/yarp-devices/pull/130

David-Estevez, 2017-12-07:
We already have a nice AravisGigE device and a beautiful GUI to control it, so I'm considering this completed and closing this issue.

closed 2017-12-07 by David-Estevez

jgvictores, 2017-12-08:
Perfect!

PeterBowman, 2021-05-24:
> Currently I am using `IFrameGrabberImage`, but I will likely change this by a `IFrameGrabber` to send the raw buffer.

`IFrameGrabber` is deprecated as of YARP 3.5 (unreleased), see https://github.com/robotology/yarp/pull/2582. I have merged our implementation into `IFrameGrabberImageRaw::getImage` at https://github.com/roboticslab-uc3m/yarp-devices/commit/89c182b481782aed80a2af4a8551fc0cfe47e940. I'm also switching from the deprecated `grabber` wrapper to `grabberDual`, which doesn't support `IFrameGrabber` anyway. I think the `yarp::sig::PixelMono` frame type fulfills the needs of this device and exposing raw unsigned chars is not necessary.