roboticslab-uc3m/yarp-devices #128: Improve examples

jgvictores, 2017-11-04:
Improve examples:
- Fix broken examples
- Import required examples from asibot-main, see https://github.com/roboticslab-uc3m/asibot-main/issues/38

jgvictores, 2017-11-06:
Have to be careful with https://github.com/roboticslab-uc3m/yarp-devices/pull/129 \:\-O

David-Estevez, 2017-12-08:
Let me leave you here what I think could be a good example for the IFrameGrabberControls2 bindings in Python (continued from #140):

```python
#Check for YARP network
yarp.Network.init()
if not yarp.Network.checkNetwork():
    print('Could not connect to YARP network. Please try running YARP server.')
    quit()

# Create and configure driver
options = yarp.Property()
options.put('device','remote_grabber')
options.put('remote', '/grabber')
options.put('local','/GrabberControls2Gui')
dd = yarp.PolyDriver(options)

# View driver as FrameGrabber
controls = dd.viewIFrameGrabberControls2()

# Check if a feature exists
print(controls.hasFeature(yarp.YARP_FEATURE_GAIN))

# Get the current value of a feature
gain = controls.getFeature(yarp.YARP_FEATURE_GAIN)
print(gain)

# Set a new value for a feature
controls.setFeature(yarp.YARP_FEATURE_GAIN, 10)

# Close device and disconnect from the YARP network
dd.close()
yarp.Network.fini()

```

jgvictores, 2017-12-09:
@David-Estevez Thanks!

PeterBowman, 2017-12-11:
> Fix broken examples

Reviewed at e0c8394 (not merged yet), all examples compile successfully.

jgvictores, 2018-02-03:
> Import required examples from asibot-main, see roboticslab-uc3m/asibot-main#38

Done at 20531227d19a189acf612aa352833d3ed17ca453

jgvictores, 2018-02-03:
> Example of IFrameGrabberControls2

Done at 6dcce3b235dab788bda1b3425207b2d55b6cb074

closed 2018-02-03 by jgvictores