roboticslab-uc3m/yarp-devices #132: In David-Estevez fork, branch for GigE support Aravis device is not working

err8029, 2017-11-10:
Al usar ccmake con las opciones Aravis y BodyYarp, me muestra el siguiente error. 
-He comprobado que las dirs de las librerias que ccmake llama sean correctas y aun asi nada
[image]
-Si solo compilo aravis en ccmake, la compilación es exitosa, pero el device no aparece en la lista

jgvictores, 2017-11-10:
Por favor copia y pega la salida por cosola de:
- `cmake ..`
- `make`
- `sudo make install`
- `sudo ldconfig`
- `yarpdev --list`

err8029, 2017-11-13:
Al intentar inciar el device me sale otro error aunque este se encuentra en la lista

teo@teo-ebola:~/eric/repos/yarp-devices-feature-gige-support/build$ yarpdev --device /AravisGigE --name /grabber --fake
[ERROR]Cannot find "/AravisGigE" plugin (not built in, and no .ini file found for it)
[ERROR]Check that YARP_DATA_DIRS leads to at least one directory with plugins//AravisGigE.ini or share/yarp/plugins//AravisGigE.ini in it
yarpdev: ***ERROR*** could not find device </AravisGigE>
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.

---------------------------------------------------- lista ------------------------------------------------------------

teo@teo-ebola:~/eric/repos/yarp-devices-feature-gige-support/build$ yarpdev --list
[INFO]Here are devices listed for your system:
[INFO]Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
[INFO]Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
[INFO]Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
[INFO]Device "sound_grabber", C++ class yarp::dev::ServerSoundGrabber, is a network wrapper.
[INFO]Device "pipe", C++ class yarp::dev::DevicePipe, has no network wrapper
[INFO]Device "group", C++ class yarp::dev::DeviceGroup, has no network wrapper
[INFO]Device "remote_controlboard", C++ class yarp::dev::RemoteControlBoard, wrapped by "controlboardwrapper2"
[INFO]Device "inertial", C++ class yarp::dev::ServerInertial, is a network wrapper.
[INFO]Device "controlboardwrapper2", C++ class yarp::dev::ControlBoardWrapper2, is a network wrapper.
[INFO]Device "batteryClient", C++ class batteryClient, has no network wrapper
[INFO]Device "analogsensorclient", C++ class AnalogSensorClient, has no network wrapper
[INFO]Device "analogServer", C++ class yarp::dev::AnalogWrapper, is a network wrapper.
[INFO]Device "virtualAnalogServer", C++ class yarp::dev::VirtualAnalogWrapper, is a network wrapper.
[INFO]Device "batteryWrapper", C++ class yarp::dev::BatteryWrapper, is a network wrapper.
[INFO]Device "Rangefinder2DWrapper", C++ class yarp::dev::Rangefinder2Dwrapper, is a network wrapper.
[INFO]Device "RGBDSensorWrapper", C++ class yarp::dev::RGBDSensorWrapper, is a network wrapper.
[INFO]Device "RGBDSensorClient", C++ class yarp::dev::RGBDSensorClient, is a network wrapper.
[INFO]Device "controlboardremapper", C++ class yarp::dev::ControlBoardRemapper, wrapped by "controlboardwrapper2"
[INFO]Device "remotecontrolboardremapper", C++ class yarp::dev::RemoteControlBoardRemapper, wrapped by "controlboardwrapper2"
[INFO]Device "robotDescriptionServer", C++ class RobotDescriptionServer, has no network wrapper
[INFO]Device "robotDescriptionClient", C++ class RobotDescriptionClient, has no network wrapper
[INFO]Device "grabberDual", C++ class yarp::dev::ServerGrabber, is a network wrapper.
[INFO]Device "JoypadControlServer", C++ class yarp::dev::JoypadControlServer, is a network wrapper.
[INFO]Device "JoypadControlClient", C++ class yarp::dev::JoypadControlClient, is a network wrapper.
[INFO]Device "transformServer", C++ class yarp::dev::FrameTransformServer, is a network wrapper.
[INFO]Device "transformClient", C++ class transformClient, has no network wrapper
[INFO]Device "locationsServer", C++ class yarp::dev::LocationsServer, is a network wrapper.
[INFO]Device "Rangefinder2DClient", C++ class Rangefinder2DClient, has no network wrapper
[INFO]Device "navigation2DClient", C++ class navigation2DClient, has no network wrapper
[INFO]Device "localization2DClient", C++ class localization2DClient, has no network wrapper
[INFO]Device "map2DClient", C++ class yarp::dev::Map2DClient, has no network wrapper
[INFO]Device "map2DServer", C++ class yarp::dev::map2DServer, is a network wrapper.
[INFO]Device "AravisGigE", available on request (found in /usr/local/lib/yarp/AravisGigE.so library), wrapped by "grabber".
[INFO]Device "opencv_grabber", available on request (found in /usr/local/lib/yarp/yarp_opencv.so library), wrapped by "grabber".
[INFO]Device "test_grabber", available on request (found in /usr/local/lib/yarp/yarp_test_grabber.so library), wrapped by "grabber".

David-Estevez, 2017-11-13:
Prueba con este comando (como el tuyo, pero sin la barra `/` )

```
yarpdev --device AravisGigE --name /grabber --fake
```

err8029, 2017-11-13:
ok este comando va.. no entiendo

David-Estevez, 2017-11-13:
El nombre del dispositivo es `AravisGigE`, no lleva barra. Lo que lleva barra son los nombres de los puertos, pues siguen una nomenclatura similar a las urls o los directorios de sistemas UNIX.

err8029, 2017-11-13:
ahora lo vi en la documentación xd , nose porque la puse

closed 2017-11-13 by David-Estevez