roboticslab-uc3m/yarp-devices #134: Travis CI broken for current yarp-devices

jgvictores, 2017-11-15:
Travis CI broken for current yarp-devices, as seen [here](https://travis-ci.org/roboticslab-uc3m/yarp-devices/jobs/302419923#L451-L472).

Looks like some CMake PPA issue:
```
sudo add-apt-repository --yes ppa:kalakris/cmake
gpg: keyring `/tmp/tmpfv9sg8ol/secring.gpg' created
gpg: keyring `/tmp/tmpfv9sg8ol/pubring.gpg' created
gpg: requesting key 63844AC3 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmpfv9sg8ol/trustdb.gpg: trustdb created
gpg: key 63844AC3: public key "Launchpad PPA for Mrinal Kalakrishnan" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK

travis_time:end:3739f06a:start=1510742953520786359,finish=1510742955132785787,duration=1611999428
[0Ktravis_fold:end:before_install.1
[0Ktravis_fold:start:before_install.2
[0Ktravis_time:start:08b0755c
[0K$ sudo apt-get update -qq
W: http://dl.hhvm.com/ubuntu/dists/trusty/InRelease: Signature by key 36AEF64D0207E7EEE352D4875A16E7281BE7A449 uses weak digest algorithm (SHA1)
W: http://ppa.launchpad.net/couchdb/stable/ubuntu/dists/trusty/Release.gpg: Signature by key 15866BAFD9BCC4F3C1E0DFC7D69548E1C17EAB57 uses weak digest algorithm (SHA1)
W: The repository 'http://ppa.launchpad.net/kalakris/cmake/ubuntu trusty Release' does not have a Release file.
E: Failed to fetch http://ppa.launchpad.net/kalakris/cmake/ubuntu/dists/trusty/main/binary-amd64/Packages  404  Not Found
E: Failed to fetch http://ppa.launchpad.net/kalakris/cmake/ubuntu/dists/trusty/main/binary-i386/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.

travis_time:end:08b0755c:start=1510742955137307607,finish=1510742963673211793,duration=8535904186
[0K
[31;1mThe command "sudo apt-get update -qq" failed and exited with 100 during .[0m

Your build has been stopped.
```

PeterBowman, 2017-11-15:
We should get rid of kalakris' PPA: https://github.com/roboticslab-uc3m/questions-and-answers/issues/17#issuecomment-302940630.

jgvictores, 2017-11-16:
Brute-force attempt at 184c5514fbb6c18971078e77deb7f89f9f666f55

jgvictores, 2017-11-16:
Brute-force was at `fix-134-travis` branch. Apart from some warnings now commented at #137 we see Travis CI breaking at https://travis-ci.org/roboticslab-uc3m/yarp-devices/jobs/303127243 
```
[ 79%] Building CXX object libraries/YarpPlugins/WiimoteSensor/CMakeFiles/WiimoteSensor.dir/WiimoteDispatcherThread.cpp.o

/home/travis/build/roboticslab-uc3m/yarp-devices/libraries/YarpPlugins/WiimoteSensor/WiimoteDispatcherThread.cpp: In member function ‘virtual void roboticslab::WiimoteDispatcherThread::run()’:

/home/travis/build/roboticslab-uc3m/yarp-devices/libraries/YarpPlugins/WiimoteSensor/WiimoteDispatcherThread.cpp:47:67: error: ‘xwii_iface_dispatch’ was not declared in this scope

         int ret = xwii_iface_dispatch(iface, &event, sizeof(event));
```

@PeterBowman Looks like wiimote stuff, maybe you could take a look?

PeterBowman, 2017-11-16:
`xwii_iface_dispatch` was introduced in v2, which means that it'll work on Xenial, but not on Trusty:

* https://packages.ubuntu.com/trusty/libxwiimote-dev
* https://packages.ubuntu.com/xenial/libxwiimote-dev

According to docs in *xwiimote.h* (Xenial):

> This function is the successor or xwii_iface_poll(). It takes an additional `size` argument to provide backwards compatibility.

Docs and source code for both functions are almost identical. I'll fix this as soon as I can.

jgvictores, 2017-11-16:
Cool, thanks!

PeterBowman, 2017-11-17:
:v: 

> Done. Your build exited with 0.

jgvictores, 2017-11-20:
Closed via bcdb7d7d324790c0cf1a1047ce1e3d20c659eab1, opened #138 to keep up with hack (using deprecated xwii function).

jgvictores, 2017-11-20:
@PeterBowman BTW, thanks a lot for 6735632f672ed0da20cdaad33c5460388ae81239 !!

closed 2017-11-20 by jgvictores