roboticslab-uc3m/yarp-devices #138: Using xwii depreciated function

jgvictores, 2017-11-20:
To solve #134, 6735632f672ed0da20cdaad33c5460388ae81239 was introduced.
1. If we continue with old distros (see https://github.com/roboticslab-uc3m/teo-main/issues/26 and related), see if we can catch xwii version via preprocessor and `#ifdef`accordingly.
1. If we move on to newer distros on all systems, move on to newer function again.

PeterBowman, 2017-11-20:
I could either parse the version string from `/usr/lib/x86_64-linux-gnu/pkgconfig/libxwiimote.pc` (not the best idea if the location/format of this file differs from one architecture/OS to another) or start playing with [`try_run()`](https://cmake.org/cmake/help/v3.0/command/try_run.html).

jgvictores, 2017-11-21:
In addition to `priority: low`, added `blocked` by https://github.com/roboticslab-uc3m/teo-main/issues/26 label. BTW, could add `interesting` label too. :smile:

PeterBowman, 2017-11-21:
I actually gave a try to the pkg-config solution mixed with find modules and seems promising, expect updates soon.

> not the best idea if the location/format of this file differs from one architecture/OS to another

This is not a big deal: `.pc` files should land in some standard location and CMake takes care of the rest with [` pkg_check_modules()`](https://cmake.org/cmake/help/v3.1/module/FindPkgConfig.html).

PeterBowman, 2017-11-22:
Should be fine at 0eb18f3, closing as done. I'll write down this page here as it explains quite a few useful things about find modules:

https://cmake.org/Wiki/CMake:How_To_Find_Libraries

closed 2017-11-22 by PeterBowman

jgvictores, 2017-11-23:
> Should be fine at 0eb18f3, closing as done. I'll write down this page here as it explains quite a few useful things about find modules:
>
> https://cmake.org/Wiki/CMake:How_To_Find_Libraries

Cool commit!

Also added the ref to the cmake wiki at https://github.com/asrob-uc3m/tutoriales/blob/master/software/programming/cmake.md

PeterBowman, 2017-12-31:
Another (and up-to-date) ref on writing nice find modules:

https://cmake.org/cmake/help/latest/manual/cmake-developer.7.html#a-sample-find-module

And how `FindXWiimote.cmake` looks now: [link](https://github.com/roboticslab-uc3m/yarp-devices/blob/fbb730a60ac4f7e65e4e2afc8d0f59e11c6b8174/cmake/find-modules/FindXWiimote.cmake).

BTW [The Difference Between Deprecated, Depreciated and Obsolete (SO)](https://stackoverflow.com/q/9208091).

jgvictores, 2018-01-15:
> Another (and up-to-date) ref on writing nice find modules:
> 
> https://cmake.org/cmake/help/latest/manual/cmake-developer.7.html#a-sample-find-module

Thanks! Updated [here](https://github.com/asrob-uc3m/tutoriales/commit/1939c0448035938a34df4010602b585f7272c8b9)!