roboticslab-uc3m/yarp-devices #140: More intuitive camera control via RPC

David-Estevez, 2017-11-21:
Currently the RPC camera controls require to remember (or to look for) a number that represents a certain property of the camera. For instance, the command to change the zoom is:

```
fgc2 set feat 16 50
``` 

Where 16 is the property "ZOOM" and 50 is the value to be set. I think it would be much more intuitive if we could use some vocabs instead of the raw numbers:

```
fgc2 set feat zoom 50
```

Other option would be to have a working "help" command, similar to the one that the `FrameGrabberControlsInterface`  (1, not 2) has.

jgvictores, 2017-11-21:
Once we solve #108 we can add some info there.

The current YARP-ish way recommends NOT using RPC and instead use the interface. IMHO, the most practical way for that is to use Python bindings+`ipython`. For this we need a good [example/python](https://github.com/roboticslab-uc3m/yarp-devices/tree/bcdb7d7d324790c0cf1a1047ce1e3d20c659eab1/example/python) folder. I'm working on that at #128 (have some good stuff at asibot-main and elsewhere).

Hope that helps!

David-Estevez, 2017-11-27:
Due to some reason I still don't comprehend, I did not see your comment 6 days ago... 

It is exactly what I needed for this issue and for https://github.com/roboticslab-uc3m/horus/issues/11, thanks a lot!!

jgvictores, 2017-11-27:
:-)

David-Estevez, 2017-12-07:
Since #147 we have two new different ways of controlling the AravisGigE device:

* A beautiful GUI  (very intuitive)
* Bindings for Python (and other languages) that can access all the camera features from the corresponding enum (less intuitive than the GUI, but more intuitive than the RPC interface).

Since we have those two options available, I think there's no need to add support to new vocabs or help  in the RPC interface. Maybe this can be closed by simply adding an example on how to use the `IFrameGrabberControls2` interface from the Python console....

What do you think, @jgvictores ?

jgvictores, 2017-12-08:
All good, I've updated the "Improve examples" issue #128 and closing!
@David-Estevez Thanks!!

closed 2017-12-08 by jgvictores