roboticslab-uc3m/yarp-devices #141: Support more GigE parameters

David-Estevez, 2017-11-21:
Current support for GigE camera parameters is very limited. The basic parameters (gain, exposure, FPS, etc) should be supported.

err8029, 2017-11-23:
Documentacióm detección de agujeros

Objetivo: detectar ajugueros en un trozo de tela con o sin textura y con posibles arrugas

Implementción: desarrollado con el sofware GRIP y usando las librerias openCV

Explicación del funcionamiento: 
El programa se constituye de 3 bloques
1.-Adquisición de imagen de la camara: este bloque nos proporciona una imagen constante de 		video en su salida
2.-Función CVerode: la erosión consiste en superponer un kernel (por defecto, o dado) a 	los bodes detectados. Con esto conseguimos que las diferencias de colores se acentuen. En 		GRIP se encuentran los siguientes parametros para este bloque:
	-input: se corresponde a la salida del bloque 1
	-iterations: el numero de veces que se aplica el algoritmo propuesto por el bloque (9)
	-border type: el tipo de borde que aplicamos a la erosion (BORDER_CONSTANT)
3.-Funcion find blobs: este bloque nos permite, encontrr grupos de pixeles con un color 	similar y, en una nueva imagen, posicionar un circulo donde estos esten. GRIP presenta los 		siguientes parametros.

PeterBowman, 2023-02-08:
Closing due to inactivity.