roboticslab-uc3m/yarp-devices #145: Simple GUI to control camera parameters with YARP

David-Estevez, 2017-11-28:
From https://github.com/roboticslab-uc3m/horus/issues/11:

>  Implement a quick and dirty GUI to control camera parameters such as zoom with YARP.

where  @jgvictores suggested:

>   *  Using a RemoteFrameGrabber instead of a rpcClient to connect to the device
>   *  Developing this code within yarp-devices, folder programs (by now, might be moved in the future)

Related to #140 , where @jgvictores said:

> Once we solve #108 we can add some info there.
> 
> The current YARP-ish way recommends NOT using RPC and instead use the interface. IMHO, the most > practical way for that is to use Python bindings+`ipython`. For this we need a good [example/python](https://github.com/roboticslab-uc3m/yarp-devices/tree/bcdb7d7d324790c0cf1a1047ce1e3d20c659eab1/example/python) folder. I'm working on that at #128 (have some good stuff at asibot-main and elsewhere).
>
> Hope that helps!

David-Estevez, 2017-11-30:
I've been several days immersed in a "situation" involving me not able to install yarp in python3 and cmake not wanting to cooperate. I've finally been able to make it work, but I've tried so many things that I don't really know which are the concrete steps that made it work. Nonetheless, I am leaving here some steps / operations that I believe contributed to solving this "situation":

* Use swig 3 instead of swig 2 (This solves some compilation problems)
* Use master branch in yarp
* To enable python bindings (with ccmake):
    1. Set YARP_COMPILE_BINDINGS to ON, configure
    2. Set CREATE_PYTHON to ON, configure
    3. Set YARP_USE_PYTHON_VERSION to 3.4, configure
* At this point, my cmake (3.4) is unable to find python 3.4, so I exit ccmake and run the following command:
   ```
   cmake .. -DPYTHON_INCLUDE_DIR=$(python3 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())")  -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.4m.so -DPYTHON_EXECUTABLE=/usr/bin/python3 
  ```
In this command the python3 library and executable paths are hardcoded, and have to be modified if it is elsewhere.
* I use ccmake again to try to fix the PYTHON_INSTDIR setting it to `../lib/python3/dist-packages` since the install prefix is set to `/usr/local` and my python won't find the library there. But this seems to be ignored when yarp installs itself. placing the python lib at `/usr/local/lib/python3/dist-packages` so...
* ... I run these commands to create symbolic links to the yarp library:
  ```
  sudo ln -s /usr/local/lib/python3/dist-packages/yarp.py /usr/lib/python3/dist-packages/yarp.py
  sudo ln -s /usr/local/lib/python3/dist-packages/_yarp.so /usr/lib/python3/dist-packages/_yarp.so
  ```

jgvictores, 2017-11-30:
Thanks @David-Estevez I think that is a very accurate description of the situation. I've opened https://github.com/roboticslab-uc3m/installation-guides/issues/26 on the YARP + Python3 situation (where CMake and swig versions also play a role) pointing to here.

David-Estevez, 2017-11-30:
I commented here since they are not very precise instructions on how to install it, but I can move it to the installation-guides if you think it might help.

btw,  thanks a lot to @jgvictores and @RaulFdzbis for your help!! :smile:

jgvictores, 2017-11-30:
Nah, I think maybe we should wait a bit to see either a well-known workflow on a fresh install, or improvements upstream that provide it.

David-Estevez, 2017-11-30:
I've tried it also on teo-ebola (fresh 16.04) and I had to do most of those steps. Swig 3 was included by default in that version's repositories, but I still had to manually set all the paths and make the symlinks.

David-Estevez, 2017-11-30:
Testing performed in https://github.com/David-Estevez/yarp-devices/commit/29134ce9f3aeca27a6b38a29ad1e0c38977cc148 revealed that currently there are no bindings to use the FrameGrabberControls interface of the remote_grabber.

@jgvictores gave me some guidelines to add support for this to yarp, more specifically:

https://github.com/robotology/yarp/blame/master/bindings/yarp.i#L460

and 

https://github.com/robotology/yarp/blame/master/bindings/yarp.i#L810

David-Estevez, 2017-12-01:
I've added view functions for both IFrameGrabberControls2 and FrameGrabberControls2_Sender but both of them return the same error:

```
error while opening /usr/local/lib/yarp/yarp_priority.so:
  /usr/local/lib/yarp/yarp_priority.so: undefined symbol: _ZTIN4yarp2os4impl27PortReaderBufferBaseCreatorE
error while opening /usr/local/lib/yarp/yarp_xmlrpc.so:
  /usr/local/lib/yarp/yarp_xmlrpc.so: undefined symbol: xmlrpc
```

And then the program ends with a Segmentation Fault.

Any clues about what is happening here @jgvictores ?

jgvictores, 2017-12-01:
There is an `xmlrpc` **optional** YARP carrier for compatibility with ROS (specifically, communicates with `ros core`). You probably have some old `.ini` in some `.../local/share/` folder which is telling the YARP plugin mechanism to look for the `.so`. Finding and erasing the `.ini` should fix the problem (or, compiling and installing the carrier).

Same should apply with the `priority` library, that also looks like a plugin.

David-Estevez, 2017-12-01:
Ok, I've deleted those files:
```
 sudo rm /usr/local/share/yarp/plugins/xmlrpc.ini 
 sudo rm /usr/local/share/yarp/plugins/priority.ini
```

The error message disappears, but the program still ends with a segmentation fault.

David-Estevez, 2017-12-01:
I think the methods that make the program crash are the ones that have a pointer in their arguments. Could this be the case?

For instance, the method:
```    
bool hasFeature(IFrameGrabberControls2 self, int feature, bool * hasFeature)
```
makes python crash, whereas the method
```
bool setFeature(IFrameGrabberControls2 self, int feature, double value)
```
seems to work.  Do I have to provide adapters for the c++ methods that use pointers in `yarp.i`?

jgvictores, 2017-12-01:
Yes, possibly you'll have to do something like [this](https://github.com/robotology/yarp/blob/1c0c7a49328cc84ba565e4faf4a0db81d1cceeaf/bindings/yarp.i#L861-L866).

David-Estevez, 2017-12-01:
Yeah, inspecting that file I can see this:
```
//////////////////////////////////////////////////////////////////////////
// Deal with ControlBoardInterfaces pointer arguments that don't translate

%extend yarp::dev::IPositionControl {
    int getAxes() {
        int buffer;
        bool ok = self->getAxes(&buffer);
        if (!ok) return 0;
        return buffer;
    }
```

Which makes me believe you are right and I have to provide my own pointer-free implementations for those methods.

David-Estevez, 2017-12-01:
Current state:

* GUI finished and working (8cd8b6c12aadb069529c746bcb360bf71c2ffcfb)
* GUI app crashes when tested with real camera (backend calls maybe?) (80306f8386723ea076bb103936351050669e95b1)
* YARP bindings support IFrameGrabberControls2, but not completely (https://github.com/David-Estevez/yarp/commit/5d624bb41e92d181366665fa69301f2b1c9a72f9)

Things left to do:

* [ ] Find out why the GUI is not working with real camera and fix it
* [ ] Add support to (at least) has, get and set commands in IFrameGrabberControls2
* [ ] Setup everything to be able to install this app (script+module+ui description file)

David-Estevez, 2017-12-01:
Basic support for has and get at https://github.com/David-Estevez/yarp/commit/f8603b9

David-Estevez, 2017-12-04:
> Find out why the GUI is not working with real camera and fix it

The GUI wasn't working because I was not storing the `PolyDriver` as a class attribute, just the `IFrameGrabberControls2` interface. The garbage collector deleted the `PolyDriver` object as soon as the init() member was finished, and that is the reason the app crashed with a Segmentation Fault.

Fixed at https://github.com/David-Estevez/yarp-devices/commit/212ca947b2f8c20499a7f52789460e751b01f0e7.

> Add support to (at least) has, get and set commands in IFrameGrabberControls2

I've added support to most of the members of the interface. Stuff I'm not supporting:

- Returning an exception / different value in case of error (Errors are not being checked).
- Supporting functions returning two values (since they have the same signature, we would have to come up with a new name for those functions, i.e. `getDualFeature()` plus dealing with returning a tuple).

Done at  https://github.com/David-Estevez/yarp/commit/f8603b9bd36d9773fb11e869c02f8f5e9b832380, PR pending!

> Setup everything to be able to install this app (script+module+ui description file)

Not done yet, I'm focusing on that now.

David-Estevez, 2017-12-04:
(Edited previous comment with right commit)

David-Estevez, 2017-12-04:
Reading the [SWIG docs](http://www.swig.org/Doc1.3/Arguments.html), it seems to be a way to automatically tag all the functions that need type conversion due to the pointers with a function that performs that conversion:

```cpp
%module example
%include "typemaps.i"

%apply double *OUTPUT { double *result };

%inline %{
extern void add(double a, double b, double *result);
extern void sub(double a, double b, double *result);
extern void mul(double a, double b, double *result);
extern void div(double a, double b, double *result);
%}
```

And this works even with several return values:

```cpp
%include "typemaps.i"
%apply int *OUTPUT { int *width, int *height };

// Returns a pair (width,height)
void getwinsize(int winid, int *width, int *height);
```

Which gives this output:
``` 
>>> w,h = genwinsize(wid)
>>> print w
400
>>> print h
300
>>>
```

But adding that to yarp, as interesting as it may seem, requires a time investment that I think it is not worth it right now.

jgvictores, 2017-12-04:
@David-Estevez That's interesting! Might use in future developments.

Even though you're not implementing that solution, can't wait to see the PR in YARP. :-)

David-Estevez, 2017-12-04:
PR for yarp code here: https://github.com/robotology/yarp/pull/1483

jgvictores, 2017-12-07:
This is good news! @David-Estevez Congratulations! :tada:

David-Estevez, 2017-12-07:
With #147 merged and https://github.com/robotology/yarp/pull/1483 already accepted and merged we can consider this issue done.

closed 2017-12-07 by David-Estevez