roboticslab-uc3m/yarp-devices #146: Create new PCAN device plugin

PeterBowman, 2017-11-29:
We'll soon get a PCAN-miniPCI to play with ([IPEH-003045](https://www.peak-system.com/PCAN-miniPCI.205.0.html)). Once the hardware is tested, a new YARP plugin should be created to manage communications between the CAN master and each node (compare with [CanBusHico](https://github.com/roboticslab-uc3m/yarp-devices/tree/9b92f99eb99d6de553dbf1ed748cfa5ccef65556/libraries/YarpPlugins/CanBusHico)).

See also: [manufacturer's website](https://www.peak-system.com/fileadmin/media/linux/index.htm) ([user manual and API](https://www.peak-system.com/fileadmin/media/linux/index.htm#download)).

PeterBowman, 2017-11-29:
Might consider closing #122 first. Also related: #74 (proposed `ICanOpen` interface).

PeterBowman, 2017-12-13:
The kernel-space software behind PCAN boards is SocketCAN, a widespread implementation of the CAN protocol available on modern Linux distros. This actually simplifies things since we are not bound to a hardware-specific API and can reuse available code. In fact, I found a YARP device in the icub-main repo: [*socketCan*](https://github.com/robotology/icub-main/tree/6ac5a5c06d77ac19c95219fe5560b5db43a9a85b/src/libraries/icubmod/socketCan). Alternatives:

1. Write our own implementation. One would be tempted to copy-paste, but care must be taken because of license restrictions (GPL).
2. Tell users to download and compile the *socketCan* device from icub-main. In other words, this adds a runtime dependency on all CAN-based devices from this repo.

jgvictores, 2017-12-14:
Since it is a YARP device plugin, I'm not sure if GPL acts virally.

More importantly, unless it is GPL due to some linkage to some other GPL that acts virally, I believe we can contact the author and see if he can re-license it to be LGPL'ed.

PeterBowman, 2017-12-23:
As spoken with @munozyanez, two YARP devices may emerge from this issue:

* a SocketCan implementation, roughly similar to [icub-main's socketCan device](https://github.com/robotology/icub-main/tree/a7979a8aa318261fed26d0f215d20abee5d8523b/src/libraries/icubmod/socketCan) (hence my doubts about licensing, but I'd like to expand it with goodies such as acceptance filtering) to interface with the new board through sockets (*netdev* mode)
* a PeakCan implementation, following the user guide on the CAN FD API, to interface with the board through a character device (*chardev* mode)

The former looks more straightforward to achieve, useful examples may be found at [munozyanez/socketcan-examples](https://github.com/munozyanez/socketcan-examples).

PeterBowman, 2017-12-24:
@jgvictores HicoCAN headers are currently copy-pasted into the [CanBusHico device directory](https://github.com/roboticslab-uc3m/yarp-devices/tree/8be3065e6529536826e22aac41ff2e163a41d157/libraries/YarpPlugins/CanBusHico), that is, `hico_api.h` does not need to be installed on the target system. I might proceed in a similar way with PeakCan, which requires two header files (`pcanfd.h` and `libpcanfd.h`); however, I'll probably make use of `find_path()` and/or create a new find module to stay closer to the usual CMake practice. I'd like to ask you for your opinion on this and whether `hico_api.h` could/should be treated the same way.

jgvictores, 2017-12-24:
@PeterBowman :+1:

PeterBowman, 2017-12-27:
Note to self: blocking read/writes with sockets and `setsockopt`: https://ubuntuforums.org/showthread.php?t=1565651&p=9794579#post9794579.

Also, an useful column about I/O calls: http://www.linux-mag.com/id/308/.

PeterBowman, 2017-12-30:
I'll probably go for Linux only by now, but let me stress that Peak offers a common API for use in both Windows and Linux (plus bindings and sample GUI apps, see https://github.com/roboticslab-uc3m/yarp-devices/issues/85#issuecomment-354543098). List of supported devices from [PCANBasic.h](https://www.peak-system.com/PCAN-miniPCI.205.0.html?&L=1#Downloads):

```c
#define PCAN_NONE                    0x00U  // Undefined, unknown or not selected PCAN device value
#define PCAN_PEAKCAN                 0x01U  // PCAN Non-Plug&Play devices. NOT USED WITHIN PCAN-Basic API
#define PCAN_ISA                     0x02U  // PCAN-ISA, PCAN-PC/104, and PCAN-PC/104-Plus
#define PCAN_DNG                     0x03U  // PCAN-Dongle
#define PCAN_PCI                     0x04U  // PCAN-PCI, PCAN-cPCI, PCAN-miniPCI, and PCAN-PCI Express
#define PCAN_USB                     0x05U  // PCAN-USB and PCAN-USB Pro
#define PCAN_PCC                     0x06U  // PCAN-PC Card
#define PCAN_VIRTUAL                 0x07U  // PCAN Virtual hardware. NOT USED WITHIN PCAN-Basic API
#define PCAN_LAN                     0x08U  // PCAN Gateway devices
```

PeterBowman, 2018-06-04:
https://github.com/roboticslab-uc3m/yarp-devices/issues/146#issuecomment-353739123

> As spoken with @munozyanez, two YARP devices may emerge from this issue:
>
> * a SocketCan implementation, roughly similar to [icub-main's socketCan [...]

Let me correct myself. The **same** CAN (FD) API offered by PEAK-System can be compiled either in chardev or netdev mode. The former performs the usual system calls (`ioctl`, `read`...), the latter resorts to including Linux headers from the SocketCan implementation (e.g. `<linux/can/...>`). Since modern kernels natively support most PC CAN devices, applications may simply open a socket in order to talk with them, that is, there is no need to compile and install the API. For the same reason, the netdev interface has been kept solely for backward compatibility, while the chardev one is made default when building the driver. Speaking of YARP devices, *CanBusPeak* will perform API calls through PEAK's interfaces (no matter what mode, chardev or netdev, since this is totally transparent to the plugin) while *CanBusSocket* should work with whatever SocketCan-enabled CAN devices, even those integrated in the AMOR robot.

PeterBowman, 2018-07-09:
Reference: [*Bit Time Requirements for CAN FD*](https://github.com/roboticslab-uc3m/yarp-devices/files/2176063/2013_hartwich_v2.pdf).

PeterBowman, 2018-07-19:
This issue will be closed after https://github.com/roboticslab-uc3m/yarp-devices/pull/190 is merged into develop. Only the character device (CanBusPeak) is considered for now due to upcoming hardware renewal on TEO. The SocketCAN device (CanBusSocket) will come later.

closed 2018-07-25 by PeterBowman

PeterBowman, 2020-06-13:
> The SocketCAN device (CanBusSocket) will come later.

See https://github.com/CANopenNode/CANopenSocket.

munozyanez, 2020-06-13:
Wow! I love it!
Maybe I can lend a hand after PhD viva.