roboticslab-uc3m/yarp-devices #153: Incorporate pulse/sample parameter

jgvictores, 2017-12-14:
We are currently counting on a fixed pulse/sample value (`0.001` (?) for velocities and thus `0.001^2` for accelerations):
- https://github.com/roboticslab-uc3m/yarp-devices/blob/4b2c591d07a857255cee07ce4a0dd675264b80aa/libraries/YarpPlugins/TechnosoftIpos/IVelocityControl2RawImpl.cpp#L25
- https://github.com/roboticslab-uc3m/yarp-devices/blob/4b2c591d07a857255cee07ce4a0dd675264b80aa/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L743
- https://github.com/roboticslab-uc3m/yarp-devices/blob/4b2c591d07a857255cee07ce4a0dd675264b80aa/libraries/YarpPlugins/TechnosoftIpos/ICanBusSharerImpl.cpp#L761

This looks dangerous and undocumented!

Incorporate a pulse/sample parameter, such as `pulsePerSample` or better name?

PeterBowman, 2017-12-14:
> Incorporate a pulse/sample parameter, such as pulsePerSample or better name?

`samplesPerPulse` and you get rid of fractions 😏

jgvictores, 2017-12-22:
Related: [yarp::dev::IMotorEncoders::setMotorEncoderCountsPerRevolution](http://www.yarp.it/classyarp_1_1dev_1_1IMotorEncoders.html#a2e6b12f99b8e1f00619e876cc39ef051)

PeterBowman, 2019-07-29:
Sample loop is 1 ms default, see e.g. docs for object 6081h (profile velocity). Regarding object 207Ah (interpolated position 1st order time), it is stated that time "is given in IU which is by default 0.8ms for steppers and 1ms for the other configurations". We don't care about steppers, I'm just curious whether the sample loop refers to that same time dimension in both objects.

PeterBowman, 2019-08-19:
> `samplesPerPulse` and you get rid of fractions smirk

Makes more sense the other way around: `pulsesPerSample`.

> Regarding object 207Ah (interpolated position 1st order time), it is stated that time "is given in IU which is by default 0.8ms for steppers and 1ms for the other configurations".

That is, we want to sample a given amount of pulses per one second. Assuming we already translated a `degrees/seconds` input into `pulses/seconds` (by applying the reduction factor and taking into consideration the number of encoder lines and pulses per line):

`pulses/second * (samples * second)/pulse`

I might be wrong in my interpretation regarding time units; however, the maths are clear and 1 millisecond as stated in the manual translates into 1000 pulses per one `(sample*second)` in my previous formula. For steppers, that would be 1250. And I like this more than fractions.

PeterBowman, 2019-08-19:
Ready at https://github.com/roboticslab-uc3m/yarp-devices/commit/36b636052cc1eb89b7bb90c71b9407914123c91b.

closed 2019-12-31 by PeterBowman