roboticslab-uc3m/yarp-devices #154: Unsupported protocol error when CMaking yarp-devices

David-Estevez, 2017-12-15:
I've updated my fork with the new advances in develop and now I cannot CMake it. The error returned is the following one:

```
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - ERROR 1: "Unsupported protocol"
-- Downloading YCMEPHelper.cmake - Attempt 2 of 3
-- Downloading YCMEPHelper.cmake - ERROR 1: "Unsupported protocol"
-- Downloading YCMEPHelper.cmake - Attempt 3 of 3
CMake Error at cmake/IncludeUrl.cmake:348 (message):
  Downloading YCMEPHelper.cmake - ERROR 1: "Unsupported protocol"
Call Stack (most recent call first):
  cmake/YCMBootstrap.cmake:105 (include_url)
  CMakeLists.txt:77 (include)
```

Since I only have enabled my own devices (`AravisGigE` and `GrabberControls2Gui`) and as far as I know I'm not using any external dependencies that require YCM, I have no clue why I'm getting this error.  So I summon all the YCM experts around to help me with this issue.

[image]

PeterBowman, 2017-12-15:
@David-Estevez it totally looks like a lack of SSL support. Which distro are you on? See [install-cmake.md#openssl-support](https://github.com/roboticslab-uc3m/installation-guides/blob/707d6cf05d5a86c9b78cc26b8b1e18b1127037ef/install-cmake.md#openssl-support) and https://github.com/roboticslab-uc3m/yarp-devices/issues/26#issuecomment-350398178.

David-Estevez, 2017-12-15:
This error happens in both Linux Mint 17.3 and Ubuntu Ubuntu 16.04.3 LTS.

I didn't follow that guide because I do not require YCM for my devices. I guess my questions are:

* Do I need YCM for something I'm not aware of? (something at whole repo level)
* **If** I do not use YCM... should I enforce it as an unused dependency on my devices?

PeterBowman, 2017-12-15:
What CMake version are you using (`cmake --version`)? Did you compile it from sources? See https://github.com/robotology/codyco-superbuild/issues/26#issuecomment-61173419.

We adopted YCM as a org-wide dependency (see kinematics-dynamics, openrave-yarp-plugins...). Here, it provides custom CMake modules (e.g. AddUninstallTarget), enhanced standard modules for legacy CMake (e.g. CMakePackageConfigHelpers), and automated handling of external dependencies (ColorDebug).

David-Estevez, 2017-12-15:
In Linux Mint 17.3 -> cmake version 3.4.0
In Ubuntu 16.04.3 LTS -> cmake version 3.5.1

Since I tried lots of things to get Python 3 bindings to compile (details at https://github.com/roboticslab-uc3m/yarp-devices/issues/145#issuecomment-348230791) I might have compiled cmake from source without openssl support. 

I'll check that in both PCs and report here.

David-Estevez, 2017-12-18:
Errors seem to be gone following your instructions ([install-cmake.md#openssl-support](https://github.com/roboticslab-uc3m/installation-guides/blob/707d6cf05d5a86c9b78cc26b8b1e18b1127037ef/install-cmake.md#openssl-support)), so recompiling cmake was probably something I tried to get the Python 3 yarp bindings to work. 

Nonetheless, to consider this issue close I would like to add some notice in the installation instructions of this repo to remind that openSSL support is required for YCM to work.

PeterBowman, 2017-12-18:
> Nonetheless, to consider this issue close I would like to add some notice in the installation instructions of this repo to remind that openSSL support is required for YCM to work.

This is something we would have to add in most repos, though. Shall we make the openSSL advice more prominent at [install-ycm.md](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-ycm.md)?

David-Estevez, 2017-12-18:
I think my concern was that some devs/users might not notice that they need YCM even if they are not using it explicitly (as it happened with myself).

I see you have recently pushed 638ac5d1eedab17bb838e340c7d9b3a3a778a2ba that solves that concern, so now we can consider this issue closed.

closed 2017-12-18 by David-Estevez