roboticslab-uc3m/yarp-devices #157: Think about implementing additional motor/axis yarp::dev interfaces

jgvictores, 2017-12-22:
Think about implementing additional motor/axis `yarp::dev` interfaces:
- http://www.yarp.it/classyarp_1_1dev_1_1IMotor.html
- http://www.yarp.it/classyarp_1_1dev_1_1IMotorEncoders.html
- http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html

jgvictores, 2017-12-22:
Related: #153

jgvictores, 2017-12-22:
Related: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/55

PeterBowman, 2019-07-29:
> http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html

Might be blocked by https://github.com/roboticslab-uc3m/teo-main/issues/44 (renaming joints).

PeterBowman, 2019-09-28:
Brief summary of what is currently implemented (✔️), what is not (❌), and what is covered in the scope of this issue (bold text) as of [latest `apocanlypse` commit](https://github.com/roboticslab-uc3m/yarp-devices/tree/cfe764c8a7ee6c665535b4543b9a345a932f1d90/libraries/YarpPlugins/TechnosoftIpos):

* ❌ `yarp::dev::IAmplifierControlRaw`
* ❌ **`yarp::dev::IAxisInfoRaw`**
* ❌ `yarp::dev::IControlCalibrationRaw`
* ✔️ `yarp::dev::IControlLimitsRaw`
* ✔️ `yarp::dev::IControlModeRaw`
* ✔️ `yarp::dev::ICurrentControlRaw`
* ✔️ `yarp::dev::IEncodersTimedRaw`
* ❌ `yarp::dev::IImpedanceControlRaw`
* ❌ `yarp::dev::IInteractionModeRaw`
* ❌ **`yarp::dev::IMotorRaw`**
* ❌ **`yarp::dev::IMotorEncodersRaw`**
* ❌ `yarp::dev::IPidControlRaw`
* ✔️ `yarp::dev::IPositionControlRaw`
* ✔️ `yarp::dev::IPositionDirectRaw`
* ❌ `yarp::dev::IPWMControlRaw`
* ✔️ `yarp::dev::IRemoteVariablesRaw`
* ✔️ `yarp::dev::IVelocityControlRaw`
* ✔️ `yarp::dev::ITorqueControlRaw`

PeterBowman, 2019-10-02:
Ready at https://github.com/roboticslab-uc3m/yarp-devices/commit/8fa67e59052ebb1373d48a88e671b0df3f6414bb. With this commit, I'm also assuming that both encoder and current reads are meant to be queried via TPDO and therefore stored in an internal buffer (https://github.com/roboticslab-uc3m/yarp-devices/issues/232#issuecomment-532871260). Also, reverse encoder count is configured via dedicate YARP .ini option.

closed 2019-12-31 by PeterBowman