roboticslab-uc3m/yarp-devices #159: Replace hardcoded/arbitrary delays with a dual semaphore queue

PeterBowman, 2017-12-23:
Current CAN bus communications rely on blocking read operations with a hardcoded 1 ms delay. Following the schema of [CanBusControlboard::ThreadImpl](https://github.com/roboticslab-uc3m/yarp-devices/blob/f2f659ec3c4575bb691799ba9948d494bdf0c01d/libraries/YarpPlugins/CanBusControlboard/ThreadImpl.cpp), the read loop behaves like this:

```c++
while (isRunning)
{
    // start listening to incoming traffic with a timeout of 'delay' milliseconds
    bool ok = read_timeout(msg_buffer, delay);

    if (!ok)
    {
        // error, timeout or EOF, loop again
        continue;
    }

    // do something with the received data
    interpretMessage(msg_buffer);
}
```

On the CAN node side, requests for data retrieval apply the same time value for delays while read operations loop constantly on another thread, as shown above (hopefully expecting that the next incoming message would refer to the exact same information previously requested):

https://github.com/roboticslab-uc3m/yarp-devices/blob/f2f659ec3c4575bb691799ba9948d494bdf0c01d/libraries/YarpPlugins/TechnosoftIpos/IPositionControl2RawImpl.cpp#L283-L296

Another approach was taken in [icub-main's sharedCan device](https://github.com/robotology/icub-main/tree/a7979a8aa318261fed26d0f215d20abee5d8523b/src/libraries/icubmod/sharedCan). It employs a pair of closely cooperating semaphores (look for `waitReadMutex` and `synchroMutex`), which are mutually exclusive (one of them initially set to zero), and non-blocking read calls. Seems cleaner, perhaps even more efficient.

In addition, advantage could be taken of the new message buffering feature coming soon in #122. For the time being, these buffers will start with a size of 1.

PeterBowman, 2018-07-16:
Interesting lecture: https://stackoverflow.com/q/62814.

jgvictores, 2018-07-16:
1. From what I've read in the past, `yarp::os::Mutex` is essentially a `bool`, so you can only involve two actors. `yarp::os::Semaphore` is more like an `int` so you can have a full queue of actors.

2. ` yarp::os::Time::delay(DELAY);  //-- Wait for read update. Could implement semaphore waiting for specific message...` is a bit complex to implement. At least as proposed, we'd almost need one semaphore/mutex ...for each type of message!

PeterBowman, 2018-08-20:
> Another approach was taken in icub-main's sharedCan device. It employs a pair of closely cooperating semaphores (look for waitReadMutex and synchroMutex), which are mutually exclusive (one of them initially set to zero), and non-blocking read calls.

For future reference, the dual semaphore construct is only used in two places: [ref1](https://github.com/robotology/icub-main/blob/c3ad642ed0aacdb3216f692cc73916936079714b/src/libraries/icubmod/sharedCan/SharedCanBus.h#L94-L98), [ref2](https://github.com/robotology/icub-main/blob/c3ad642ed0aacdb3216f692cc73916936079714b/src/libraries/icubmod/sharedCan/SharedCanBus.h#L120-L126).

PeterBowman, 2018-08-22:
Perhaps this SO answer is closer to my original intent: https://stackoverflow.com/a/47614705.

PeterBowman, 2018-09-19:
We had to increase tenfold the hardcoded 1 ms delay in order to correctly fetch the actual motion status from the iPOS:

https://github.com/roboticslab-uc3m/yarp-devices/blob/9e7c30923bc399dcf282aa1501ef3aaf6f9cd8df/libraries/YarpPlugins/TechnosoftIpos/IPositionControl2RawImpl.cpp#L158

The default value was causing trouble in our [teo-self-presentation demo](https://github.com/roboticslab-uc3m/teo-self-presentation) due to `IPositionControl::checkMotionDone` not updating an internal flag with the actual motion status fetched from the driver. This resulted in reporting a false state, either by finishing the motion sequence too early or by getting stuck in a "target not reached" mode.

PeterBowman, 2019-04-30:
Idea: TPDOs can be configured to emit messages at fixed intervals, and send them with higher frequencies in case the drive state which is being watched changes; see [3.6, *TxPDOs mapping example*](https://github.com/roboticslab-uc3m/yarp-devices/files/3131971/P091.063.CANopen.iPOS.UM.2019.pdf) (asynchronous transmission) and 10.3, *Cyclic Synchronous Position Mode example* (synchronous).

PeterBowman, 2019-08-05:
> In addition, advantage could be taken of the new message buffering feature coming soon in #122. For the time being, these buffers will start with a size of 1.

WIP at https://github.com/roboticslab-uc3m/yarp-devices/issues/209.

> Idea: TPDOs can be configured to emit messages at fixed intervals, and send them with higher frequencies in case the drive state which is being watched changes; see [3.6, _TxPDOs mapping [...]

To be discussed at https://github.com/roboticslab-uc3m/yarp-devices/issues/223.

PeterBowman, 2019-08-07:
Idea: implement a map of binary semaphores ([`yarp::os::Semaphore`](http://www.yarp.it/classyarp_1_1os_1_1Semaphore.html)) with timeout ([`waitWithTimeout()`](http://www.yarp.it/classyarp_1_1os_1_1Semaphore.html#a80f68f77306c5b4eb91bb9720174b450)), instantiate a new semaphore on every SDO request (value), associate it to a SDO dictionary object index (key), release the semaphore either in the read thread upon receiving the acknowledge/response or when timeout expires.

PeterBowman, 2019-08-13:
Ready at https://github.com/roboticslab-uc3m/yarp-devices/commit/fec00cfd35e4a84d5697df0e25d6123f00fcdec4. All SDO read requests now poll an available semaphore and block until the response arrives or a timeout occurs. Additionally, most SDO writes await an acknowledge signal from the driver.

PeterBowman, 2019-08-20:
> All SDO read requests now poll an available semaphore

Assuming only one active connection can be handled by a single SDO channel, there is no need to poll anything. Commit https://github.com/roboticslab-uc3m/yarp-devices/commit/29b3e1ea145676aed328648eaf0121e2b080a90a simplified things a bit.

PeterBowman, 2019-09-17:
This same semaphore-wait-with-timeout thingy proved useful for non-SDO thingies. Look for a *StateObserver.hpp* file in https://github.com/roboticslab-uc3m/yarp-devices/commit/972f8758d246bbb2a85c40cf5af11bd5855597a0. Now, it is possible to signal an event to a observer in a more generic manner: no context data, context data is an integer-type (be it an array of `std::uint8_t`s or non-byte-length types), or context data is a class type.

closed 2019-12-31 by PeterBowman

PeterBowman, 2020-01-28:
> Assuming only one active connection can be handled by a single SDO channel, there is no need to poll anything.

Issue https://github.com/roboticslab-uc3m/yarp-devices/issues/160 allowed simultaneous SDO requests to occur. Commit https://github.com/roboticslab-uc3m/yarp-devices/commit/fd5dd7db9550c66ad3b383b7ba1e90f4f12528c2 guards against this and reports "overrun" errors whenever the current SDO client's observer registers a response/confirmation message a different SDO client was supposed to receive from a previously issued request/indication. The SDO handshake has been made foolproof with a triple check: object index and object subindex must match, and command specifier bits must agree with the expected upload/download protocol.