roboticslab-uc3m/yarp-devices #161: Migrate tables to ods/md

jgvictores, 2018-01-06:
The following markdown tables can become unmanagable, and are already difficult:
- [yarp-devices-install.md#os-requirements](https://github.com/roboticslab-uc3m/yarp-devices/blob/4547168177750c06ce902dd9552341760cb8cb0f/doc/yarp-devices-install.md#os-requirements)
- [yarp-devices-install.md#specific-dependencies](https://github.com/roboticslab-uc3m/yarp-devices/blob/4547168177750c06ce902dd9552341760cb8cb0f/doc/yarp-devices-install.md#specific-dependencies)

I'd follow an ods/md workflow as recommended [here](https://github.com/roboticslab-uc3m/teo-software-manual/blob/8789d4c4975bd5d00748ee1227ecb3b04851b5a3/README.md#workflow-for-assets) and implemented [here](https://github.com/roboticslab-uc3m/teo-software-manual/blob/8789d4c4975bd5d00748ee1227ecb3b04851b5a3/appendix/a-teo-diagrams.md#dh-parameter-table).

jgvictores, 2018-01-08:
Nice proposal by @David-Estevez is `.csv`, which can be imported, edited, and exported via libreoffice.
This is good, so we do not have to maintain two files.

David-Estevez, 2018-01-08:
This is what the new tables look like:
* https://github.com/David-Estevez/yarp-devices/blob/c43b936fcfbff70f239043bbaf0fce5fdd66ff7a/doc/specific-dependencies.csv
* https://github.com/David-Estevez/yarp-devices/blob/c43b936fcfbff70f239043bbaf0fce5fdd66ff7a/doc/os-requirements.csv

@jgvictores Could you please check them out and give me you feedback about them (like/dislike/change something)?

jgvictores, 2018-01-08:
Incredibly nice!
A few comments/questions:
1. Only works in desktop mode. Good for me, GitHub actually has a link in addition to the browser option.
1. Markdown links are obviously markdown links, and not rendered as links in csv. Good for me.
1. Obscure part for now: guess you know how the *,**,*** will work out?
1. Does it look like both tables would be less wide if transposed?

David-Estevez, 2018-01-08:
1. You can always see the page on desktop mode, I guess.
2. This point bothers me a little, but I see no clear solution.
3. I don't know what do you mean with this point :sweat_smile: 
4. They probably would, but the transposed table might not scale well with more dependencies (which is not a concern right now, but might be in the future). I can transpose them, no problem.

jgvictores, 2018-01-08:
1. Ok!
1. It's ok!
1. Auch. That didn't render as expected LOL. Anyway. For instance, in [specific-dependencies.csv](https://github.com/David-Estevez/yarp-devices/blob/c43b936fcfbff70f239043bbaf0fce5fdd66ff7a/doc/specific-dependencies.csv), `LeapMotionSensor` has an asterisk that seems intended for a footnote... This makes sense in the original context, but not sure how it works out here, and have no ideas for fixing it.
1. Yes, please, transposing would be great!

David-Estevez, 2018-01-08:
3. As a matter of fact, there are three footnotes in that table...

jgvictores, 2018-01-08:
3. Yes: \*, \*\*, and \*\*\* (now visibly rendered lol).

jgvictores, 2018-01-11:
Changed issue title, `Migrate tables to ods/md` -> `Migrate tables to csv`. Related: https://github.com/roboticslab-uc3m/best-practices/issues/20

jgvictores, 2018-02-08:
Finished at ffce5d9918cf5b8c6a81a2dea1d447edde772b15.

closed 2018-02-08 by jgvictores